realsenseai/librealsense #4015: Any way to setup the ROI for auto exposure for the depth (IR cameras) using the JSON file ?

tarekmuallim, 2019-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |   | 
| Operating System & Version |   Win 10 / Linux Ubuntu  | 
| Kernel Version (Linux Only)    |                                            | 
| Platform                                 | PC  |
| SDK Version                            |                          | 
| Language                            |  C++                           | 
| Segment			|                          | 

Hi,
Is there any way to setup the ROI for auto exposure of the depth sensor (IR cameras) using the JSON file uploading?
I didn’t find any reference to explain what parameters can be set in JSON file and what values can added.
I am looking for a detailed list of all parameters which can be setup using the JSON file.

MartyG-RealSense, 2019-05-19:
I hope the link below will help you with your json ROI question.

https://github.com/IntelRealSense/librealsense/issues/2681#issuecomment-436391721

If you visit the Visual Presets documentation page then you will find links to example json files that list a range of different configurations in each one.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets

tarekmuallim, 2019-05-20:
Thank you for your answer.
I am working with c++. If it's possible, could you please provide a c++ example for setting ROI?

Regarding json file examples, it don't contain all the parameters which can be set using json file. Providing a full list of all the parameters which can be set using json file will help us a lot.

MartyG-RealSense, 2019-05-20:
Apparently, support for setting the auto-exposure ROI has been added to ROS.  

https://github.com/IntelRealSense/librealsense/issues/2190

In regard to C++, as far as I know the ROI can be set with an instruction called **set_region_of_interest**

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_sensor.hpp#L356

I do not have a standalone example script for it though.

json is part of the camera's Advanced Mode.  Intel decided not to provide full documentation for Advanced Mode features, as they interact with each other in complex ways and so Intel handles them with machine learning algorithms instead.  

[image]

tarekmuallim, 2019-05-20:
Thanks. 
I will try to use this function.

RealSenseCustomerSupport, 2019-06-10:
----------------------------------------------

Hi.
Does this function work for you?

tarekmuallim, 2019-06-10:
Actually, I was busy and I didn't try it yet.
But I think the functions should work fine.
I will try it asap.
You may close this issue if you want.

closed 2019-06-24 by RealSenseCustomerSupport

woutoe, 2020-03-05:
As this is the first issue you'll find when searching for how to set the auto exposure roi in c++ and it isn't answered properly; It might be useful to share my code for how to do this for both the depth sensor (= stereo sensors) and the rgb color sensor. Tested for the realsense D415 camera. 

```c++
#include <iostream>
#include <librealsense2/rs.hpp>

// Set the configuration
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 15);
cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_ANY, 15); 

// Make sure you first start the pipe
pipe.start(cfg);

// Set the Auto Exposure (AE) Region of Interest (ROI). Should be done after starting the pipe, will give an error otherwise
// Create the ROI for auto exposure (set these values to whatever you need)
rs2::region_of_interest roi;
roi.min_x = 20;
roi.min_y = 450;
roi.max_x = 285;
roi.max_y = 700;

// Given the pipe, we can get a device and query its sensors using:
std::vector<rs2::sensor> sensors_started = pipe.get_active_profile().get_device().query_sensors();

// Set the ROI for auto exposure to all sensors on the device
for (rs2::sensor sensor_in : sensors_started){
	rs2::roi_sensor sensor(sensor_in);

	try {
		sensor.set_region_of_interest(roi);
	} catch (const rs2::invalid_value_error& e) {
		std::cout << e.what() << std::endl;
	}
}
```

The first sensor in the `sensors_started` vector is for the depth cameras. The second sensor in the vector is for the rgb color camera. This script sets them to the same ROI, but this is not required.

Similar issues:
#2681 #4363

MartyG-RealSense, 2020-03-05:
@woutoe   Thanks so much for sharing your solution with the community!

EuroEdit, 2020-04-16:
@woutoe When setting the ROI as you suggest, I get: "UVC header is not available" when trying to retrieve timestamps etc. Any idea why setting the ROI might affect the UVC header?