realsenseai/librealsense #4016: How can I get color frame and IMU data on TX2?

228282416, 2019-05-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   ubunut18.04  | 
| Kernel Version (Linux Only)    |  4.9.140                                   | 
| Platform                                 | NVIDIA Jetson TX2(JetPack4.2) |
| SDK Version                            |       2.21.0      | 
| Language                            |                        | 
| Segment			|  Robot                  | 

### Issue Description
There are some questions I can't fix.
 1.I can't get color frame and IMU data by realsense-viewer.But I can get depth frame.The warnning of device is as follows:
**20/05 11:32:14,241 WARNING [547604545136] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Z16  ]
 20/05 11:32:20,632 WARNING [547579367024] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.**
2.I can't connected to D435i via USB3.0 .The warnning of device is as follows:
**Failed to open scan_element /sys/devices/3530000.xhci/usb2/2-1/2-1.2/2-1.2:1.5/0003:8086:0B3A.0005/HID-SENSOR-200073.1.auto/iio:device4/scan_elements/in_accel_z_en Last Error: Permission denied**
Who can help me? Thank you very much!

ev-mp, 2019-05-20:
@228282416 hello,
1. The warnings notify that the kernel patches were not applied, so there will be no metadata available. These do not affect Depth/IR/RGB streaming/controls functionality and can be disregarded.
2. This warning can be fired if the udev-rules file distributed with the SDK is not up to date. 
With regards to usb2/3 issue - what is the kernel log when plugging the device in?

228282416, 2019-05-20:
@ev-mp Thanks for your responses.My kernel version is 4.9.140.Should I change my kernel version?And there some information about device when I plugging in.
[image]

228282416, 2019-05-20:
I get many warnning messages when I making librealsense project,such as:
1.
` /usr/include/features.h:184:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
  warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"`
2.`/usr/include/c++/7/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]`
3.`/home/sikong/librealsense/third-party/glad/glad.c:1696:16: warning: ISO C forbids conversion of object pointer to function pointer type [-Wpedantic]
  glGetString = (PFNGLGETSTRINGPROC)load("glGetString");`

ev-mp, 2019-05-20:
@228282416 , according to the log the device is properly enumerated. 
It is clear, though, that you're using an unmodified kernel.
- You may consider switch to kernel 4.4  as a quick searching with `tx2` and `imu` keywords provides a handful of discussions that imply Jetsonhacks' patches are provided for 4.4 only.  In any case - there are confirmed reports of users able to run all the streams of D435i on JetsonTX.
- The compilation warnings ,though rather un-pretty, are harmless, and since most of them related to some 3rd party code they will probably remain for a while.

228282416, 2019-05-21:
@ev-mp Thank you again.So,that's mean I have to reflash Jetpack3.2.1 in TX2 ?What a......Well ,I choose to change D435i to another camera.

Tengyun-Mo, 2019-05-22:
Finally,I have installed D435i driver on TX2(Jetpack4.2+kernel4.9.140+librealsense v2.22.0 ).What a pleasure it is.

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Hi @Tengyun-Mo,

Since the issue resolved, I will close this.

closed 2019-09-20 by RealSenseCustomerSupport