realsenseai/librealsense #4017: How to imshow RGB camera

YH53, 2019-05-20:
Required Info
| Camera Model | { D415 } |
| Operating System & Version | {Win 10 |

| SDK Version | 2.0 |
| Language | C++ |
Platform: visual studio 2015

I am new to it, I'm learning the examples in the SDK, in rs-imshow.cpp,The depth stream is shown in this way：

rs2::frameset data = pipe.wait_for_frames(); 
rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
const int w = depth.as<rs2::video_frame>().get_width();
const int h = depth.as<rs2::video_frame>().get_height();
Mat image(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);
imshow(window_name, image);

How can I imshow color stream in a similar way?

RealSenseCustomerSupport, 2019-05-24:
----------------------------------------------

Hi YH53,

You may try following snippet of code.

        rs2::frameset data = pipe.wait_for_frames();
        rs2::frame color = data.get_color_frame();

        const int w = color.as<rs2::video_frame>().get_width();
        const int h = color.as<rs2::video_frame>().get_height();
        Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);
        imshow(window_name, image);

Thanks!

YH53, 2019-06-01:
@RealSenseCustomerSupport 
Thanks！
I‘ve try it.But the color it shows might be wrong.For example, blue things will turn yellow.I guess it seems that some parameter is wrong.    How to do it?

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@YH53 What format do you configure for color? Please configure RGB format for it. I guess maybe BGR format you selected which will cause blue things look like yellow.

YH53, 2019-06-13:
@RealSenseCustomerSupport 
Where can I configure for color?  How to configure RGB format for it?  Thanks!

closed 2019-06-17 by YH53