realsenseai/librealsense #4019: D435  pixel_distance_in_meters

scizors, 2019-05-20:
Hello,
 CAMERA D435
 the value below is set random pixel number.
 I would like to know which pixel_distance you mean is it with RGB camera or stereo camera pixel
python snippet 
        dpt_frame = pipeline.wait_for_frames().get_depth_frame().as_depth_frame()
        pixel_distance_in_meters = dpt_frame.get_distance(100,100)
        print(pixel_distance_in_meters)
output
6.038000106811523
5.866000175476074
5.9800004959106445
6.098000526428223
5.866000175476074
5.9800004959106445
5.9800004959106445
5.9800004959106445
 and so on

ev-mp, 2019-05-20:
@scizors hello,
The line
>dpt_frame = pipeline.wait_for_frames().get_depth_frame().as_depth_frame()

instructs Librealsense to retrieve the frame produced by the Depth sensor, therefore the `frame.get_distance(i,j)` call retrieves the depth data stored in this frame.
Also note that the specific API is available exclusively for `depth_frame`  - RGB stream provides 2D data only (no depth).

scizors, 2019-05-20:
thanks for the reply

ya I know RGB does not have depth values.

the frame you have mentioned. Is it from the stereo camera

if yes the i and j values are from the stereo camera values

scizors, 2019-05-20:
If yes I can close the issue

ev-mp, 2019-05-20:
Maybe you're looking for the `aligned` depth, i.e. a depth frame sized as RGB frame with  1:1 mapping between data in depth and color pixels. 
This use-case is elaborated in the corresponding example https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

scizors, 2019-05-20:
why there is a clipping aligned depth .Can I remove clipping and do aligned 

what is frame.get_distance(i,j) i and j what it means is depth data stored in this frame from stereo camera

ev-mp, 2019-05-20:
@scizors hello, this thread will be closed as your previous response suggests.
The alignment-related discussion will be consolidated and continued in #4021.

closed 2019-05-20 by ev-mp