realsenseai/librealsense #4021: Align Depth to Color get depth distance

scizors, 2019-05-20:
camera D435
python code Align Depth to Color

how can I get distance from depth to color 
code error
dpt_frame = pipeline.wait_for_frames(). aligned_frames.get_depth_frame() .as_depth_frame()
pixel_distance_in_meters = dpt_frame.get_distance(369,42)
error
   dpt_frame = pipeline.wait_for_frames().aligned_frames.get_depth_frame() .as_depth_frame()

AttributeError: 'pyrealsense2.pyrealsense2.composite_frame' object has no attribute 'aligned_frames'

ev-mp, 2019-05-20:
@scizors hello,
The aligned frame a synthetic stream generated in Librealsense by a dedicated processing block.
Please  review the documentation and examples to understand how the alignment is invoked and consequently used:

Introduction:
- https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#frame-alignment

C++ Tutorials:
- https://github.com/IntelRealSense/librealsense/tree/master/examples/align
- https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced

Python example:
- https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

scizors, 2019-05-21:
thanks for the reply

scizors, 2019-08-07:
Hello,
 How can I zoom in and out using aligned stream. Is it possible

closed 2019-11-15 by scizors