realsenseai/librealsense #4023: 2.21.0 release build failing on Ubuntu 18.04 due to glf

Algomorph, 2019-05-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | Need to be able to build the viewer first to check... | 
| Operating System & Version |   Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.2.5 | 
| Language                            |  (All?)                          | 
| Segment			|  ? | 

### Issue Description
Build fails w/ gcc 7 (issue partially similar to https://github.com/IntelRealSense/librealsense/issues/2352):

```/usr/bin/ld: CMakeFiles/rs-latency-tool.dir/rs-latency-tool.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/build.make:99: recipe for target 'wrappers/opencv/latency-tool/rs-latency-tool' failed
make[2]: *** [wrappers/opencv/latency-tool/rs-latency-tool] Error 1
CMakeFiles/Makefile2:491: recipe for target 'wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/all' failed
make[1]: *** [wrappers/opencv/latency-tool/CMakeFiles/rs-latency-tool.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2```

Algomorph, 2019-05-20:
Just encountered the same bug on the master branch.
Here is partial CMake config used, I think should have everything that's relevant, let me know if not:
[image]

Algomorph, 2019-05-20:
Build fails with -DBUILD_WITH_OPENMP=ON, build succeeds with -DBUILD_WITH_OPENMP=OFF

radfordi, 2019-05-23:
Hi @Algomorph.  OpenMP support was dropped about 6 months ago here 65eb8b593ca85f6cf96b04f0313755abc1666f11.  It looks to me like not much is different when you build with `BUILD_WITH_OPENMP` today.  In particular, we don't actually build with OpenMP.  I'm not sure if support for OpenMP was dropped on purpose or not, but setting that doesn't even try and link with, or use the command line options to enable, it.  Instead, it just disables linking with `pthreads`, possibly leading the error you ran into.  #4049 may or may not fix it for you, but it does build for me on 18.04.

Algomorph, 2019-05-24:
@radfordi , I see, thank you, I see what the issue is. I'm using prebuilt debian packages right now, it seems sufficient ATM. Going to wait and see what happens to the OpenMP support. This issue should probably be kept open until either the OpenMP support is restored (per that branch you point out) or someone fixes the CMake to remove the BUILD_WITH_OPENMP option altogether.

commit 8cdc47e1 (2019-05-29): Merge pull request #4093 from radfordi/openmp

Restore OpenMP Support (Fixes #4023)
closed 2019-06-11 by ev-mp