realsenseai/librealsense #4024: Struggling to get export_localization_map to work

mmoyes, 2019-05-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 2.21 & 2.22 | 
| Operating System & Version |    Linux (Ubuntu 19.04)  | 
| Kernel Version (Linux Only)    |  5.0.0-15-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.21 & 2.22                         | 
| Language                            |  C++                         | 
| Segment			|  AR                         | 

### Issue Description
Struggling to get export_localization_map to work. Works fine in realsense-viewer. 

I do the usual:
```
rs2::context ctx;
pipe = rs2::pipeline(ctx);
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
pipe.start(cfg);
```

Then I've tried various iterations of:
```
std::vector<rs2::sensor> sensors = pipe.get_active_profile().get_device().query_sensors();
					for ( auto &sensor : sensors) {
						if ( sensor.is<rs2::pose_sensor>() ) {
							sensor.as<rs2::pose_sensor>().stop();
							auto blah = sensor.as<rs2::pose_sensor>().export_localization_map();
						}
					}
```
The sensor.stop() I tried replacing with pipe.stop() and removing completely. I'm working off the assumption something needs to be stopped as in realsense-viewer

I get the following errors (dependent on variation or stop)

```
15:12:53.241 [25213] [E] Device-AD30: MessageID 0x1003 (SLAM_GET_LOCALIZATION_DATA) failed with status 0x1
15:12:53.241 [25213] [E] Device-AD30: USB Error (0x1)
RealSense error calling rs2_get_raw_data(buffer:nullptr):
    null pointer passed for argument "buffer"
mtTrack: ../../libusb/io.c:2116: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
```
or
```
15:13:35.505 [25393] [E] Device-AD30: State [ACTIVE_STATE] got event [ON_ASYNC_START] ==> [ERROR_STATE]
15:13:35.505 [25393] [E] Device-AD30: Entered state [ERROR_STATE]
```

Any pointers or working examples would be much appreciated.

ev-mp, 2019-05-21:
@mmoyes hello,
There is a limitation that after stopping the sensors the map cannot be exported immediately due to internal processing (serialization).  You can work around this by adding a sleep between `stop` and `export` calls. (from my experience ~5sec for 5 min running session).
Also note that in order to generate even a minimal map the sensor need a certain amount of running time  (~10 sec) during which the sensor should to be rotate it a little to obtain enough environmental features.

mmoyes, 2019-05-22:
Hi @ev-mp ,
Thanks for the response. Great that seems to work well.
I'm now doing...
```
tracking_sensor.open(pose_stream);
tracking_sensor.start(callback);
// slowly moving around for a minute to try and collect tracking features
tracking_sensor.stop();
// sleep for 5 seconds
std::vector<uint8_t> map = tracking_sensor.as<rs2::pose_sensor>().export_localization_map();
//write map to disk
```
Then I move the camera to a new location within the same space.
Then `tracking_sensor.open(pose_stream);`... `tracking_sensor.as<rs2::pose_sensor>().import_localization_map(map);`...etc

However, I do not seem to be getting the updated tracking information in. I'm expecting to see the position jump to the original coordinates. Is this not correct? Is there a way to force a relocate?

vincentchu, 2019-05-22:
@ev-mp Thanks again for the response. Is this expected to work in python as well? I tried sleeping 10 seconds and continue to get exceptions:

```
Traceback (most recent call last):
  File "record.py", line 114, in <module>
    slam_map = pose_sensor.export_localization_map()
RuntimeError: null pointer passed for argument "buffer"
```

My python code looks like:

```python
pipeline = rs.pipeline()
profile = pipeline.start()
pose_sensor = pipeline.get_active_profile().get_device().first_pose_sensor()

# Record frames for awhile ...

pipeline.stop()
time.sleep(10)
slam_map = pose_sensor.export_localization_map() # Fails here
print(slam_map)
```

ev-mp, 2019-05-22:
@vincentchu hello, please repeat the scenario and make sure to move and/or tilt the camera gently during streaming, e.g, by holding it in hand. 
In my observation 15-20 seconds of streaming is equivalent to 50-200kb localization map.

mmoyes, 2019-05-22:
As in #4040 not sure if there is a way to use or verify the imported map. Are there any methods currently implemented for re-localization calls?

vincentchu, 2019-05-22:
@ev-mp I'm still unable to get any data out of the device, in both C++/python.

```c++
auto profile = pipe.start();
auto device = profile.get_device();
 auto pose_sensor = device.first<rs2::pose_sensor>();

// collect for 15-20s

printf("Stopping pose_sensor\n");
pose_sensor.stop();

std::printf("Exporting map ... sleeping \n");
std::this_thread::sleep_for(std::chrono::seconds(10));

std::vector<uint8_t> results;
results = pose_sensor.export_localization_map();
```

Using my C++ code, I get this output:
```
Stopping pose_sensor
14:58:00.453 [0] [E] Device-C800: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0
Exporting map ... sleeping
14:58:00.559 [0] [E] Device-C800: Entered state [ERROR_STATE]
Exception detected ...null pointer passed for argument "buffer"
```

Is there some issue with my firmware? The output from `rs-enumerate-devices`:

```
vince@local:c++ → rs-enumerate-devices
Device info:
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	905312111088
    Firmware Version              : 	0.0.18.5502
    Physical Port                 : 	vid_8087 pid_0B37 bus_0 port_3
    Product Id                    : 	0B37

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF
```

I'm using the Mac support for the T265 on the `development` branch.

vincentchu, 2019-05-26:
@ev-mp After some struggling, I think my problem was related to an older version of librealsense because I was able to successfully export mapping data. My next question is around using this exported mapping data.

Reading the code, the C++ signature says that the returned data is `std::vector<uint8_t>` (which is a `list` type in Python). Does this mapping data represent tracked cartesian points in 3d? What's the appropriate way to utilize this data? 

I took a quick look and the number of points returned does not seem to be a multiple of 3 (if they were cartesian points). Is there some other serialization format that I need to use?

Thanks.

radfordi, 2019-05-26:
@vincentchu, the map data is meant to be opaque.  The way you interact with the map is, before saving it, to define a named location relative to the current session's coordinates with `set_static_node`:
```c++
        bool set_static_node(const std::string& guid, const float3& position, const float4& orientation);
```
and then after loading that saved map in a subsequent session, you can ask for its location, relative to the current session's coordinates, with `get_static_node`.
```c++
        bool get_static_node(const std::string& guid, float3& position, float4& orientation);
```

The `set` in the original session will not be successful unless the device is returning high confidence, so you may need to try a few times as the confidence can change at any time.

The `get`, in subsequent sessions will not be successful until the device is able to find that place on the map.  This may never happen if the environment has changed sufficiently or if you are in a previously unmapped place, etc.  Once the `get` returns a pose, in the ideal case,  it will continue to return the exact same pose the next time you ask, but in practice the device will continue to refine the pose of the saved location as it gets more information.  In particular, as you get closer it might have a more accurate pose for the saved location.

The easiest way to get started is to just do
```c++
    while(!set_static_node("origin", float3{0,0,0}, float4{0,0,0,1}))
          ;
```
before saving the map and then, after loading the map, to call `get("origin")` until it returns `true` and pretend the returned "origin" will never change.

vincentchu, 2019-05-26:
@radfordi Thank you for that explanation. I understand that the exported maps are meant to be opaque, but is there no way (documented or otherwise) to deserialize underlying data? I'm particularly interested in visualizing the maps that the camera is generating and using.

radfordi, 2019-05-26:
@vincentchu, there is currently no documented description of the underlying map data.

vincentchu, 2019-05-26:
@radfordi Is this planned on the roadmap? I would love to see this feature!

radfordi, 2019-05-26:
@vincentchu, it's not on the roadmap, but it is being discussed.  We understand the utility of being able to visualize the maps.

mmoyes, 2019-05-28:
Thanks @radfordi setting and getting origin makes sense. However, unable to replicate. Getting lots of

> Device-DC40: ERROR: Bulk transfer message 0x100B (SLAM_GET_STATIC_NODE) response to host got LIBUSB_ERROR_TIMEOUT. Host did not return answer

and

> Device-DC40: ERROR: Bulk transfer message 0x100B (SLAM_GET_STATIC_NODE) request to device got LIBUSB_ERROR_TIMEOUT. Bytes requested 136, bytes transferred 0

when I try `get_static_node`

Tried increasing usbfs_memory_mb.

Using Api 2.22

Attached sloppy test code in case it helps:
[mapIO.cpp.txt](https://github.com/IntelRealSense/librealsense/files/3228899/mapIO.cpp.txt)

giovannicairoli, 2019-06-05:
Goodmorning everyone,

I wanted to say thank you, these are the answers I was looking for to start saving and importing the map useful for the localization in a known environment. 
Before starting coding, I would like to be able to save the map by exploiting the viewer (as mentioned from @mmoyes at the beginning). Probably it sounds so stupid but it doesn't work. Could you please help me?
I have both the D435i and the T265 connected. I supposed that for exporting the map I needed to start recording using the button on the left. 
But this error appears:
Realsense error Calling: UNKNOWN in rs2_create_record_device_ex(device:33C373C0, file:C):
Error opening file: C:\Users\Giovanni/20190605_113800.bag.

Could you please help me?

I'd like to say thank you to you in advance.

Giovanni

radfordi, 2019-06-05:
Hi @mmoyes,

The messages you are seeing are needlessly verbose and like just indicate that you need to move the system around until it converges.  We have a PR to quiet them #4152.

The call to `set_static_node` will not succeed until the system has "converged" which is indicated by `tracker_confidence` (0-none, 1-low, 2-med, 3-high) being at least `3`.  These correspond to the path colors (0-grey, 1-red, 2-yellow, 3-green) in the `realsense-viewer`'s 3D view.

radfordi, 2019-06-05:
Hi @giovannicairoli,

`rs2_create_record_device` is for recording the raw data (images, gyro, accel) not for saving a localization map.  I suspect `rs2_export_localization_map`, or the C++ version `export_relocalization_map`, is what you are looking for.  I don't believe we have added this functionality to the `realsense-viewer` yet.

giovannicairoli, 2019-06-05:
Hi, @radfordi,

So, if I understood, there is no possibility to export, import and localize the camera in the realsense viewer! Right? So I will start trying coding following the code posted before.
I will replay to you if I need more help.

Thank you.

Giovanni

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

Since the issue clarified, still any support for this topic? If not, will close this.

vincentchu, 2019-06-19:
@RealSenseCustomerSupport Thank you, please feel free to close.

mmoyes, 2019-06-20:
@RealSenseCustomerSupport 
Is more work going in to smoothing out the problems in the current system. I am having to continuously try to get the stored position. System seems to crash occasionally when doing so. Is documentation or an recommended example coming?

Thanks

closed 2019-06-20 by RealSenseCustomerSupport

Sttrr, 2019-09-09:
> Thanks @radfordi setting and getting origin makes sense. However, unable to replicate. Getting lots of
>
>
> and
>
>
> when I try `get_static_node`
>
> Tried increasing usbfs_memory_mb.
>
> Using Api 2.22 [...]

 Hello, @mmoyes I also encountered the same problem when I try `get_static_node`,and the program crashed. how did you solve this problem？thank you!!!

mmoyes, 2019-09-09:
@Sttrr I'm still struggling to get this working consistently (in a usable way). Using version 2.26 onwards helps. I mostly wrap a try catch in a while loop: While (get static node not successful) { try get static node }