realsenseai/librealsense #4025: Reading .bag files on AWS

tRosenflanz, 2019-05-20:
Is it possible to read prerecorded .bag files on AWS EC2 instance?

We will have a large number of large .bag files and postprocessing them on EC2 could greatly speed it up versus doing on local machines. However, the bag files refuse to be read and time out after first `wait_for_frames()` call using python wrapper. 

We tried following these directions https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md but the kernel patches fail with `Loading the modified modules into kernel...modprobe: FATAL: Module hid_sensor_gyro_3d not found.` although the AMI uses Ubuntu 18 but this isn't that surprising overall since we are dealing with virtual machines.

ev-mp, 2019-05-21:
@tRosenflanz hello, while I have no first-hand experience with EC2, the kernel patches are needed only when working with a physical device.
Running rosbag files with Librealsense does not require any kernel patches/ udev rules modifications at all.

As for running in the cloud - you may review this [Jupiter tutorial](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb) that runs on cloud-based rosbag records (#2088)

tRosenflanz, 2019-05-29:
Thank you for the tutorial. Issue ended up being on our side. Some files we tested on AWS actually had corrupted channels while the ones we tested locally didn't.

closed 2019-05-29 by tRosenflanz