realsenseai/librealsense #4028: open pipeline fails with invalid value

kartur, 2019-05-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.22.0                          | 
| Language                            | C/C++ OpenCV }                          | 
| Segment			|  others  | 

### Issue Description
The RealSense Viewer works fine, but I can't get the realsense working when programtically using the library. The camera is connected properly and shown as USB3.2 in the realsense-viewer. I always get some invalid value exception when trying to get the pipeline started.

[image]
[image]

ev-mp, 2019-05-21:
@kartur hello,
Execute the demos uninterrupted to allow it to terminate with more meaningful error messages.
Additionally - try several SDK-distributed demos -  `realsense-viewer`, `rs-capture` and  `rs-enumerate-devices` - and share the log/screenshots.

kartur, 2019-05-21:
Hi ev-mp,

ah I see, if I don't stop at the exception, it actually continues and works. Here is what's in the output console:
`
Exception thrown at 0x00007FFC2B239129 in align.exe: Microsoft C++ exception: librealsense::invalid_value_exception at memory location 0x00000053C3B9D120.
Exception thrown at 0x00007FFC2B239129 in align.exe: Microsoft C++ exception: std::runtime_error at memory location 0x00000053C3B9DE10.
Exception thrown at 0x00007FFC2B239129 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
`
(The "number provided was invalid" messages appears multiple times)

But when using my own program, it still doesn't work, even uninterrupted.

Here the requested data:
[image]
[image]
[rs-enum-dev.txt](https://github.com/IntelRealSense/librealsense/files/3202702/rs-enum-dev.txt)

ev-mp, 2019-05-22:
@kartur hello, what is the configuration you're passing to `cfg` when calling `pipe.start(cfg)`, can you share the actual code?
Also - the log you've posted refers to align.exe:
>Exception thrown at 0x00007FFC2B239129 in align.exe:

Is this the SDK's `rs-align` demo, or your code, please elaborate.

kartur, 2019-05-22:
@ev-mp regarding the align demo, I did not change the code there. I just use it to confirm the realsense (lib) is working properly. I get the same errors/exception in the other demos too, e.g. hello-realsense.
[image]

I would prefer not to share my code (it has a lot of dependencies anyway). But I can just say it worked before. I'm not sure why it doesn't work now, it could be the new SDK version, but I'm not 100% sure. Personally I think it would be best to get the SDK examples working properly first.

kartur, 2019-05-26:
@RealSenseCustomerSupport

Why close this issue? I've got the latest FW and SDK version, and the examples don't work properly with D435i..

ev-mp, 2019-08-13:
@kartur, the issue was reopened, as requested.
Can you reiterate the reproduction scenario and the error symptoms?
Thanks.

rhlvora, 2020-02-02:
I think I'm getting the same thing.

If, as an example, I run rs-pose in the sdk samples I crash when the code tries to start the pipeline.

I have a D415, and I'm running on windows 10 in visual studio 2019.

[image]

[image]

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

rhlvora. The error is correct. You are trying to start a pipe with POSE streaming enabled which is only supported on an Intel T265 camera not the D415.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@kartur. We are unable to reproduce this issue. Hello-realsense example runs as expected on both a D415 and D435i. Can you provide some more information on what you might be doing differently?

Thanks

rhlvora, 2020-02-07:
> rhlvora. The error is correct. You are trying to start a pipe with POSE streaming enabled which is only supported on an Intel T265 camera not the D415.

Sounds good, thanks!

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

@kartur

Do you still need assistance with this?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-04-05 by RealSenseCustomerSupport

RohitSingh1226, 2020-07-30:
Is there any way of running the ar-basic and ar-advanced example on D415, I think its not able to run cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

@RealSenseCustomerSupport

ev-mp, 2020-07-30:
@RohitSingh1226 no.
The `RS2_STREAM_POSE` streaming format is an [unique capability of T265 device](https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#t265-tracking-camera).

foxtierney, 2024-09-03:
I also have run into this problem just today. It previously was not happening (I unfortunately don't know what I did between it working and not working since I took a good deal of time checking other places in my code first).

Set-up:
- Windows 10
- Visual Studio Community 2019 Version 16.11.13
- Intel RealSense SDK 2.34.0.1448 (I need this version since last I checked the virtual camera align doesn't work after 2.34)

My symptoms are as follows:
- Sample project can run but throws many errors (below)
- Personal project also throws errors but breaks
- Personal project but with hello-realsense.cpp only throws errors and cannot recover either

Error output from untouched align.exe sample run in the IDE:
```
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: librealsense::invalid_value_exception at memory location 0x000000903FF2BC00.
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: std::runtime_error at memory location 0x000000903FF2CA00.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 (KernelBase.dll) in align.exe: WinRT originate error - 0xC00D36B3 : 'The stream number provided was invalid.'.
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: librealsense::io_exception at memory location 0x00000090416FF680.
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: librealsense::io_exception at memory location 0x00000090416FF680.
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: librealsense::io_exception at memory location 0x00000090416FF680.
The thread 0x3044 has exited with code 0 (0x0).
The thread 0x44c8 has exited with code 0 (0x0).
Exception thrown at 0x00007FFD765BB699 in align.exe: Microsoft C++ exception: librealsense::io_exception at memory location 0x00000090416FF680.
```

I suspect this is some kind of deep problem embedded in some configuration file...

MartyG-RealSense, 2024-09-03:
Hi @kartur  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/5693#issuecomment-737682648 who experienced a memory location exception suggested that it could occur when switching from Debug to Release build configurations in a Visual Studio project.

foxtierney, 2024-09-04:
@MartyG-RealSense, Thank you for pointing that out. Unfortunately, I already saw that and attempted that method twice (I probably should have mentioned it at first). I with two slightly different orders of operation:
First
1. Uninstall RealSense SDK via the control panel
2. Click clean in the visual studio project
3. Build the project (I did hit built and not rebuild)
Second
1. Uninstall Realsense SDK via the control panel
2. Restart my computer
3. Click clean in Visual studio
4. Rebuild the project
Neither of these seemed to work. Maybe there is some specific order or small operation like having something open or not open when doing the uninstall that has an effect.

MartyG-RealSense, 2024-09-04:
The link below describes how to compile the RealSense SDK's official example projects in Visual Studio.

https://www.intel.com/content/www/us/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

When using RealSense example code in your own personal project in Visual Studio, a different build process is required as there are linkages to the RealSense library that need to be established.  This can be done with three **.props** files provided by the SDK.  The link below has information about this.

https://support.intelrealsense.com/hc/en-us/community/posts/4412175411859/comments/4412182682387

foxtierney, 2024-09-05:
>This can be done with three .props files provided by the SDK.

Thank you. I have done exactly that and will explain my full process and details below.

>As a note for those that come to this thread later, it seems that Visual Studio Community 2022 may not be able to use the .props files. However, the 2019 version does. Unfortunately, 2019 is somewhat challenging to get now but you can find the initial release at archive.org: https://archive.org/details/vs_Community

After last time I have done the following:

1. Uninstalled RealSense SDK and manually removed any remaining files I could find in `Program Files (x86)` and the `AppData` folders (I didn't find anything in these last folders)
2. Uninstalled Visual Studio 2019 and manually removed any remaining files from both `Program Files` and `Program Files (x86)` as well as from the `AppData`
3. Deleted the entire Visual Studio project folder (so that none of the remaining files remain)
4. Reinstalled Visual Studio Community 2019, selecting the C++ options
5. Reinstalled RealSense SDK version 2.3.40
6. Restarted my computer
7. Opened Visual Studio and created an empty project
8. Went to `View > Other Windows > Property Manager`, right clicked my project, clicked `Add Existing Property Sheet` and selected one of the `.props` files from `C:\Program Files (x86)\Intel RealSense SDK 2.0`. I then repeated this for the other two `.props` files.
9. Added `rs-hello-realsense.cpp` from `C:\Program Files (x86)\Intel RealSense SDK 2.0\samples\hello-realsense` to the project by right clicking `Source Files` from the `Solution Explorer` window, then `Add > Existing Item...` and navigating to the file.
10. Run the project (Without touching the build version or 32/64 bit selection) by clicking on `Local Windows Debugger`
11. The program runs, throws the previously shown exceptions, but does proceed to show the depth in the console as expected.

When trying it with my own .cpp file, I also get the errors but it does seem to move past them... Maybe it's not an issue now or I was forgetting something previously. In any case, my issue seems to be resolved, though the exceptions persist.

MartyG-RealSense, 2024-09-05:
Thanks so much for sharing your process in such detail.  I'm pleased to hear that the programs can run despite the exceptions.

I would recommend selecting the architecture that your program is using (32-bit x86 or 64-bit x64) instead of using AnyPC (which has caused compilation problems in past cases).