realsenseai/librealsense #4031: Jetson Nano + D435i Issues

kat-mcg-lin, 2019-05-21:
> How can we get the D435i to work properly?? I know the notes say it won't get recongized, but there must be some way?! Are there any other external patches or something? Jetson Hacks made a video about connecting a nano to a D435i and they got it to work there.... what did they do that isn't included in the git?!

I've asked this question elsewhere, hoping to get some answers here. I've been trying to get the D435i to work for awhile, I have the T265 working just fine, but the D435i isn't being recognized after following the popular instructions from Jetson Hacks

JvanZijl, 2019-08-14:
@kat-mcg-lin Have you been able to solve this issue?

malith1992, 2019-09-05:
I have the same issue.

dorodnic, 2019-09-05:
Latest builds come with librealsense2 packages for Jetson. Please try by following distribution.md (skip the dkms) Works well on a nano for me. Also, same can be accomplished by building with libuvc (see relevant instructions under doc folder)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mickeyouyou, 2019-09-10:
Do you mean following `distribution_linux.md` ? skip the dkms install by apt ? there is no arm version packages I think.

dorodnic, 2019-09-10:
Yes
There are arm packages 🙂

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mickeyouyou, 2019-09-10:
ok  thank you point that, I must have a try now .

mickeyouyou, 2019-09-10:
meet some errors when run `sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u` for ubuntu 18 

```bash
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'arm64'
```

dorodnic, 2019-09-10:
Maybe because of bionic vs xenial? Sorry for that. @ev-mp any idea?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mickeyouyou, 2019-09-10:
It's okay. :smile: 

So do you have any suggestions for installation doc of D435i?  for now , D435I device could not detected by nano .

dorodnic, 2019-09-10:
Me and @ev-mp will check how to offer debians for bionic as well, but for xenial arm build we basically followed libuvc-installation.md document / script. Key parameter is -DFOrce_libuvc=true which lets us bypass kernel patches. In 2.25 I think we ensured IMu would also be detected using this method 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mickeyouyou, 2019-09-11:
hi @dorodnic I build liubrealsense master(version 2.28) successfully. Nano can detect D435i . 

but when run `realsense-viewer` meet some error like bellow :

```bash
 11/09 13:10:42,564 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 11/09 13:10:42,634 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 11/09 13:10:45,105 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 11/09 13:10:46,288 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 11/09 13:10:46,499 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 11/09 13:10:47,139 WARNING [547222434240] (messenger-libusb.cpp:94) bulk_transfer returned error, endpoint: 134, error: Resource temporarily unavailable, number: 11
 
```

and CPU resources  like 3/4 80%+ 

[image]

is it normal ?

dorodnic, 2019-12-10:
Normal - yes.
Good - no. We are still looking into ways this can be improved.

bgilreath, 2020-02-12:
Any progress on this? My issue results in same bulk_transfer with D435i on Jetson Nano, 4.9.140-tegra, RealSense ROS v2.2.8, Running with LibRealSense v2.28.1, Device FW version: 05.12.01.00

arjunsengupta1998, 2020-02-17:
I got the following issue while running realsense-ros on rviz:

 15/02 05:20:53,264 WARNING [547163709824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/02 05:20:53,325 WARNING [547163709824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/02 05:20:53,430 WARNING [547163709824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/02 05:20:53,491 WARNING [547163709824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Can anyone explain what this means?

sadransh, 2020-03-02:
@arjunsengupta1998  Did you find any way to solve this issue? 
@mickeyouyou how about you?

mickeyouyou, 2020-03-04:
Yeah  I build realsense with https://github.com/JetsonHacksNano/installLibrealsense

arjunsengupta1998, 2020-03-04:
I am still getting these errors. I installed it from JetsonHacks as well. 
@mickeyouyou did you run the installLibrealsense script or build realsense script?
I ran the install script first. However, pyrealsense2 wasn't installed with it, so I built the sdk from source. Pyrealsense2 was installed, but still getting those warnings.

q8wwe, 2020-03-09:
Im getting the same issue on Jetson TX2 :+1: messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

any solution

jorgemia, 2020-03-26:
I get `[548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61` when starting the depth stream of the D435i with a Jetson Nano.

Then I start getting `WARNING [548212306368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11` 

I'm running Realsense 2.33.1. and D435i firmware 05.12.03.00

RealSenseCustomerSupport, 2020-06-08:
----------------------------------------------

Hi,

Are you still experiencing issues with this?

Try installing the packages using the instructions here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Just be sure to first remove all traces of anything RealSense to prevent any potential conflicts using the instructions here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

Thanks

RealSenseCustomerSupport, 2020-06-11:
----------------------------------------------

FYI if we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-06-22 by RealSenseSupport