realsenseai/librealsense #4033: rs-measure.cpp in python

scizors, 2019-05-21:
do we have rs measure in python if yes can any one share it

MartyG-RealSense, 2019-05-21:
A RealSense community member did some conversions of Measure into Python.  Links to their scripts, and advice from their creator, can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/3948#issuecomment-491228272

There is also the 'box_dimensioner_multicam' example program for Python:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

scizors, 2019-05-22:
thanks for the reply 
https://github.com/IntelRealSense/librealsense/issues/2204#issuecomment-414497056

closed 2019-05-22 by scizors

pfcouto, 2022-12-03:
Hello @scizors  @MartyG-RealSense, based on this issue. Is there any solution that can help me get de width and height of an detected object using the depth? Or something other than the pixels, since it isn't very accurate? Can you help?

Thanks for your help!

MartyG-RealSense, 2022-12-04:
Hi @pfcouto  If you are using Python then the RealSense SDK's **TensorFlow** wrapper may be helpful to you.

https://dev.intelrealsense.com/docs/tensorflow-with-intel-realsense-cameras

pfcouto, 2022-12-04:
Hi @MartyG-RealSense I am using Detectron2 with pytorch to inference. You don't know any way to get the width of an object?