realsenseai/librealsense #4034: MATLAB issue, device.as error

upxela, 2019-05-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.22.0                          | 
| Language                            |  matlab                          | 

### Issue Description
I am using the MATLAB wrapper from the Windows Installer and MATLAB R2018A to read through a prerecorded bag file. To do this, I will set_real_time to false. This is my MATLAB code:

```
pipeline = realsense.pipeline();
config = realsense.config();
config.enable_device_from_file(name, false);

pipe_prof = pipeline.start(config);
device = pipe_prof.get_device();
playback = device.as('playback');
playback.set_real_time(false);
```

However, while doing so I get the following error:

>Brace indexing is not supported for variables of this type.
Error in realsense.device/as (line 96)
                    dev = realsense.playback(out{:});
Error in readBag (line 10)
playback = device.as('playback');

upxela, 2019-06-10:
Has anybody else had similar problems with this/reproduced the error?

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@upxela Could you please try below to see if you still have issue?

    playback = realsense.playback(device)

upxela, 2019-06-13:
@RealSenseCustomerSupport Following your advice I changed the code to:

```
pipeline = realsense.pipeline();
config = realsense.config();
config.enable_device_from_file(name, false);

pipe_prof = pipeline.start(config);
device = pipe_prof.get_device();
% playback = device.as('playback');
playback = realsense.playback(device); 
playback.set_real_time(false);
```

My new error message is as follows:

> Error using realsense.playback
Error: File: playback.m Line: 55 Column: 27
'current_status' is already defined as a method in the 'playback' class.

> Error in readBag (line 11)
playback = realsense.playback(device);

lramati, 2019-06-13:
As an immediate fix, you can replace line 60 of playback.m with `function stop(this)`

upxela, 2019-06-13:
@RealSenseCustomerSupport  @lramati Thanks for that. Now I have a problem with constructing the playback. Here is my error message:

>Not enough input arguments.
>Error in realsense.playback (line 6)
           this = this@realsense.device(handle, index);
>  Error in readBag (line 11)
playback = realsense.playback(device);

What exactly are `handle` and `index`? I am guessing that I need to change `playback = realsense.playback(device); ` to have the parameters match `handle` and `index`.

lramati, 2019-06-13:
The parameters `handle` and `index` were supposed to come from inside the device instance.
You can fix the `.as('playback')` syntax by replacing `out{:}` on lines 88, 94, and 96 with `out`. A change in how devices are handled in the backend appears to have not been propagated to this function.

upxela, 2019-06-13:
@lramati As clarification, you meant to change `playback = realsense.playback(device);` back to `playback = device.as('playback');` after replacing the `out{:}`?

Doing so, I still get the same error:

>Brace indexing is not supported for variables of this type.
Error in realsense.device/as (line 96)
                    dev = realsense.playback(out, -1);
Error in readBag (line 10)
playback = device.as('playback');

lramati, 2019-06-13:
It just worked on my machine. I'm also confused because there doesn't actually appear to be any brace indexing on the line it's erroring on. Could you confirm the changes got saved and maybe restart the Matlab environment just in case?

upxela, 2019-06-13:
Restarted matlab and it worked. Thanks!

closed 2019-06-13 by upxela