realsenseai/librealsense #4035: Disparity shift formula

BlackAlpaca, 2019-05-21:
| Camera Model                       | D435 | 
| Firmware Version                   | TBD | 
| Operating System & Version |  Win 10) |                           | 
| Platform                                 | PC |
| SDK Version                            |  2.x.x                          | 
| Language                            |  C++                           | 
| Segment			|  others                           | 

### Issue Description

Hi everyone

My question is mostly regarding the disparity shift.
For our application there is a fixed maximum length which we need to be able to see in certain usecases. So I was thinking of using disparity shift to narrow down the range. For usecases X and Y, a shift of for example 5 or 10 suffices.

Using the [tuning guide](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) I can see a graph which shows the influence of the disparity shift on the D415 camera.

But now I was wondering if there is a formula to calculate the minZ and the maxZ for a certain disparity shift value?
For disparity shift value 50 with a D415 the minZ is 30cm and the maxX is 110cm. Is there a formula to find these values? (Also usable for the 435 camera)

Thanks in advance!

MartyG-RealSense, 2019-05-21:
My research of your question found a research paper that expands upon the disparity shift graph in the tuning guide to provide much more detail on the subject, again testing with the D415 camera.

https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6387018/#sec3-sensors-19-00489title

The paper provides formulas for calculating MinZ and MaxZ if you know the disparity shift value.

[image]

BlackAlpaca, 2019-05-21:
@MartyG-RealSense As I know the parameters for the D435 I can easily apply it for that camera.
Thank you for the very useful and to-the-point reply, this was exactly what I was looking for!

Wishing you all the best, thanks again!

closed 2019-05-21 by BlackAlpaca