realsenseai/librealsense #4036: Viewing T265 Freezes realsense-viewer

tpwrules, 2019-05-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |  4.15.0-50-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.22.0                 | 

### Issue Description
I installed librealsense 2 as described in the [Linux documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).

Running realsense-viewer shows the attached T265. Clicking the slider to turn on the Tracking Module immediately freezes realsense-viewer, no matter which streams are selected out of the Available Streams list. There is no freeze when streaming from an attached D435. realsense-viewer still freezes no matter if the D435 is plugged in or not.

The rs-pose and rs-pose-and-image demos seem to work. They show the data updating as the camera is moved.

ev-mp, 2019-05-21:
@tpwrules hello,
 - What is the graphic card you're using - is it a discrete or an on-board type ?
 - Can you try to experiment with toggling off GLSL shaders in the `realsense-viewer` and see whether this has any effect?
[image]

tpwrules, 2019-05-22:
The graphics card is an Nvidia Quadro P2000

Those settings have no effect. The graphics works fine with a D435 (I can see the color and depth in 3D).

minihops123, 2019-05-24:
I have exactly the same issue. I have a  Quadro M2000M  and an NVIDIA driver version 410.104 installed

martinakos, 2019-06-12:
Same issue here. The same firmware and librealsense SDK version as the OP. I have a Quadro M2000M and NVIDIA driver version is 396.82

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Hi @tpwrules, @martinakos, minihops123

This looks a hardware dependent issue. Since I don't have that platform, can you verify if the latest librealsense (ex: 2.25.0) resolve the issue?

levingerdes, 2019-10-01:
I have the same issue with librealsense 2.25.0 (I've also tried 2.29.0) and a slightly different setup:

Ubuntu 18.04.3 LTS with kernel version 5.0.0-29-generic
NVIDIA Quadro P1000, NVIDIA driver version 430.50
T265 firmware version 0.1.0.279

The realsense-viewer works absolutely fine with the D435, but turning on the tracking module of the T265 freezes the realsense-viewer.

Having a T265 connected to the PC when starting the realsense-viewer from a terminal shows me these errors:
```
14:27:34.224 [23028] [E] infra/Poller: poll: epoll_wait error 4
14:27:34.224 [23028] [E] Dispatcher: handleEvents(): Poller::poll() ret -1
```
Without having clicked anything in the viewer. If I have both the D435 and the T265 connected, I still see these error messages but I can view all streams from (only) the D435 without issues.

Unfortunately, the mentioned GLSL settings have no effect either.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @levingerdes,

Any dmesg log? If only T265 connected, is it working?

closed 2019-11-08 by RealSenseCustomerSupport

NhatTanXT3, 2019-12-09:
I have the same issue with 2.29.0 with my setup:
- Ubuntu 18.04.3 LTS with kernel version 5.0.0-36-generic
- Nvidia Quadro M2200, driver 390.116
- T265 firmware version 0.1.0.279
Below is my dmesg log:
```
[ 4897.830675] usb 1-3: new high-speed USB device number 9 using xhci_hcd
[ 4897.979681] usb 1-3: New USB device found, idVendor=03e7, idProduct=2150, bcdDevice= 0.01
[ 4897.979686] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4897.979689] usb 1-3: Product: Movidius MA2X5X
[ 4897.979691] usb 1-3: Manufacturer: Movidius Ltd.
[ 4897.979694] usb 1-3: SerialNumber: 03e72150
[ 4906.248899] usb 1-3: USB disconnect, device number 9
[ 4906.806596] usb 2-4: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[ 4906.827625] usb 2-4: New USB device found, idVendor=8087, idProduct=0b37, bcdDevice=ff.ff
[ 4906.827629] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4906.827632] usb 2-4: Product: Intel(R) RealSense(TM) Tracking Camera T265
[ 4906.827635] usb 2-4: Manufacturer: Intel(R) Corporation
[ 4906.827638] usb 2-4: SerialNumber: 9092121114170000
[ 4907.369269] uvcvideo: Failed to query (SET_CUR) UVC control 10 on unit 2: -32 (exp. 2).
```
When connecting the T265, system only recognise the Movidius, after open realsense-viewer or launch rs_t265.launch, system recognise the Tracking camera. Btw, ROS wapper works fine, but only the realsense-viewer freezes

tpwrules, 2020-01-14:
This issue is still present with SDK 2.31.0, firmware 0.2.0.879, kernel 4.15.0-74-generic on 18.04.3 LTS, and NVidia driver 430.64 on a Quadro P2000.