realsenseai/librealsense #4038: Where is D435i firmware version 5.11.6.200?

ryan4729, 2019-05-21:
librealsense 2.21.0 release notes recommend 5.11.6.200 firmware for the D400 series. However, I can't find this on the firmware download site (https://downloadcenter.intel.com/download/28573/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t). The latest one on the site is 5.11.4

oursland, 2019-05-21:
I have a copy of it, but I don't think it's a good idea to install it.  See #3978.  FW 5.11.1.100 appears to be your best bet as the D435i will enumerate correctly and be recognized by librealsense.

ryan4729, 2019-05-21:
I'm actually using it and it seems to work fine, I'm just wondering why it disappeared from the site - maybe it's no longer recommended?

ev-mp, 2019-05-22:
@ryan4729 , @oursland -
The fw 5.11.6.200 was temporally removed. I believe it will be restored soon.
In any case we do not recommend to downgrade from 5.11.6 to previous versions.

zjcs, 2019-05-24:
Downgrade is dangerous, https://github.com/IntelRealSense/librealsense/issues/3857

RealSenseCustomerSupport, 2019-05-24:
----------------------------------------------

Hi All,

Please see the comments from ev-mp.
And please try not change back and forth between FW versions for now, as it is pointed out that someone had issues of doing so.

Thanks!

closed 2019-06-10 by RealSenseCustomerSupport