realsenseai/librealsense #4039: Dependency on linux-headers-generic

YoshuaNava, 2019-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.10.3-5.11.6.200 | 
| Operating System & Version | Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    | 4.15 lowlatency | 
| Platform                                 | PC |
| SDK Version                            | Most recent | 
| Language                            | C++ | 
| Segment			|  Robot  | 

### Issue Description
By inspecting the dependencies listed in the librealsense package.xml file [see [link](https://github.com/IntelRealSense/librealsense/blob/master/package.xml#L32)], I have found that it lists `linux-headers-generic`. This is a valid rosdep key [[source](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3840)], which means that under normal circumstances it should be installed by the rosdep tool.

Now, my questions:

* Why is this package listed as a dependency? I think it reduces compatibility with systems that have a different type of kernel (realtime, lowlatency, etc), and can inadvertedly trigger the installation of unexpected depends.

* Would it be possible to remove it? If affirmative, I could submit a PR as soon as possible.

Thank you.

YoshuaNava, 2019-05-24:
Are the Linux generic packages actually used during build?

doronhi, 2019-07-10:
@YoshuaNava ,
Your notes sound valid and it could be that this depend is an overkill and can be replaced.
`package.xml` is part of the ROS installation and librealsense package is still not available as a ROS debian package.
That in mind, it would help to validate your notes if you could please describe the operations that triggered reading the file `package.xml`?

YoshuaNava, 2019-07-10:
Hi doronhi,
I'm currently generating Debian packages of librealsense and realsense2_camera. While running  rosdep to install all the pkg depends, prior to generating debians, I observed that it tried to install Linux image generic.

I tried removing it locally, and it didn't seem to cause any issues.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@YoshuaNava Any other questions about this? Looking forward to your update. Thanks!

YoshuaNava, 2019-12-04:
No more questions at the moment.

Thank you.

closed 2019-12-10 by dorodnic