realsenseai/librealsense #4040: Manually set position and force re-localisation

marekskocz, 2019-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5502 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | Intel NUC  |
| SDK Version                            |  2.21.0                        | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Feature Request
It would be great if there was a manual "set position" functionality in the API, which would allow the user to tell the T265 where it is (approximately) with respect to an imported map. Ideally, this would be combined with the re-localisation functionality (user sends approximate pose to T265, T265 tries to re-localize and sends confirmation to user as soon as re-localisation is done).

Is there any chance we might see such a feature in the upcoming releases?

RealSenseCustomerSupport, 2019-05-31:
----------------------------------------------

Hi @marekskocz

Thank you for the information regarding your feature request.  In order to make sure we either have what you need already or take into consideration what you need, I'm hoping you can give me some further information around the usage model you are trying to accomplish with a feature like this.  Can you provide some further details on what you would use this for?

We currently have the ability to load and save maps.  That's already there in our librealsense library.  We also have the ability to set_static_node and get_static_node that may be able to cover the features that you are requesting.

radfordi, 2019-05-31:
@marekskocz, you can define a coordinate frame with `set_static_node` before you save a map in one session.  After you load that map into a new session you can query location of that coordinate frame in the current session's coordinates with `get_static_node`.   As long as the system knows where the frame you defined is, we will continue to return an updated estimate of that frame's location every time you call `get_static_node`.

marekskocz, 2019-06-03:
Thank you for your reply. I am aware of the set_static_node and get_static node functionality, but this is not enough for our purposes.

Having a manual set position / re-localisation has two goals:

- using the camera in a large area (larger than the size of the internal map), by storing multiple maps and importing them as the camera roams from one area to another. After switching between maps, we could right away call a re-localisation, as we know the exact location of the camera (instead of having to wait until it re-localises by itself, without having an initial estimate of its pose in the map).

- Allowing quicker re-localisation after power-up in an existing environment. If the system was shut down at a certain location, and we know the camera hasn't moved, we could give the camera an estimate of its location without having to wait until it re-localises by itself.

radfordi, 2019-06-04:
Thanks for the suggestion @marekskocz.  I agree with both of your points.  If the pose is accurate it would accelerate relocalization.

marekskocz, 2019-06-05:
@radfordi Any chance we might see such functionality in future releases? Or any tips how we could implement this ourselves?

giovannicairoli, 2019-09-05:
Hello,

@RealSenseCustomerSupport Is there a sample or something similar about load, save maps and relocate where someone not expert can start?

Thank you

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @giovannicairoli,

You can first refer to below example in unit test which show the sequence concept to have export/import map and set/get static nodes. One full sample app is also planned.
https://github.com/IntelRealSense/librealsense/blob/b54f47ba78307be1f337f0b62dcbb4ee1613df34/unit-tests/unit-tests-live.cpp#L5450

RealSenseCustomerSupport, 2020-01-28:
----------------------------------------------

There have been some updates to librealsense with regards to import/export maps and relocalization.  The latest version of librealsense now being 2.32.1...also an sample from this PR giving example of importing/exporting maps as well as get/set static nodes - https://github.com/IntelRealSense/librealsense/pull/5356

@marekskocz - have a look at the latest and updates.  Let us know if we can close this thread out.

closed 2020-04-30 by RealSenseSupport