realsenseai/librealsense #4041: D435 come out of the box with Right IR stream disabled  and why

scizors, 2019-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (latest) | 
| Operating System & Version |   {Win (10) 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python and viewer }                          | 
| Segment			|  others                         | 

### Issue Description
 D435 come out of the box with Right IR stream disabled  and why 
How to acivate it in python

ev-mp, 2019-05-22:
@scizors hello, 
The Right IR stream is enabled iff device is enumerated as USB3 and the `Advanced Mode` is set on.
You can use Realsense viewer to verify both:
[image]

scizors, 2019-05-22:
How to activate advance mode in python and what is that means

ev-mp, 2019-05-22:
@scizors, 
The `advanced mode` feature enables Intel-specific proprietary APIs to fine-tune the device performance.
This [white paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-
technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf) will present some cases achievable with this API.

The feature is persistent between device power ups - so it should be set only once either in `realsense-viewer` or any other tool.
For technicalities please refer to [python documentation](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples#list-of-examples) - `RS400 Advanced Mode` example.

scizors, 2019-05-22:
thanks a lot foe the quick reply

closed 2019-05-22 by scizors