realsenseai/librealsense #4042: Infrared stream missing depending on system

nicovanduijn, 2019-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.13.00 | 
| Operating System & Version |  Linux Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-50-generic vs.  4.9.56-intel-pk-standard    | 
| Platform                                 | PC vs. UP Core  |
| SDK Version                            |  2.21.0                          | 
| Segment			|  Robot                          | 

### Issue Description
Depending on the system I plug my realsense into, different streams are available, despite using the same firmware, same SDK, even the same USB cable (2.1). 

How can this be? Is this related to the kernel?

The output of `rs-enumerate-devices` on my laptop:
[image]

While on the UP Core, the infrared streams are missing: 
[image]

I really need the IR stream, and I cannot use USB 3.1...  Any ideas?

ev-mp, 2019-05-22:
@nicovanduijn hello, in USB2 mode the D400 cameras provide a subset of functionalities due to bandwidth and power constrains. This includes limiting IR streaming to left imager only.
This limit is set by design and not related to kernel/OS.

nicovanduijn, 2019-05-22:
@ev-mp I'm aware of this. But my problem is that both setups use USB2, one of them has the infra stream 1, the other has no infra stream at all.

I'd be happy having one

ev-mp, 2019-05-22:
@nicovanduijn - I'm sorry I misinterpret the post.
Can you share the log for `lsusb -d 8086:0b07 -v` from the Up Core machine ?

nicovanduijn, 2019-05-22:
Sure thing:
```
Bus 001 Device 006: ID 8086:0b07 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x8086 Intel Corp.
  idProduct          0x0b07 
  bcdDevice           50.9d
  iManufacturer           1 Intel(R) RealSense(TM) 435
  iProduct                2 Intel(R) RealSense(TM) 435
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength         1115
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              110mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Intel(R) RealSense(TM) 435 with RGB Module Depth
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              5 Intel(R) RealSense(TM) 435 with RGB Module Depth
      VideoControl Interface Descriptor:
        bLength                14
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength          119
        dwClockFrequency       48.000000MHz
        bInCollection           2
        baInterfaceNr( 0)       1
        baInterfaceNr( 1)       2
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
        bmControls           0x00200000
[... 632 lines cut ...]
        bLength                             6
        bDescriptorType                    36
        bDescriptorSubtype                 13 (COLORFORMAT)
        bColorPrimaries                     0 (Unspecified)
        bTransferCharacteristics            0 (Unspecified)
        bMatrixCoefficients                 0 (Unspecified)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           50
  bNumDeviceCaps          3
FIXME: alloc bigger buffer for device capability descriptors
Device Status:     0x0001
  Self Powered
```

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@nicovanduijn Do you still have issue also if connect to USB3 port on UP Core? Could you please try 4.15 kernel on UP Core?

nicovanduijn, 2019-06-24:
As mentioned in my initial post, USB3 is not an option for our use-case (due to GPS interference).

Using a different kernel would be one option, but this really is just the tip of the ice berg of our problems. The main issue is that we it doesn't work well in bright light due to the exposure issues (#4140 #4022 ), which we're not sure if they are broken in 5.09.13 or just invisible because I can't see the IR image.

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@nicovanduijn According to #4140, the exposure issue will be fixed in firmware 5.11.12.0. Will update you when this firmware ready.
Considering only kernel version is different between your PC and UP Core, please have a try with kernel 4.15 on UP Core if possible. Thanks!

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@nicovanduijn Any luck if upgrade kernel to 4.15 on UP Core?

nicovanduijn, 2019-07-02:
Due to the container system we use, I cannot update the kernel for now

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@nicovanduijn Considering FW v5.9.13 is a little too old version, could you please try the latest FW v5.11.6.250 to see if you still get this IR stream missing issue on UP Core?

FW v5.11.6.250 download link: https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@nicovanduijn Did you get chance to try FW v5.11.6.250? Looking forward to your update. Thanks!

nicovanduijn, 2019-07-12:
I did not observe the issue with other firmwares so far

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@nicovanduijn Glad to hear that. Will close this issue accordingly. Thanks!

closed 2019-07-12 by RealSenseCustomerSupport