realsenseai/librealsense #4043: Unable to load .json config into device (C#)

xen20, 2019-05-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.04.00  | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.22.0.856                           | 
| Language                            |  C#                           | 
| Segment			|  others                          | 

### Issue Description
Hello.
I've been trying to load a .json settings file with C# using the following code (I have looked @ issue #3275 , and used the same approach, pretty much): 
```
        static void Main(string[] args)
        {
            Context ctx = new Context();
            Console.WriteLine(ctx.Version);
            DeviceList devices = ctx.QueryDevices();

            if (devices.Count == 0)
                Console.WriteLine("No device detected ");

            Pipeline pipe = new Pipeline();

            Device camera = devices[0];
            Config config = new Config();
            config.EnableStream(Stream.Depth, 848, 480, Format.Z16, 60);
            config.EnableStream(Stream.Color, Format.Rgb8);

            PipelineProfile profile = pipe.Start(config);
            var loadJSON = AdvancedDevice.FromDevice(profile.Device);
            //loadJSON.AdvancedModeEnabled = true; //crashes the program
            string jsonContents = System.IO.File.ReadAllText("settings.json");
            loadJSON.JsonConfiguration = jsonContents.Replace("\r\n", "\n");
         }
```
However, when trying to read the configuration back by simply getting something like:
`string currentConfig = loadJSON.JsonConfiguration` , I do not get the values read from the file. What am I missing?
Thanks.

xen20, 2019-05-29:
Issue still persists.

xiaoyanluo, 2019-06-05:
@xen20 same as me

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@xen20 @xiaoyanluo  Could you please provide the difference details? The .json file you load and read will be helpful.

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@xen20 @xiaoyanluo Could you please update if you still need support on this?

xiaoyanluo, 2019-07-08:
Thank you! All the best.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@xen20 How about your side? Do you still have questions for this ticket?

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@xen20 If no other questions, will close it accordingly. Thanks!

closed 2019-07-15 by RealSenseCustomerSupport