realsenseai/librealsense #4044: rs2_transform_point_to_point

scizors, 2019-05-22:
D435 

Transform 3D coordinates relative to one sensor to 3D coordinates relative to another viewpoint */
python
color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_point)
means
one sensor what it means from 3D depth point to 3D color point

is it possible to do other way around: 3D color point to 3D depth point
if yes how

lramati, 2019-06-13:
As mentioned in the documentation, the function can transform from any viewpoint to any other viewpoint using the extrinsics between them. In your case, you'd want colot_to_depth_extrin, which can be obtained much the same way as the depth_to_color_extrin

scizors, 2019-06-13:
but align is doing the same thing. I need to know the difference between align frame and  depth to color extrin

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@scizors Align needs rs_transform_point_to_point but also need rs2_deproject_pixel_to_point and rs2_project_point_to_pixel
https://github.com/IntelRealSense/librealsense/blob/v2.24.0/src/proc/align.cpp#L35

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@scizors Any other questions about this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

@scizors If no other questions will close it accordingly. Thanks!

closed 2019-07-09 by RealSenseCustomerSupport