realsenseai/librealsense #4046: Advanced_mode in new version

CChape, 2019-05-22:
Hi I was following old examples for loading a json congifuration file. 
But it's seems, that some examples does not work anymore in new version.
I got this error:
"rs400_advanced_mode/rs400_advanced_mode.hpp:No file or folder of this type"

Is there a new way to load a json file for camera configuration I'm using a D435. 
Thanks in advance.

ev-mp, 2019-05-22:
@CChape hello,
Loading json presets via c++ is available via:
```c++
            std::string json_filename(TBD)
            std::ifstream file(json_filename);
            if (file.good())
            {
                std::string str((std::istreambuf_iterator<char>(file)), std::istreambuf_iterator<char>());
                if (auto advanced = dev.as<advanced_mode>())
                {
                    advanced.load_json(str);
                    ....
```
#3710

CChape, 2019-05-22:
Hi @ev-mp 
Thanks for your quick answer. I tried to add your code, but "advanced_mode" was not declared in this scope
And I add the "#include <librealsense2/rs_advanced_mode.hpp>". 
Do I need to add a namespace or something?
Thanks!

ev-mp, 2019-05-22:
@CChape, correct -
The `advanced mode` is isolated into  `rs400` namespace, mostly due to legacy reasons.
So you need to add these two lines:
```c++
#include <librealsense2/rs_advanced_mode.hpp>
...
using namespace rs400;
...
```

CChape, 2019-05-22:
@ev-mp All good!
Thanks for your fast and clear answers!

closed 2019-05-22 by CChape