realsenseai/librealsense #4048: context.query_devices() confuses T265

tpwrules, 2019-05-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |  Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    | 4.15.0-50-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.22.0 (master) }                          | 
| Language                            |  Python 3.7   | 

### Issue Description

The following program displays the pose from an attached T265:
```
import pyrealsense2 as rs
ctx = rs.context()
print(list(ctx.query_devices()))
pipe = rs.pipeline()
pipe.start()
print(pipe.wait_for_frames().first_or_default(rs.stream.pose).as_pose_frame().get_pose_data().translation)
pipe.stop()
```

On the listed setup, it produces the following output when a T265 is plugged in:
```
$ python3 query_devices_problem.py
[<pyrealsense2.device: Intel RealSense T265 (S/N: <redacted>)>]
Traceback (most recent call last):
  File "query_devices_problem.py", line 5, in <module>
    pipe.start()
RuntimeError: No device connected
```

When line 2 is commented out, it produces correct output:
```
$ python3 query_devices_problem.py
x: -3.32125e-06, y: 0.000118504, z: -2.13213e-05
```

It seems that querying for devices makes the T265 inaccessible to the pipe system. This problem does not occur with a D435 (it instead crashes because the D435 does not offer pose data).

ev-mp, 2019-05-22:
@tpwrules hello,
When working with T265 you have to share and reuse the initial `context` across all the `pipe` objects created. This limitation does not apply to other devices.
>import pyrealsense2 as rs
**_ctx_**= rs.context()
print(list(ctx.query_devices()))
pipe = rs.pipeline(**_ctx_**)
pipe.start()
...

See #3734, #3697

tpwrules, 2019-05-22:
Oh, I did not realize that I had to pass the context to the pipeline. That indeed fixes the issue.

closed 2019-05-22 by tpwrules