realsenseai/librealsense #4050: Misalignment when using the D435i

jonblob, 2019-05-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version | Win 10 |                              | 
| Platform                                 | PC |
| SDK Version                            | 2.22.0 | 
| Language                            | Python 3 | 
| Segment			| Robot | 

### Issue Description
Alignment of depth and color frame returns misaligned images (both depth to color and color to depth).
I tested aligning the images using pyrealsense2.align as well as the rs-align tool, both yielding similar results where the aligned depth image seems to be shifted to the left with respect to the color image.

[image]
Figure 1: Color frame and corresponding aligned depth frame using pyrealsense2.align

[image]
Figure 2: Results of rs-align.

Source code:

```
import numpy as np
import cv2
import pyrealsense2 as rs

#Camera streams
res_color = (848,480)
res_depth = (848,480)
fps = 15

#Colorizer for the depth frame
colorizer = rs.colorizer()

#Initialize D435
pipe_D435 = rs.pipeline()
config_D435 = rs.config()
config_D435.enable_stream(rs.stream.depth, res_color[0], res_color[1], rs.format.z16, fps)
config_D435.enable_stream(rs.stream.color, res_depth[0], res_depth[1], rs.format.bgr8, fps)
align_to = rs.stream.color
align = rs.align(align_to)
       
#Start pipeline
prof_D435 = pipe_D435.start(config_D435)

while True:
    frames_D435 = pipe_D435.wait_for_frames()
    
    #alignment
    frames_D435 = align.process(frames_D435)
    
    #Get color and depth frames for display
    color_frame = frames_D435.get_color_frame()
    depth_frame = frames_D435.get_depth_frame()
    
    color_frame = np.asanyarray(color_frame.get_data())
    depth_frame = np.asanyarray(colorizer.colorize(depth_frame).get_data())
    
    camera_images = np.vstack((color_frame, depth_frame))
    
    #Display images
    cv2.imshow("RealSense", camera_images)            
    key = cv2.waitKey(1)
    if key & 0xFF == ord('q'): 
        cv2.destroyWindow("RealSense")
        pipe_D435.stop()  
        break
 ```

ev-mp, 2019-05-23:
@jonblob hello,
What is the log of `rs-enumerate-devices -c` ?

jonblob, 2019-05-23:
The output of `rs-enumerate-devices -c` is
```
Device info:
    Name                          :     Intel RealSense D435I
    Serial Number                 :     846112070024
    Firmware Version              :     05.11.06.200
255.255.255.255
    Recommended Firmware Version  :     05.10.03.00
    Physical Port                 :     \\?\usb#vid_8086&pid_0b3a&mi_00#7&2d610992&1&0000#{24e552d7-6523-47f7-a647-d3465bf1f5ca}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Usb Type Descriptor           :     3.2

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format
    Infrared 2    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 15Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y16
    Infrared 2    1280x800      @ 15Hz     Y8
    Infrared 1    1280x800      @ 15Hz     Y8
    Infrared 1    1280x720      @ 30Hz     Y8
    Infrared 2    1280x720      @ 30Hz     Y8
    Infrared 2    1280x720      @ 15Hz     Y8
    Infrared 1    1280x720      @ 15Hz     Y8
    Infrared 2    1280x720      @ 6Hz      Y8
    Infrared 1    1280x720      @ 6Hz      Y8
    Infrared 1    848x480       @ 90Hz     Y8
    Infrared 2    848x480       @ 90Hz     Y8
    Infrared 1    848x480       @ 60Hz     Y8
    Infrared 2    848x480       @ 60Hz     Y8
    Infrared 1    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 30Hz     Y8
    Infrared 1    848x480       @ 15Hz     Y8
    Infrared 2    848x480       @ 15Hz     Y8
    Infrared 2    848x480       @ 6Hz      Y8
    Infrared 1    848x480       @ 6Hz      Y8
    Infrared 2    848x100       @ 100Hz    Y8
    Infrared 1    848x100       @ 100Hz    Y8
    Infrared 1    640x480       @ 90Hz     Y8
    Infrared 2    640x480       @ 90Hz     Y8
    Infrared 2    640x480       @ 60Hz     Y8
    Infrared 1    640x480       @ 60Hz     Y8
    Infrared 2    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 30Hz     Y8
    Infrared 2    640x480       @ 15Hz     Y8
    Infrared 1    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 6Hz      Y8
    Infrared 1    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 25Hz     Y16
    Infrared 1    640x400       @ 15Hz     Y16
    Infrared 2    640x400       @ 15Hz     Y16
    Infrared 2    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 60Hz     Y8
    Infrared 1    640x360       @ 30Hz     Y8
    Infrared 2    640x360       @ 30Hz     Y8
    Infrared 2    640x360       @ 15Hz     Y8
    Infrared 1    640x360       @ 15Hz     Y8
    Infrared 1    640x360       @ 6Hz      Y8
    Infrared 2    640x360       @ 6Hz      Y8
    Infrared 2    480x270       @ 90Hz     Y8
    Infrared 1    480x270       @ 90Hz     Y8
    Infrared 2    480x270       @ 60Hz     Y8
[... 798 lines cut ...]
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: -0.0444818884134293  -0.00510000018402934  -0.011739999987185

Extrinsic from "Accel"    To      "Gyro" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0

Extrinsic from "Accel"    To      "Accel" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0  0  0
```

Seems like the extrinsics between color and all other sensors are broken, however I did not receive any warnings/errors when using align.

jonblob, 2019-05-23:
I just tested my code and `rs-align` using a colleague's D435, which returned properly aligned images. For that camera, the extrinsics between color and all other sensors as shown in `rs-enumerate-devices -c` also seem to be fine.

ev-mp, 2019-05-23:
@jonblob hello, unfortunately this is the same issue as in #3788, #3949
>Extrinsic from "Depth"    To      "Color" :
N/A

In v2.22 we release a workaround that apparently worked only for some users. We'll return to you on this.

jonblob, 2019-05-23:
I am using v2.22, so apparently the workaround does not work in my case. Thank you for your fast response though! I hope this issue can be resolved soon.

ev-mp, 2019-05-23:
@jonblob hello,

Please create and share the camera dump generated with the attached python script 
[realsense_flash_debug.txt](https://github.com/IntelRealSense/librealsense/files/3211870/realsense_flash_debug.txt)

Instructions:
1. Install python 2.7/3.6
2. Install realsense python wrapper
`pip install pyrealsense2`
3. Rename the attached script extension: `realsense_flash_debug.txt` into `realsense_flash_debug.py`
4. Run the script
`python realsense_flash_debug.py`
The script will run for ~2min and generate device FW dump `realsense_debug_dump.bin`
4. Attach the resulted file

jonblob, 2019-05-24:
Hello, I ran the script as described and attached the FW dump. I renamed the .bin to .txt to be able to attach it.
[realsense_debug_dump.txt](https://github.com/IntelRealSense/librealsense/files/3214577/realsense_debug_dump.txt)

jonblob, 2019-06-03:
Can you please tell me how long it will take until an update is released? Without being able to align color and depth images, I can not use the camera.

zhuzunjie17, 2019-06-03:
Exactly the same issue

xiaoyanluo, 2019-06-04:
> Exactly the same issue
我也是很多问题，你呢？求交流啊 WX Ai1616166

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@jonblob The latest librealsense v2.23.0 and FW 5.11.6.250 should fix the issue. Please have a try. Thanks!

FW link: https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip

jonblob, 2019-06-13:
Thank you for the update! I installed the firmware version 5.11.6.250 and built the new librealsense v2.23.0.
Unfortunately, the issue is still not fixed. Both using rs-align and using my scripts with pyrealsense, the output of align is a black image (both for depth to color and color to depth). When executing rs-enumerate-devices or checking the extrinsics via pyrealsense, the extrinsics of all other streams to the color stream are still zeros, exactly the same as in my previous upload.

Are there any other things to try? Otherwise I would ask to return the camera or exchange it for a D435, since I can not proceed with my research without being able to align depth and color images.

RealSenseCustomerSupport, 2019-06-14:
----------------------------------------------

@jonblob Really sorry for the trouble! I'll further check what you can check in this situation and will update you early next week.

jonblob, 2019-06-14:
Thank you very much!

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@jonblob

Please install Calibration API and try to restore golden calibration table by "Intel.Realsense.CustomRW.exe -g"
to see if the issue can fix or not. If still can't fix, please exchange the camera with the distributor. Sorry for the trouble!

 Calibration API download link: https://downloadmirror.intel.com/28517/eng/CalibrationAPI-2.6.8.0-Setup.zip
User guide: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

jonblob, 2019-06-18:
Thank you for your help. I ran "Intel.RealsenseCustomRW.exe -g", resulting in

```
CustomRW for Intel RealSense D400, Version: 2.6.8.0

  Device PID: 0B3A
  Device name: Intel RealSense D435I
  Serial number: ************
  Firmware version: 05.11.06.250

Calibration on device successfully reset to default gold factory settings.
```

Unfortunately, rs-enumerate-devices still shows "N/A" for the extrinsics between color and any other stream. rs-align and the align function in pyrealsense2 also do not work. If possible, I would like to exchange the camera for a D435 with correct extrinsic calibration so I can continue properly. The issue can then be closed.

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

@jonblob Sorry to hear that the issue not fixed. Yes, please exchange the camera.

closed 2019-06-19 by RealSenseCustomerSupport