realsenseai/librealsense #4051: TrackingManager: Error while loading image - libusb_open failed (0xFFFFFFFD), will try again later

jdwolk, 2019-05-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.4.0 | 
| Operating System & Version |  Ubuntu Mate  | 
| Kernel Version (Linux Only)    |  4.14.5-92                                        | 
| Platform                                 | Odroid XU4  |
| SDK Version                            | 2.22.0                        | 
| Language                            |  C                        | 
| Segment			| Robotics                        | 

### Issue Description
Hi,

Today I upgraded the firmware on my D435 so I could experiment (separately) with the T265 in the latest version of `librealsense`. My application is robotics so most of my experimenting is happening via `realsense-ros`.

After upgrading firmware, whenever I run any of the experiment commands (i.e. `rs-depth`, etc) or even `rs-enumerate-devices` I get `TrackingManager: Error while loading image - libusb_open failed (0xFFFFFFFD), will try again later` and it never corrects.

Notably, when run via `realsense-ros`, I have `initial_reset` set to true; after the reset occurs the camera seems to work ok. It's just all these other applications where it doesn't seem to work.

jdwolk, 2019-05-23:
Actually, I just noticed this only occurs when I have both my T265 and D435 plugged in at the same time

ev-mp, 2019-05-23:
@jdwolk hello, afaik there was a fix  for that in the librealsense ros wrapper version released yesterday.
Can you try that version and update?

jdwolk, 2019-05-23:
If you're referring to [2.2.4](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.4), I'm already using that like I mentioned above. Tried re-building just to make sure and it definitely still happens.

Also like I mentioned, it's not ros-specific since the error comes up when I run any of the `rs-` example commands, i.e. even `rs-enumerate-devices`. I'm now realizing that this happens _whenever_ the T265 is involved - right now I just tried with only the T265 and it still happens for example.

jdwolk, 2019-05-23:
Also noticed a few more things. Everything works great on my laptop, it's only when I try on the odroid XU4 that I get errors.

Here's `lsusb` output on the Odroid (on laptop it shows up as an Intel device):

```
CAMERA:~ % lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 03e7:2150  
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 016: ID feed:1203  
Bus 001 Device 015: ID 04f2:0939 Chicony Electronics Co., Ltd 
Bus 001 Device 014: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And `dmesg`:

```
[ 1689.913180] usb 3-1.1: new high-speed USB device number 6 using xhci-hcd
[ 1690.013675] usb 3-1.1: New USB device found, idVendor=03e7, idProduct=2150
[ 1690.013682] usb 3-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1690.013687] usb 3-1.1: Product: Movidius MA2X5X
[ 1690.013692] usb 3-1.1: Manufacturer: Movidius Ltd.
[ 1690.013697] usb 3-1.1: SerialNumber: 03e72150
```

jdwolk, 2019-05-23:
Ah, it was my mistake.

During the [manual build/install](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) I forgot to run the scripts to patch for my device and set up udev rules, i.e.:

```
./scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh
./scripts/setup_udev_rules.sh
```

After building w/ those it works like a charm on my Odroid XU4 :sparkles:

I'll close this

closed 2019-05-23 by jdwolk