realsenseai/librealsense #4052: passthrough for x, y points (SDK 2.0)

vusdo, 2019-05-23:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C }                          | 
| Segment		    	|                             | 

### Issue Description
I saw a sample that removes the background through depth.

I have not only depth through SDK2.0 ...
I want to remove x and y through the angle of view.

For example with PCL ..
pcl passthrough is to multiply.

What should I do with SDK 2.0?
How can I set the range?

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

Hi @vusdo,

Do you have your own solution? We don't have this feature so far. Probably, someone can contribute his idea.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @vusdo

This thread has been opened for quite some time.  If there is nothing else needed, then the thread should be closed.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi @vusdo

If there is nothing else needed, then the thread should be closed.
***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-02-21 by RealSenseCustomerSupport