realsenseai/librealsense #4053: Projector temperature in ROS Wrapper

Brolin56, 2019-05-23:
Hi,
As per the datasheet linked [here](https://www.mouser.com/datasheet/2/612/Intel-RealSense-D400-Series-Datasheet-1383890.pdf) in page 38, it is mentioned that the laser projector will shutoff at temperature >60 degree Celsius. I am using pointcloud data from the camera. So when the laser shuts down my node crashes. Can you please add a publisher that can publish the projector temperature so that I can use this data to avoid my node crash.

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@Brolin56  According to the datasheet, "When laser power and depth streaming is enabled and if
stereo depth module temperature is >60°, laser power is halved. If temperature is not lowered below temperature limit within a certain interval, the laser is shut off."   Did you get the laser shut down in your case now? Or you're just thinking of such possibility?

brolinA, 2019-06-21:
I am sure that the laser projector is being shutdown. If I could read the temperature in the ROS wrapper then I would be able to show a warning message telling that the laser is getting too hot.

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

@brolinA Got your point! Will feedback this request to our Engineering team. Any update will let you know. Thanks!

brolinA, 2019-06-27:
Thank you. Please do it ASAP

doronhi, 2019-08-06:
Added in [realsense2_camera, PR#878](https://github.com/IntelRealSense/realsense-ros/pull/878)

RealSenseCustomerSupport, 2019-08-20:
----------------------------------------------

@Brolin56 As Doronhi replied, the request now implemented in latest release https://github.com/IntelRealSense/realsense-ros/releases. Could you please check and try? Thanks!

closed 2019-08-26 by RealSenseCustomerSupport