realsenseai/librealsense #4054: librealsense2-dkms install error: linux-headers-4.15.18-041518-generic is not supported

ccfire, 2019-05-23:
My laptop env:
ubuntu18.04  linux-4.15.18-041518-generic.

After run:
sudo apt-get install librealsense2-dkms
Output error:
Building for 4.15.18-041518-generic 4.18.0-20-generic
Building initial module for 4.15.18-041518-generic
**ERROR (dkms apport): kernel package linux-headers-4.15.18-041518-generic is not supported**
Error! Bad return status for module build on kernel: 4.15.18-041518-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.

Does the dkms not support 4.15.18-041518 ?

ev-mp, 2019-05-23:
@ccfire hello,
The DKMS package is built tested and distributed for Ubuntu LTS kernels, such as 4.15.0-50.
The version  you're using is a mainline _4.**15.18-041518**-generic_ (note the difference in semantics vs _4.**15.0-**._)
You can install the Ubuntu LTS version of 4.15 with:
`sudo apt install linux-image-4.15.0-50-generic linux-headers-4.15.0-50 linux-headers-4.15.0-50-generic`
and then rerun the dkms package.

ccfire, 2019-05-23:
@ev-mp thanks for reply, I will try now

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@ccfire Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@ccfire Any update? Will close the ticket if no other questions. Thanks!

closed 2019-06-25 by RealSenseCustomerSupport