realsenseai/librealsense #4055: How to use color camera with emgucv in win7？

jinxzzz, 2019-05-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {win7}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C#}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Due to the needs of the project, I need to call the D435 RGB camera through c# programming, which requires EmguCV. I installed the win7 sdk driver according to the tutorial（https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md）, but I still can't use the RGB camera through the  basic programming statement{VideoCapture capture(0)}  in EmguCV. I checked the device manager but found no image device, the universal serial bus device has a D435 color and depth camera.I want to know how to call RGB camera with EmguCV.
In addition, why can I find the camera in the image device in the device manager under the win10 system, but in the universal serial bus device under the win7 system?

dorodnic, 2019-06-20:
Windows 7 UVC driver does not support 1.5 version of the UVC specification used by the camera, hence you cannot use D400 as "regular" web-cam in Windows 7. It's an OS limitation. This is why we designed and implemented a custom driver to enable Win7 users. What you are asking is not supported without using librealsense

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@jinxzzz Any other question for this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@jinxzzz If no other questions, will close the ticket at this point.

closed 2019-07-05 by RealSenseCustomerSupport