realsenseai/librealsense #4056: Bug in the script realsense-ubuntu-lts.sh

frankfalse, 2019-05-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System |   Ubuntu 16.04 xenial | 
| SDK Version                             2
| Language                            |  C/C++                          | 

### Issue Description
During the installation of librealsense SDK version 2.22 on Ubuntu 16.04 LTS, and in particular when we apply the patches to the kernel modules needed by the library, the script **patch-realsense-ubuntu-lts.sh failed** because it can't clone the git repository of the kernel modules.

The line with the error is the number 53:
git clone -b $kernel_branch git://kernel.ubuntu.com/ubuntu/ubuntu-${ubuntu_codename}.git --depth 1 ./${kernel_name}

I have found the new position of the repository and I've modified the script. So I have finished the installation.

Someone can correct the script please?

ev-mp, 2019-05-23:
@frankfalse hello, 
What is the kernel version you're runing ? (`uname -r`)
I've tried the script with Ubuntu 16.04.4 LTS kernels 4.4 and 4.15 and was not able to reproduce any repo-related error.
Can you post the errors you've encountered  and the changes you made to make it work?

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@frankfalse Could you please feedback your kernel version as ev-mp requested? Thanks!

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@frankfalse If no other questions, will close the ticket accordingly. Thanks!

closed 2019-07-08 by RealSenseCustomerSupport