realsenseai/librealsense #4057: Save into .bag file pyrealsense2

filipematosinov, 2019-05-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux (Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello, 
I am wondering how is the best way to save a stream (color + depth) into a  .bag using pyrealsense. I have seen several examples to read from a bag file but nothing to save. My particular application needs to save to streams (from two cameras) to two .bag files.

realsens01, 2019-05-29:
Hi @filipematosinov ,

If I understand your question correctly, then you can do that by setting up config objects and passing them to the start function of the pipeline object. There is some example code in issue https://github.com/IntelRealSense/librealsense/issues/3044. I am collecting data from two D435 cameras in this fashion. There are a number of issues to be aware of (such as the one that I linked).

filipematosinov, 2019-06-06:
Thank you @realsens01. I am now able to save them correctly. However, to read it how can I read it frame by frame (i.e not reading at the actual fps of the video) in order to preform some calculations to each frame which are slower than the actual fps.

realsens01, 2019-06-06:
Some time ago, I tried doing this with the Realsense playback feature, but I had better luck with the [rosbag python interface](http://wiki.ros.org/rosbag/Code%20API#py_api). Take a look at the rosbag wiki for [examples](http://wiki.ros.org/rosbag/Cookbook#Python). On Ubuntu, the relevant package for me was installed via ``apt install python3-rosbag``. I wrote a Python class that basically pulls all of the information out of the bag using rosbag. Here's some code that illustrates the general idea:

```
import rosbag
bag    = rosbag.Bag('realsense.bag')
topics = ['/device_0/sensor_1/Color_0/image/data']
for message_data in bag.read_messages():
     (topic, msg, t) = message_data
     # Do something with the frame data in msg here.
```

This pulls out the color image stream. Leave out ``topics`` to process every type of ROS message, or change ``topics`` to include different streams.

**However**, I see posts on the forum that seem to indicate that there's been relevant developments in the Realsense API. Others on the forum might be able to give you a better idea about what is the most appropriate approach, at present. Hopefully, someone more knowledgeable will chime in.

filipematosinov, 2019-06-07:
Thanks for the advice @realsens01  :). Let's see if someone comes up with a high level realsense API approach. In the meantime I will have a look at tje python ROSBAG package, I was trying to avoid using ROS but maybe is better not to be scared!!

filipematosinov, 2019-06-07:
Moreover is it possible to save multiple cameras in the same bag? From your example it seems you can have several devices but using the pyrealsense API we have to call config (config.enable_record_to_file(...) ) which is created for each camera if I am correct

realsens01, 2019-06-18:
My experience with the rosbag API has been far better than librealsense API. Admittedly, my experience is somewhat limited in both cases, but everything I tried worked as the documentation said it would. On the other hand, I've had problems with Realsense for months, and the documentation / support haven't been much help.

I do not believe that it is possible to save multiple cameras in the same bag with the Realsense API, but I am not certain. You are correct: I initialize and configure each of my two cameras separately, and they record to a pair of bag files.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@filipematosinov Did you get it through now? Any other questions? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@filipematosinov Just as realsens01 mentioned, currently saving multiple cameras in the same bag not supported. Any other questions about this ticket?

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@filipematosinov If no other questions about it, will close it accordingly. Thanks!

closed 2019-07-22 by RealSenseCustomerSupport