realsenseai/librealsense #4058: t 265  realsense-viewer  not show imu&&img .only show Trajectory

hitxuhao, 2019-05-23:
| | |
|---|---|
|**librealsense**|2.21.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense T265|
|**Serial Number**|908412110462|
|**Firmware Version**|0.0.18.5502|

Please provide a description of the problem

ev-mp, 2019-05-23:
@hitxuhao hi,
Please check whether  this reproducible in v2.22.  
Sharing `realsense-viewer` and `rs-capture` screenshots will help as well.

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

Hi @hitxuhao,

Any update? Did you also try the latest librealsense?

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @hitxuhao,

Is the issue clarified/resolved? If no addition support for this issue, will close it.

closed 2019-12-05 by RealSenseCustomerSupport