realsenseai/librealsense #4059: All Xcode examples do not work. Does anybody have a guideline from start to finish?

omarfakhr, 2019-05-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@omarfakhr So you have referred to https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md and still has problem?  Any error prompted when you get the issue?

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@omarfakhr Any update from your side? Looking forward to your reply. Thanks!

omarfakhr, 2019-06-25:
I am on vacation right now. Can I check it when I come back?

Thanks a lot
Omar Fakhr

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@omarfakhr No problem. Enjoy your vacation! Let's check it later.

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@omarfakh Did you get chance to try now? Thanks!

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@omarfakh Sorry that we didn't get feedback from you several weeks. Will close this at this point. Please feel free to open new ticket if you still have issue or questions. Thanks!

closed 2019-07-22 by RealSenseCustomerSupport