realsenseai/librealsense #4063: Point Cloud

nvoltron, 2019-05-23:
| | |
|---|---|
|**librealsense**|2.21.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|831612073758|
|**Firmware Version**|05.11.04.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Please provide a description of the problem
I can see the point could image from Realsense viewer. 
I can see only streo and RGB image in 2D view. 
How can I see point cloud image from Realsense viewer? 
Best regards, 

[image]

ev-mp, 2019-05-26:
@nvoltron hello, the pointcloud is rendered when the viewer's "3D" view is turned on in the upper right corner. (try the "2D | 3D" toggle)

ev-mp, 2019-06-10:
@nvoltron, did you manage to get point cloud view?
Please update

nvoltron, 2019-06-10:
Yes, I can see the point cloud view toggle 3D and deselect "Render Quad". 
Thanks,

ev-mp, 2019-06-10:
Thanks for confirmation. Closing now.

closed 2019-06-10 by ev-mp