realsenseai/librealsense #4064: some issue with T_265_stereo

Drawodash, 2019-05-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 / T200 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
HI, i try to lunch T-265_stereo https://github.com/IntelRealSense/librealsense/commit/805f1536e7cdf5488d734716db2a7bc7424534af

but when i lunch it i have a issue like 

python3 t265_stereo.py
Traceback (most recent call last):
File "t265_stereo.py", line 113, in 
pipe.start(cfg, callback)
RuntimeError: No device connected

even if the realsense viewer saw all right.

so if someone hnow why?

ev-mp, 2019-05-26:
@Drawodash hello,
Building Librealsense2 with T265 support requires internet connection. 
In case you build the `pyrealsense` module manually, there is a possibility that T265 code was removed by CMake configuration due network-related issues
if that's the case - please rebuild the wrapper with CMake, and check CMake logs for TM2 notifications

RealSenseCustomerSupport, 2019-07-10:
----------------------------------------------

Hi @Drawodash,

Is the issue clarified? which OS version did you use? Did you follow the instruction in 't265_stereo.py'?

    **$ apt-get install python3-venv # install python3 built in venv support**
    **$ python3 -m venv py3librs # create a virtual environment in pylibrs**
    **$ source py3librs/bin/activate # activate the venv, do this from every terminal**
    **$ pip install opencv-python # install opencv 4.1 in the venv**
    **$ pip install pyrealsense2 # install librealsense python bindings**

Drawodash, 2019-07-10:
> Hi @Drawodash,
>
> Is the issue clarified? which OS version did you use? Did you follow the instruction in 't265_stereo.py'?
>
> ```
> **$ apt-get install python3-venv # install python3 built in venv [...]

hi,

I was using a Ubuntu console from a Windows10 OS. then i tried from a jupyter notebook and it worked correctly. so i think my problem was that the linux terminal didn't recognize the camera.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

Hi @Drawodash,

It's good it's working finally. I guess Ubuntu in Windows 10 is not well integrated. Previously, I tried parallels to install a different OS on MacBook and also faced the USB camera not detected issue.
T265 Ueeds to be recognized in two stage. 1) MA2X5X USB device recognized, 2) After FW loaded, T265 USB device recognized

closed 2019-07-24 by RealSenseCustomerSupport