realsenseai/librealsense #4068: Problem to run D435 on Fedora 27

karimbernardet, 2019-05-24:
Hi
My config : 

Camera Model                 D435
 Linux Fedora 27
 Kernel Version     4.18.19-100                                
Platform                                 PC
          
I am trying to have the camera working on my laptop. I did that : 
[root@10-83-102-17 karim]# ./patch-uvcvideo-fc23.sh
Intall Fedora kernel development tools
Échec de la synchronisation du cache pour le dépôt « larsks-apm_planner », désactivation.
Échec de la synchronisation du cache pour le dépôt « bintray--resin-io-redhat », désactivation.
Échec de la synchronisation du cache pour le dépôt « isv_ownCloud_desktop », désactivation.
Dernière vérification de l’expiration des métadonnées effectuée il y a 1:04:48 le ven. 24 mai 2019 10:05:32 CEST.
Le paquet fedora-packager-0.6.0.1-3.fc27.noarch est déjà installé, ignorer
Le paquet kernel-devel-4.18.13-100.fc27.x86_64 est déjà installé, ignorer
Le paquet kernel-devel-4.18.18-100.fc27.x86_64 est déjà installé, ignorer
Le paquet kernel-devel-4.18.19-100.fc27.x86_64 est déjà installé, ignorer
Dépendances résolues.
Rien à faire.
Terminé !
Checking required packages ...
Échec de la synchronisation du cache pour le dépôt « larsks-apm_planner », désactivation.
Échec de la synchronisation du cache pour le dépôt « bintray--resin-io-redhat », désactivation.
Échec de la synchronisation du cache pour le dépôt « isv_ownCloud_desktop », désactivation.
Dernière vérification de l’expiration des métadonnées effectuée il y a 1:04:52 le ven. 24 mai 2019 10:05:32 CEST.
Le paquet bash-4.4.23-1.fc27.x86_64 est déjà installé, ignorer
Le paquet bc-1.07.1-3.fc27.x86_64 est déjà installé, ignorer
Le paquet binutils-2.29-13.fc27.x86_64 est déjà installé, ignorer
Le paquet bison-3.0.4-8.fc27.x86_64 est déjà installé, ignorer
Le paquet bzip2-1.0.6-24.fc27.x86_64 est déjà installé, ignorer
Le paquet diffutils-3.6-3.fc27.x86_64 est déjà installé, ignorer
Le paquet elfutils-0.170-10.fc27.x86_64 est déjà installé, ignorer
Le paquet elfutils-devel-0.170-10.fc27.x86_64 est déjà installé, ignorer
Le paquet findutils-1:4.6.0-19.fc27.x86_64 est déjà installé, ignorer
Le paquet flex-2.6.1-5.fc27.x86_64 est déjà installé, ignorer
Le paquet gawk-4.1.4-8.fc27.x86_64 est déjà installé, ignorer
Le paquet gcc-7.3.1-6.fc27.x86_64 est déjà installé, ignorer
Le paquet git-2.14.5-1.fc27.x86_64 est déjà installé, ignorer
Le paquet gzip-1.8-4.fc27.x86_64 est déjà installé, ignorer
Le paquet hmaccalc-0.9.14-8.fc27.x86_64 est déjà installé, ignorer
Le paquet hostname-3.18-4.fc27.x86_64 est déjà installé, ignorer
Le paquet kmod-25-1.fc27.x86_64 est déjà installé, ignorer
Le paquet m4-1.4.18-5.fc27.x86_64 est déjà installé, ignorer
Le paquet make-1:4.2.1-4.fc27.x86_64 est déjà installé, ignorer
Le paquet net-tools-2.0-0.47.20160912git.fc27.x86_64 est déjà installé, ignorer
Le paquet openssl-1:1.1.0i-1.fc27.x86_64 est déjà installé, ignorer
Le paquet openssl-devel-1:1.1.0i-1.fc27.x86_64 est déjà installé, ignorer
Le paquet patch-2.7.6-5.fc27.x86_64 est déjà installé, ignorer
Le paquet perl-Carp-1.42-395.fc27.noarch est déjà installé, ignorer
Le paquet perl-devel-4:5.26.2-407.fc27.x86_64 est déjà installé, ignorer
Le paquet perl-generators-1.10-6.fc27.noarch est déjà installé, ignorer
Le paquet perl-interpreter-4:5.26.2-407.fc27.x86_64 est déjà installé, ignorer
Le paquet pesign-0.112-20.fc27.x86_64 est déjà installé, ignorer
Le paquet redhat-rpm-config-79-1.fc27.noarch est déjà installé, ignorer
Le paquet rpm-build-4.14.2.1-1.fc27.x86_64 est déjà installé, ignorer
Le paquet tar-2:1.29-7.fc27.x86_64 est déjà installé, ignorer
Le paquet xz-5.2.3-4.fc27.x86_64 est déjà installé, ignorer
Dépendances résolues.
Rien à faire.
Terminé !
Mise à jour / installation...
   1:kernel-4.18.19-100.fc27          ################################# [100%]
Construction pour plate-formes cibles: x86_64
Construction pour cible x86_64
Exécution_de(%prep) : /bin/sh -e /var/tmp/rpm-tmp.9A7BkW
Initialized empty Git repository in /root/rpmbuild/BUILD/kernel-4.18.fc27/linux-4.18.19-100.fc27.x86_64/.git/
[master f6340aa] Stable update
 2000 files changed, 20317 insertions(+), 12729 deletions(-)
[... 253 lines cut ...]
[105689.889259]  video_usercopy+0xbd/0x5d0 [videodev]
[105689.889265]  ? copy_overflow+0x20/0x20 [videodev]
[105689.889270]  v4l2_ioctl+0x45/0x50 [videodev]
[105689.889274]  do_vfs_ioctl+0xa4/0x630
[105689.889277]  ? kmem_cache_free+0x1a5/0x1d0
[105689.889279]  ? kmem_cache_free+0x1a5/0x1d0
[105689.889282]  ksys_ioctl+0x70/0x80
[105689.889284]  __x64_sys_ioctl+0x16/0x20
[105689.889287]  do_syscall_64+0x5b/0x160
[105689.889291]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[105689.889293] RIP: 0033:0x7f25da50c267
[105689.889294] Code: b3 66 90 48 8b 05 19 3c 2c 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d e9 3b 2c 00 f7 d8 64 89 01 48 
[105689.889328] RSP: 002b:00007f25d4f21b98 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[105689.889330] RAX: ffffffffffffffda RBX: 0000000000000001 RCX: 00007f25da50c267
[105689.889331] RDX: 00007f25d4f21ba0 RSI: 00000000c0405602 RDI: 0000000000000014
[105689.889333] RBP: 0000000000000014 R08: 0000000000000006 R09: 000000002036315a
[105689.889334] R10: 00007f25a4005330 R11: 0000000000000246 R12: 00007f25d4f21c80
[105689.889335] R13: 00000000c0405602 R14: 00007f25d4f21e90 R15: 00007f25a40039d0
[105689.889337] ---[ end trace 466380dfafe45a25 ]---

Thanks for any help

Cheers

Karim

ev-mp, 2019-05-27:
@karimbernardet hello,
The mentioned kernel patch was contributed by community and hasn't been updated for a while - it isn't going to work with kernel 4.18.
On the other hand - patching kernel is not a prerequisite for running Librealsense,  and, as a fact, multiple platforms successfully use unpatched kernels (e.g. RaspberryPi ).

Please post the application failure log for inspection - the attached kernel error regarding the missing fourcc are expected and do not indicate issue with Librealsense.

Additionally  - note that the device is being recognized as USB2:
>[105689.676565] usb 1-2: new high-speed USB device number 22 using xhci_hcd
[105689.804541] usb 1-2: New USB device found, idVendor=8086, idProduct=0ad6, bcdDevice=50.8f

karimbernardet, 2019-05-28:
Hi
Thanjks for the reply. Version of the firmware of the camera is 5.8.15.
I have upgraded my laptop to fedora 28 to have version 2 of realsense. With no patched kernel and when I start realsense-viewer I have :

[karim@10-83-102-17 ~]$ realsense-viewer 
 28/05 16:32:45,565 ERROR [140445375661824] (types.h:178) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,565 ERROR [140445375661824] (types.h:178) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,565 ERROR [140445375661824] (types.h:178) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,568 ERROR [140445375661824] (types.h:178) xioctl(UVC_GET_LEN) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,568 ERROR [140445375661824] (types.h:178) xioctl(VIDIOC_G_CTRL) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,568 ERROR [140445375661824] (types.h:178) xioctl(UVC_GET_LEN) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,568 ERROR [140445375661824] (types.h:178) xioctl(UVC_GET_LEN) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,569 ERROR [140445375661824] (types.h:178) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,569 ERROR [140445375661824] (types.h:178) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,569 WARNING [140445375661824] (sensor.cpp:313) Unregistered Media formats : [ UYVY ]; Supported: [ ]
 28/05 16:32:45,668 ERROR [140445375661824] (types.h:178) xioctl(VIDIOC_G_CTRL) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,668 WARNING [140445375661824] (sensor.cpp:691) Exception was thrown when inspecting properties of a sensor
 28/05 16:32:45,668 ERROR [140445375661824] (types.h:178) xioctl(VIDIOC_G_CTRL) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:45,668 WARNING [140445375661824] (sensor.cpp:691) Exception was thrown when inspecting properties of a sensor
 28/05 16:32:46,567 ERROR [140445287966464] (types.h:178) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:46,567 ERROR [140445287966464] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
 28/05 16:32:46,931 ERROR [140445660936896] (types.h:178) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device
RealSense error calling rs2_is_enabled(dev:0x7fbbf8000b20, enabled:0):
    set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device

and the system says : 
[ 4650.308330] usb 1-1: new high-speed USB device number 4 using xhci_hcd
[ 4650.434921] usb 1-1: New USB device found, idVendor=8086, idProduct=0ad6, bcdDevice=50.8f
[ 4650.434932] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 4650.434935] usb 1-1: Product: Intel(R) RealSense(TM) 430
[ 4650.434937] usb 1-1: Manufacturer: Intel(R) RealSense(TM) 430
[ 4650.437544] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4650.437760] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 430 (8086:0ad6)
[ 4650.439644] uvcvideo: Unable to create debugfs 1-4 directory.
[ 4650.440034] uvcvideo 1-1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4650.440039] uvcvideo 1-1:1.0: Entity type for entity Processing 2 was not initialized!
[ 4650.440043] uvcvideo 1-1:1.0: Entity type for entity Intel(R) RealSense(TM) 430 with was not initialized!
[ 4650.440046] uvcvideo 1-1:1.0: Entity type for entity Camera 1 was not initialized!
[ 4650.440487] input: Intel(R) RealSense(TM) 430: Int as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/input/input22

Cheers

Karim

karimbernardet, 2019-05-28:
and I am using 
[karim@10-83-102-17 ~]$ rpm -qa | grep realse
librealsense-2.10.1-1.fc28.x86_64
librealsense-doc-2.10.1-1.fc28.x86_64
librealsense-devel-2.10.1-1.fc28.x86_64

karimbernardet, 2019-05-30:
Hi
Using vmplayer, I have setup a vm with an Ubuntu with 18.04.02 LTS. The camera seems correctly detected but when I try to update the firmware I got

root@osboxes:~# intel-realsense-dfu -b 003 -d 007 -f -i /home/osboxes/Signed_Image_UVC_5_11_4_0.bin 

 - Intel® RealSense™ DFU Tool v1.1.0-

Device busnum = 3
Device devnum = 7
FW file path = /home/osboxes/Signed_Image_UVC_5_11_4_0.bin
FW Version of .bin File   = 5.11.4.0

Error: libusb_interrupt_transfer failed. err = -1 - LIBUSB_ERROR_IO
!

karimbernardet, 2019-05-30:
I am using the usb cable of my phone (I do not have the original one), this is maybe the problem in fact ...

karimbernardet, 2019-05-31:
I bought a new cable
It works now on the virtual machine
Sorry :-)

ev-mp, 2019-06-10:
@karimbernardet hello. 
It is not clear though whether you've ultimately managed to upgrade the device FW, but in your case it is highly recommended.
The originally mentioned FW version 5.8.15 is rather old and is most certainly not compatible with kernel 4.18 and latest SDK versions (v2.16.1+) due to the up-streamed fix [here](https://lkml.org/lkml/2017/9/6/775). 
That patch has been [added to Librealsense source tree](https://github.com/IntelRealSense/librealsense/commit/bab07bb4d7dba23979f00239b7f98de4e7f99f6a) in v2.16.1 for lower kernels as well, but since you're using a different distro then you would receive it via Fedora updates

closed 2019-06-25 by RealSenseCustomerSupport