realsenseai/librealsense #4069: how to set my depth_image like this one

gravitychen, 2019-05-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Operating System & Version |   Win 10  | 
| Language                            |  python              | 

### Issue Description

Hello . I have a question here.

I want to change my depth image like 

[image]

so,I adjust the parameter 
[image]

But finally I got this 
[image]

Do you know hot to enhance the face quality? ( we basically cannot see the eyes , nose and lips clearly like the example beyond)

thank you very much

Best regards

AntoineWefit, 2019-05-24:
You should use the [RealSense Viewer](https://software.intel.com/en-us/realsense/d400/get-started) to set your settings and save them.
For the amount of details you need, I think that you should also be as close as possible to get max accuracy from the camera.
Try the attached [config.txt](https://github.com/IntelRealSense/librealsense/files/3217251/config.txt) file (save it as "config.json") and enable some post processing to smooth the mesh.

Hope this'll help !

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@gravitychen Some modifications were made based on config.txt provided by AntoineWefit and please see https://gist.github.com/sherrysun2018/a56e6091d7ba51220ab0169240e84775

Please have try this json preset to see if the view is better or not.
Also you can add some post processing as AntoineWefit suggested. Thanks!

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@gravitychen Did you get chance to try? Any update? Looking forward to your reply.Thanks!

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@gravitychen If no other questions, will close the ticket.

closed 2019-07-05 by RealSenseCustomerSupport

gravitychen, 2019-07-12:
sorry about my late reply, I found that you can simply set the max distance and min distance ,and change the depth unit lower,then that's all