realsenseai/librealsense #4070: rs-data-collect D435i issue

kat-mcg-lin, 2019-05-24:
Currently trying to use D435i depth camera on a Jetson Nano

Device is recongized fine and the realsense viewer works perfectly. However, when I try to use rs-data-collect gyro and accel are not "supported by this device" so I cannot collect any data on the actual movement of the camera. I followed the config instructions from the git about rs-data-collect, so I do not think it is an issue with the config file.

How do I resolve this?

I've attached the output of what happens when I run rs-data-collect

```
Running rs-data-collect: -c ~/documents/data_collection/config_files/d435i_config.cfg -f ~/documents/data_collection/tests/test2.csv 

Request added for DEPTH,640,480,30,Z16,0
Request added for INFRARED,640,480,30,Y8,1
Request added for INFRARED,640,480,30,Y8,2
Invalid syntax in configuration line COLOUR,640,480,30,RGB8,0
Request added for ACCEL,1,1,63,MOTION_XYZ32F
Request added for GYRO,1,1,200,MOTION_XYZ32F

Device selected: 
Intel RealSense D435I. USB Type: 3.2
S.N: 843212070694
FW Ver: 05.11.04.00
255.255.255.255

User streams requested: 5, actual matches: 3

The following request(s) are not supported by the device: 
Type: Gyro, Idx: 2, MOTION_XYZ32F, [1x1], 200fps
Type: Accel, Idx: 2, MOTION_XYZ32F, [1x1], 63fps

Data collection started.... 

Collecting data for 1 sec
Collecting data for 2 sec
Collecting data for 3 sec
Collecting data for 4 sec

Data collection accomplished with [100-100] frames recorded per stream
Serializing captured results to ~/documents/data_collection/tests/test2.csv
Task completed

```

ev-mp, 2019-05-26:
@kat-mcg-lin hello, the log notifies about typo in the configuration file you use (highlighted):
>Request added for INFRARED,640,480,30,Y8,2  
Invalid syntax in configuration line **COLOUR**,640,480,30,RGB8,0  
Request added for ACCEL,1,1,63,MOTION_XYZ32F  

Switching to **_COLOR_** should probably fix it- I just rechecked it with the [SDK reference section](https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect#config-file-format) and v2.22.

In case it doesn't - please provide the FW and SDK versions and post the config. file for review

kat-mcg-lin, 2019-05-28:
I can make the switch from colour to color, but the gyro and accel are the two pieces of information I want to collect, but are the ones that are not working 

`The following request(s) are not supported by the device: `

`Type: Gyro, Idx: 2, MOTION_XYZ32F, [1x1], 200fps`
`Type: Accel, Idx: 2, MOTION_XYZ32F, [1x1], 63fps`

ev-mp, 2019-05-28:
@kat-mcg-lin hello, this message could be due to the previous error encountered.
Please rerun the tool using the rectified/referenced config file and update.

kat-mcg-lin, 2019-05-28:
oh awesome! You're right it worked!
Thank you :)

closed 2019-05-28 by kat-mcg-lin