realsenseai/librealsense #4071: "Frame didn't arrived within 15000"

filipematosinov, 2019-05-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (5.11.4.0) | 
| Operating System & Version |   {Ubuntu 18.04| 
| Kernel Version (Linux Only)    |  (4.18.0-20-generic9                                         | 
| Platform                                 | Up Squared |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am using two cameras connected to an Up Squared board using USB3.0 ports. I am experiencing the error : "Frame didn't arrived within 15000" constantly until I completelçy power off the board, remove the cameras and power it on again. Only rebooting the board doesn't solve the problem. This happenings with every script I run including rs-save-to-disk example. The board is not connected to any display at the moment and is being controlled over ssh.

ev-mp, 2019-05-26:
@filipematosinov hello, 
What is the camera type and the SDK versions? 
Can you provide logs for review:
- librealsense application log can be enabled via `export LRS_LOG_LEVE=DEBUG && rs-save-to-disk`
- console log for `rs-enumerate-devices`
- Kernel log `dmesg -wT` activated since the device plugin (in a separate ssh session).

LDWinner, 2019-06-04:
I have also encountered this issue. Have you fixed it?

filipematosinov, 2019-06-06:
No I haven't but I found a workaround. Whenever it happens I just flash the last firmware again (with the option -f to force to install again) and then everything is okay :)

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@filipematosinov @LDWinner What SDK version did you use when get the issue?

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@filipematosinov @LDWinner Could you please update if you still need support on this? Thanks!

filipematosinov, 2019-07-05:
Yes, I am running the sdk version 2.24. 

The requested logs (during this logs no error occured: 
[rs-enumerated-to-dik.log](https://github.com/IntelRealSense/librealsense/files/3362154/rs-enumerated-to-dik.log)
[rs-save-to-disk.log](https://github.com/IntelRealSense/librealsense/files/3362155/rs-save-to-disk.log)

RealSenseCustomerSupport, 2019-07-12:
----------------------------------------------

@filipematosinov What's the camera type? Could you please also provide the output of "rs-enumerate" ?

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

@filipematosinov  Seen from your log "baseline = -49.896957 mm" I guess you're using D435. What streams and resolution configured when you get the issue? And did you get chance to try the latest FW 5.11.6.250 in https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

@filipematosinov Any other questions about this?

AntoineWefit, 2019-07-25:
@filipematosinov @LDWinner looks like triggering a "USB reset" when the board is booting could be a workaround :)
Hope this'll help !

filipematosinov, 2019-07-29:
> @filipematosinov @LDWinner looks like triggering a "USB reset" when the board is booting could be a workaround :)
> Hope this'll help !

@AntoineWefit  thanks for the suggestion I will look into that and try it out :)

filipematosinov, 2019-07-29:
> @filipematosinov Seen from your log "baseline = -49.896957 mm" I guess you're using D435. What streams and resolution configured when you get the issue? And did you get chance to try the latest FW 5.11.6.250 in https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip

@RealSense-Customer-Engineering Yes I am now using the new firmware. Now I don't receive the same error anymore (or at least not often) but this one instead: RuntimeError: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

I am using depth and RGB streams both at 848x480 @ 15 fps

RealSenseCustomerSupport, 2019-08-16:
----------------------------------------------

@filipematosinov Did you see any streaming issue when the RuntimeError occurred? And which sample did you run when get the issue?

RealSenseCustomerSupport, 2019-08-20:
----------------------------------------------

@filipematosinov Could you please update so that we can further support on this? Thanks!

filipematosinov, 2019-08-20:
> @filipematosinov Did you see any streaming issue when the RuntimeError occurred? And which sample did you run when get the issue?

Yes, the script just stops. I am getting that error at random times, using the python API or the rs-save-to-disk

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

@filipematosinov Could you please provide the librealsense logs for RuntimeError issue? Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@filipematosinov Did you get the logs? Thanks!

RealSenseCustomerSupport, 2019-10-12:
----------------------------------------------

@filipematosinov Didn't get response from you for weeks. Did you still have the issue? Looking forward to your update.Thanks!

closed 2019-10-25 by RealSenseCustomerSupport

sycdlcrain, 2020-06-10:
Hi all,

I was getting this error message trying to run the examples on a raspberry pi. Rebuilding with a smaller image size worked for me. 
For example in rs-capture.cpp change:
`window app(1280, 720, "RealSense Capture Example");`
to
`window app(424, 240, "RealSense Capture Example");`