realsenseai/librealsense #4072: Cannot connect to D435i. Using Jetson TX2

lifuguan, 2019-05-25:
| Board | SDK | OS version | device |
| --- | ---- | ---- | ---- |
| Jetson TX2 | JetPack 4.3 | 18.04 bionic | D435i |

## I've compiled the whole profile following the instructions, but when I ran the `rs-capture` it told me I don't connect the the device.....what the hell???

#### Here is my `rs-capture` info :

```bash
robomaster@robomaster-desktop:~$ rs-capture 
RealSense error calling rs2_pipeline_start(pipe:0x55a53380d0):
    No device connected
```
#### Here is my `lsusb` info : 
```bash 
robomaster@robomaster-desktop:~$ lsusb
Bus 002 Device 005: ID 8086:0b3a Intel Corp.   # It is the D435i
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 060b:7a03 Solid Year 
Bus 001 Device 003: ID 413c:301a Dell Computer Corp. 
```

#### Here is my `dmesg`  info : 

```bash
[  668.165820] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[  668.174063] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input10
[  668.175374] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  668.175393] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  668.180343] uvcvideo: Unable to create debugfs 2-5 directory.
```

# Is their any suggestions or solutions for me? Appreciate that.

ev-mp, 2019-05-27:
@lifuguan hello,
Can you specify which Installation Guide did you follow and whether you've applied the `udev-rules` from  [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites).
Please attach kernel and application logs:
 - `sudo dmesg -C && dmesg -wT`
 - `export LRS_LOG_LEVEL=DEBUG && rs-capture`

BeautyBeast, 2019-05-28:
@lifuguan hello,
I plan to use the same devices and versions as yours. Did any other issues happened during the installation of the D435i? 
BTW, would you please share the tutorial? Thanks

lifuguan, 2019-05-29:
@BeautyBeast hello,
      I found a profile which is forked from the JetsonHacks. And he said he scuessfully install the librealsenseTX2 on jetpack 4.2.
- https://github.com/Tengyun-Mo/buildLibrealsense2TX2

lifuguan, 2019-05-29:
@ev-mp hello, 
      Thanks for answering my question!   My kernel is : `4.9.140-tegra`.
The official(JetsonHack) didn't say if they support jetpack4.2 or not. But I found a profile which claimed that he modified the official code to support Jetpack 4.2. 
Here is his profile's address : https://github.com/Tengyun-Mo/buildLibrealsense2TX2.git
#### `sudo dmesg -C && dmesg -wT` 's info : 
```
robomaster@robomaster-desktop:~$ dmesg -wT
[日 5月 26 15:03:46 2019] usb 2-1: new SuperSpeed USB device number 2 using tegra-xusb
[日 5月 26 15:03:47 2019] usb 2-1: New USB device found, idVendor=8086, idProduct=0b3a
[日 5月 26 15:03:47 2019] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[日 5月 26 15:03:47 2019] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[日 5月 26 15:03:47 2019] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[日 5月 26 15:03:47 2019] usb 2-1: SerialNumber: 904523050797
[日 5月 26 15:03:47 2019] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[日 5月 26 15:03:47 2019] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[日 5月 26 15:03:47 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[日 5月 26 15:03:47 2019] uvcvideo: Unable to create debugfs 2-2 directory.
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.0: Entity type for entity Processing 2 was not initialized!
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.0: Entity type for entity Camera 1 was not initialized!
[日 5月 26 15:03:47 2019] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input7
[日 5月 26 15:03:47 2019] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[日 5月 26 15:03:47 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[日 5月 26 15:03:47 2019] uvcvideo: Unable to create debugfs 2-2 directory.
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.3: Entity type for entity Processing 7 was not initialized!
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.3: Entity type for entity Extension 8 was not initialized!
[日 5月 26 15:03:47 2019] uvcvideo 2-1:1.3: Entity type for entity Camera 6 was not initialized!
[日 5月 26 15:03:47 2019] usbcore: registered new interface driver uvcvideo
[日 5月 26 15:03:47 2019] USB Video Class driver (1.1.1)
[日 5月 26 15:03:54 2019] usb 2-1: usb_suspend_both: status 0
[日 5月 26 15:03:54 2019] usb usb2: usb_suspend_both: status 0
[日 5月 26 15:04:42 2019] usb 2-1: usb_suspend_both: status 0
[日 5月 26 15:04:42 2019] usb usb2: usb_suspend_both: status 0
[日 5月 26 15:04:47 2019] usb 2-1: usb_suspend_both: status 0
[日 5月 26 15:04:47 2019] usb usb2: usb_suspend_both: status 0
[日 5月 26 15:04:52 2019] usb 2-1: usb_suspend_both: status 0
[日 5月 26 15:04:52 2019] usb usb2: usb_suspend_both: status 0
```
#### `rs-capture`'s info : 
```bash
robomaster@robomaster-desktop:~$ export LRS_LOG_LEVEL=DEBUG && rs-capture
RealSense error calling rs2_pipeline_start(pipe:0x55a89c8c80):
    No device connected
```

closed 2019-06-21 by RealSenseCustomerSupport