realsenseai/librealsense #4079: What disparity shift value to use for range from 1m to 1.5m

tispratik, 2019-05-28:
We have a usecase wherein our camera is mounted at 4ft high from a conveyor belt. We are trying to focus on objects within the range of 100cm to 150cm. I am hoping to get better depth accuracy by increasing the disparity shift number. But reading through https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf I couldn't really understand what is the correct value to use. If we use disparity shift of 50, according to the document we will get minZ=30cm and maxZ=110cm... I am not sure if I should set minZ to 70 or something else to get into the 100cm to 150 cm range.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | NVIDIA GTX 1070  |
| SDK Version                            |                            | 
| Language                            |  python }                          | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

tispratik, 2019-05-28:
Found Baseline & Focal Length for D415 in the product specs.

[image]

[image]

Now the question is how to get Focal Length in Pixels from mm.

agrunnet, 2019-05-28:
You can read the focal length in pixels through the SDK (intrinsics). 

But you won’t be able to improve depth accuracy by changing the disparity shift. (At least not unless you use it to move closer than the minZ)

tispratik, 2019-05-28:
Thanks for the information @agrunnet. I am wondering what help does disparity shift provide if not for helping in getter depth values?

agrunnet, 2019-05-28:
The disparity shift is mainly for reducing the minimum distance. Which allows you to get nearer to objects. (Which does then improve the noise as the square of the distance away). 
There is a white paper on the intel RealSense website that describes the best known methods for tuning performance which includes a discussion about this so should be helpful.

tispratik, 2019-05-29:
Thanks @agrunnet, as described in that paper, why does setting a disparity shift of 50 set the MaxZ to 110cm? Shouldn't MaxZ be infinite?

RealSenseCustomerSupport, 2019-06-27:
----------------------------------------------

As the whitepaper mentioned "Because of the inverse relationship between disparity and depth, MaxZ will decrease much faster than minZ as the disparity shift is increased."
The disparity shift method decreases the minZ, but it sacrifices the maxZ.

closed 2019-07-19 by RealSenseCustomerSupport

hrdkjain, 2019-09-16:
Using the MinZ and MaxZ values shown in Disparity Figure on [this page](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance), approximate formulae for depths should be:
MinZ = focal length(pixels) x baseline(mm) / (126+disparityShift)
MaxZ = focal length(pixels) x baseline(mm) / (disparityShift)