realsenseai/librealsense #4080: Downward facing T265

vidullan, 2019-05-28:
Hi,

I'm using an Intel 265 tracking camera and am having issues when using it downward facing.

I have no issues when it is mounted forward facing. But if placed facing downwards, then the translations are reported correctly but the rotation values are incorrect and sometimes seem to be coupled. 

Is this a known issue?

SlavikLiman, 2019-05-28:
Hi @vidullan ,
As a human, facing downwards, will have a hard time to navigate, the T265 in your case is probably also does. 
1. What is your platform?
2. Is there a specific reason you decided to install it facing the ground?
3. How far is the T265 located from the ground?

patrickpoirier51, 2019-06-02:
@SlavikLiman  
I presume this is the standard use case for a Drone Platform flying from 0 to 50 Meter outdoor.  
There are many UAV SLAM systems using this configuration.

vidullan, 2019-06-03:
Hi @SlavikLiman,
I'm not sure I entirely agree with your comparison with humans and the T265. Sure, we as humans will have a hard time navigating facing downwards if facing a carpet or something, but imagine a case where there was a nice pattern printed on the floor. In that case if we knew the pattern we would find it just as easy (barring a stiff neck from looking down all the time). 

The camera theoretically should not have a problem if the ground has distinguishable features for V-Slam. The problem here is that the implementation does not take into account that the camera can be pointed towards the ground and is messing up the coordinate frames when started in that orientation. I was wondering if the Intel engineers/whoever writes the firmware would be able to fix this.

@patrickpoirier51 is correct in that this is a standard case for a drone as there usually are better features to be found pointing towards the ground rather than at the horizon when flying along.

radfordi, 2019-06-03:
Hi @vidullan,

You are exactly correct that the initial "yaw" of the world coordinates will be "random" when you have a downward facing camera.  The system will work just fine, but you will not be able to predict which direction is forward.  We have a way to handle this internally by allowing the user to define the forward direction in device's body coordinate system and then to define which direction in the world coordinates it could be lined up with initially.  e.g. the "pointing out of the top of the camera is the initial forward direction" and it "should point in the world x direction initially".  The T265 accepts this kind of world coordinate definition internally, but we haven't exposed it yer.  I'll move it up on the priority list.

We are also working on a change to make tracking more robust in the downward facing camera case.   There's nothing inherently worse about the downward facing setup; in fact it's what I would recommend for a drone.

Thanks for your feedback.

patrickpoirier51, 2019-06-03:
@radfordi That would be greatly appreciated as we are presently working as a Google Summer Of Code project on the T265. 
We can demonstrate indoor flying with Ros Based Forward Facing T265 running ArduPilot on a 450 size drone and we are getting ready to flip camera for a downward facing indoor and outdoor experiments.

radfordi, 2019-06-03:
@patrickpoirier51, until we have a general setting for this, you can get a semi-consistent yaw by starting on a slightly inclined plane.

patrickpoirier51, 2019-06-03:
@radfordi , so you mean having the camera looking down at an angle of 45 Deg. or less so the IMU gets some consistent gravity vector on the Y axis in order to initialize the system ?

radfordi, 2019-06-03:
@patrickpoirier51, no, I meant having camera face exactly down with respect to the drone body, but slightly tilting the drone itself when you start streaming poses.  You could even point the camera where you want, start stream poses, then set it down.

The direction the camera is looking at start defines the -z world axis, so a slight tilt will give that a more consistent direction.  However, even if the device is looking perfectly downward and we have to pick a random direction for -z, the poses and orientation should always be self-consistent modulo that initial random yaw.

gitak2019, 2019-06-06:
I too would like to use the t265 with a downward orientation.  I ended up mounting them on two drones with a slight tilt so that it initialized consistently at power up.  It would be easier to flat mount the t265 though on future builds.  So, a solution that allowed downward facing mounting with a consistent initialization would be appreciated.

vidullan, 2019-06-09:
Hi @radfordi ,

So I tried to take the approach of initializing the camera with the drone tilted when it is mounted downward facing but it does not seem to work.

I checked the values using the provided example script: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_rpy.py

The roll, pitch, yaw values reported are good when it is kept titled, but as the drone is placed back on ground causing pitch to approach -90 degress, there is a coupling where roll and yaw both increase and the values I assume are being reported in an inertial frame that has changed. When the drone is tiled back again reducing pitch to greater than -90 degrees, the inertial frame switches back to the original camera initialized frame. 

This makes it unusable as the inertial frame keeps shifting based on inclination. Not sure if I misunderstood you, but hopefully the internal fix is exposed soon.

vidullan, 2019-06-09:
@radfordi Also are you aware if the tracking is degraded if there are moving objects in the frame? Due to the fisheye lenses, the front left and right propellers are visible in the frame. 

Sometimes we noticed that the x,y,z, values zero themselves losing track of the drone during forward motion. Not sure if this is a known issue with the camera, or something else wrong on our end. This is one of the reasons why we wanted to downward mount it.

radfordi, 2019-06-10:
Hi @vidullan,

The roll/pitch/yaw coordinates output by [t265_rpy.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_rpy.py) are designed to work with a forward facing camera.  They have a singularity for a downward facing camera.  You'll want to tweak the script with that in mind based on how you have mounted your camera.

Here's [a change to t265_rpy.py](https://gist.github.com/radfordi/1ead04df10335b17808294bc8716de9c) that gives a consistent roll/pitch/yaw when I start facing forward and then point the camera downward.

Yes, moving objects can degrade performance, but the T265 is robust to a certain amount of motion.

RealSenseCustomerSupport, 2019-07-25:
----------------------------------------------

Hi @vidullan,

Any update? Is the intermediate solution provided by @radfordi help your issue?

patrickpoirier51, 2019-07-25:
I can confirm that initializing system with an angle works ok for us
https://files.gitter.im/ArduPilot/VisionProjects/w2Tt/image.png

radfordi, 2019-07-25:
Thanks for confirming @patrickpoirier51.

vidullan, 2019-07-26:
@RealSenseCustomerSupport Hi, thanks for checking in. Since we would have to initialize the system at an angle to have it downward facing we have given up on that till a fix that allows us to power it up downward facing is issued.

In the meantime we have mounted it at a 45 degree angle facing the ground and that seems to work well enough for our purposes.

patrickpoirier51, 2019-07-26:
@radfordi  Jim please note that in order to get the camera facing down with ArduPilot , @hoangthien94 did some modification the the Transpose Matrix, you can look at his script here:
https://github.com/hoangthien94/vision_to_mavros/blob/master/scripts/t265_to_mavlink.py

vizzbee, 2019-08-23:
Hey @patrickpoirier51 @radfordi 
I am attempting indoor navigation using t265 and put it preferrably in downward pose. Did the package by @hoangthien94 work?

patrickpoirier51, 2019-08-23:
Yes it does, and its now fully documented as an ArduPilot WIKI
http://ardupilot.org/copter/docs/common-vio-tracking-camera.html

Its quite easy using the python script as the camera position (Forward -vs- Down Looking)  is a parameter that can be passed as input arguments from the command line.

Enjoy !!

hoangthien94, 2019-08-23:
@vizzbee Just to add to @patrickpoirier51's answer: ArduPilot now supports T265 in forward/downward facing configuration, both with and without ROS. The wikis are available for step-by-step instructions on how to get the system up and running:
- [Wiki for T265 + ROS + ArduPilot](http://ardupilot.org/dev/docs/ros-vio-tracking-camera.html)
- [Wiki for T265 + ArduPilot ](http://ardupilot.org/copter/docs/common-vio-tracking-camera.html#vio-tracking-camera-for-non-gps-navigation)

For the sake of brevity, explanation of the system in the wikis are kept short. More in-depth explanation as well as discussion/asking for any advices when something is not working (ArduPilot-related), you may refer to the following blog posts and [ArduPilot Vision Gitter channel](https://gitter.im/ArduPilot/VisionProjects) :
- T265 + ROS + ArduPilot: [part 1](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-1-getting-started-with-the-intel-realsense-t265-on-rasberry-pi-3b/43162), [part 2](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-2-complete-installation-and-indoor-non-gps-flights/43405), [part 3](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-3-indoor-autonomous-flights-and-performance-tests/43626).
- T265 + ArduPilot: [part 4](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-4-non-ros-bridge-to-mavlink-in-python/44001), [part 5](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-5-camera-position-offsets-compensation-scale-calibration-and-compass-north-alignment-beta/44984).

Hope this helps.

mzahana, 2019-12-20:
@patrickpoirier51 just out of curiosity, did you do outdoor experiments with T265? How was the performance?
The localization i got outdoors is really worse than indoors. I am not sure if there are any specific setting for it to work good outdoor.

neilyoung, 2019-12-20:
Same experience here. One plausible explanation in another thread was the lack of features at short range. In my Tests the drift was very large but in all cases the ranges were too short. There was no single case in which a distance was measured longer than in reality.

patrickpoirier51, 2019-12-20:
@mzahana  Hello,  long time no talk :-)

I manage to make it work on a specific configuration:
- Looking Down 
- A '' skirt'' around the unit acting as a Sun Shade
- Good feastures on the ground @ 15 Meters

Here is the result
https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-4-non-ros-bridge-to-mavlink-in-python/44001/28

mzahana, 2019-12-20:
@patrickpoirier51 YES! It’s been a while. I have been through some transitions :) and it is nice to chat again.  Very nice work on the T265. I am using it with PX4 and it’s straight forward to integrate it through MAVROS. Works nicely indoors. In outdoors, I experienced considerable drift, but did not really have the time to investigate  or improve as I was working on different VIO setups. I guess there are 2 lessons I see here.
* Adjusting the orientation of T265 to capture more features outdoors
* Avoiding excessive sun light by for example, using the nice trick you have of the “skirt” to provide sun shade.

Thanks for the information. Will try that after my vacation!

neilyoung, 2019-12-21:
There is not much wrong with this cam at 20 x 20 m, and maybe also at 20 x 20 x 20. One need to go farther to see issues.

partlygloudy, 2020-02-13:
> 
> We have a way to handle this internally by allowing the user to define the forward direction in device's body coordinate system and then to define which direction in the world coordinates it could be lined up with initially. e.g. the "pointing out of the top of the camera is the initial forward direction" and it "should point in the world x direction initially". The T265 accepts this kind of world coordinate definition internally, but we haven't exposed it yet. I'll move it up on the priority list.

@radfordi Any update on when this functionality may become available? My team has had some success initializing the t265 at an angle before orienting it to face the ground, however, for the particular application we're working on, that won't be an option. We know the camera will always be facing the ground on startup, but we won't be able to touch or move the vehicle in any way. The functionality you describe here is exactly what we're looking for.

dookei, 2020-03-30:
Hi @radfordi just pinging. Any news? We use upward config and also have the same issue when in downward.

RealSenseSupport, 2020-09-10:
Thank you for highlighting the need for updates to accommodate downward facing configuration. We have moved our focus to our next generation of products and consequently, we will not be addressing this any further additions to the latest released FW for downward facing configuration in the T265.

closed 2020-09-15 by RealSenseSupport

DiegoHerrera1890, 2021-11-08:
Hello, dear community. 
I am facing an issue with the T265 camera. May you guys help me?
Thank you.
Here is the link to my issue:
https://github.com/IntelRealSense/librealsense/issues/9933