realsenseai/librealsense #4081: Unable to catpure video

themikem, 2019-05-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.04.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-50-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |   v2.22.0                          | 
| Language                            |  various                         | 
| Segment			|  others                          | 

### Issue Description
unable to capture RGB video via V4L2 via /dev/video2 or in realsense-viewer.

Connection log from dmesg:
```[Mon May 27 21:18:29 2019] usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
[Mon May 27 21:18:29 2019] usb 4-1: New USB device found, idVendor=8086, idProduct=0b07
[Mon May 27 21:18:29 2019] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Mon May 27 21:18:29 2019] usb 4-1: Product: Intel(R) RealSense(TM) Depth Camera 435 
[Mon May 27 21:18:29 2019] usb 4-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[Mon May 27 21:18:29 2019] usb 4-1: SerialNumber: 840213020831
[Mon May 27 21:18:29 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[Mon May 27 21:18:29 2019] uvcvideo: Unable to create debugfs 4-4 directory.
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.0: Entity type for entity Processing 2 was not initialized!
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.0: Entity type for entity Camera 1 was not initialized!
[Mon May 27 21:18:29 2019] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:1d.0/0000:03:00.0/usb4/4-1/4-1:1.0/input/input32
[Mon May 27 21:18:29 2019] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[Mon May 27 21:18:29 2019] uvcvideo: Unable to create debugfs 4-4 directory.
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.3: Entity type for entity Processing 7 was not initialized!
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.3: Entity type for entity Extension 8 was not initialized!
[Mon May 27 21:18:29 2019] uvcvideo 4-1:1.3: Entity type for entity Camera 6 was not initialized!
```

Realsense viewer:
```~$ realsense-viewer 
 27/05 21:14:37,476 WARNING [140019537132224] (synthetic-stream-gl.cpp:79) Initializing rendering, GLSL=1
 27/05 21:14:37,476 WARNING [140019537132224] (synthetic-stream-gl.cpp:88)  0 GPU objects initialized
 27/05 21:14:37,476 WARNING [140019537132224] (synthetic-stream-gl.cpp:131) Initializing processing, GLSL=1
 27/05 21:14:37,558 WARNING [140019537132224] (synthetic-stream-gl.cpp:145)  0 GPU objects initialized
 27/05 21:14:38,306 WARNING [140019146680064] (sensor.cpp:365) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
 27/05 21:14:46,364 WARNING [140019104716544] (backend-v4l2.cpp:908) Frames didn't arrived within 5 seconds
 27/05 21:14:51,369 WARNING [140019104716544] (backend-v4l2.cpp:908) Frames didn't arrived within 5 seconds
 27/05 21:14:56,374 WARNING [140019104716544] (backend-v4l2.cpp:908) Frames didn't arrived within 5 seconds
```

Also have tried via gstreamer and others to capture video, no luck.

Have tried installing both from apt repos and building SDK from scratch.  Again, same results in either case.

I find this message especially odd:
```Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
```

How can a supported media format be unregistered when it is in both lists...?

Any other ideas?  @RealSenseCustomerSupport 

Thanks for your help!

pedrombmachado, 2019-05-29:
Hi @themikem ,
   More information is required... Let's first check if your RS camera is being recognised as a USB3 device. Can you please do a
```
$ lsusb -t
```
Also, check if there are any warnings:
```
$ dmesg | tail
```
Have you tested using the pyrealsense2? Try the following [example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py)

themikem, 2019-06-05:
Here's the lsusb output:
```
~$ lsusb -t
/:  Bus 14.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 13.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 12.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 11.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    |__ Port 2: Dev 3, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 3, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 6, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 14, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 11, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 1: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 10, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 4: Dev 16, If 0, Class=Human Interface Device, Driver=usbhid, 12M
                |__ Port 2: Dev 15, If 0, Class=Vendor Specific Class, Driver=, 12M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
    |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 9, If 0, Class=Human Interface Device, Driver=usbfs, 12M
        |__ Port 1: Dev 9, If 1, Class=Human Interface Device, Driver=usbfs, 12M
        |__ Port 2: Dev 13, If 0, Class=Human Interface Device, Driver=usbfs, 12M
        |__ Port 2: Dev 13, If 1, Class=Human Interface Device, Driver=usbfs, 12M
    |__ Port 6: Dev 8, If 0, Class=Human Interface Device, Driver=usbfs, 12M
    |__ Port 6: Dev 8, If 1, Class=Human Interface Device, Driver=usbfs, 12M
    |__ Port 14: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 12M
```

The realsense is the only camera connected, and it is on bus 4 dev 3 per "normal" lsusb output:
```
~$ lsusb
Bus 014 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 013 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 8086:0b07 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 002 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 1b1c:1b2d Corsair 
Bus 001 Device 013: ID 1b1c:1b34 Corsair 
Bus 001 Device 009: ID 1b1c:1b3b Corsair 
Bus 001 Device 005: ID 1b1c:1b2c Corsair 
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 012: ID 0b05:1867 ASUSTek Computer, Inc. 
Bus 001 Device 014: ID 1b1c:0c0b Corsair 
[... 15 lines cut ...]
[   98.565199] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   98.565202] usb 4-2: Product: Intel(R) RealSense(TM) Depth Camera 435 
[   98.565204] usb 4-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[   98.565207] usb 4-2: SerialNumber: 804113020567
[   98.569949] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[   98.569956] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[   98.570698] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[   98.582036] uvcvideo: Unable to create debugfs 4-3 directory.
[   98.582217] uvcvideo 4-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   98.582221] uvcvideo 4-2:1.0: Entity type for entity Processing 2 was not initialized!
[   98.582224] uvcvideo 4-2:1.0: Entity type for entity Camera 1 was not initialized!
[   98.582338] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:1d.0/0000:03:00.0/usb4/4-2/4-2:1.0/input/input36
[   98.585448] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[   98.585456] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[... cut ...]
[   98.591123] uvcvideo 4-2:1.3: Entity type for entity Camera 6 was not initialized!
```

This appears to mirror the errors shown when running the realsense application, but I don't know how the UUIDs translate to the pixel format types in that earlier error message.  Perhaps that is the source of having UYVY on both the unsupported and supported lists? (i.e. perhaps the text is masking a UUID mismatch...?)

I'll take a look at the python example, but given we're seeing low-level errors here I don't hold out much hope of that changing anything.

Any ideas / suggestions?

ev-mp, 2019-06-10:
@themikem hello,
I've checked the reports but couldn't find anything immediate to point to. Can you rerun `realsense-viewer`, redirect log output to file with `DEBUG` level, run streaming/controls and then post the log file for review?
[image]

An explanation regarding - 
>I find this message especially odd:
`                       Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]`

Librealsense is mostly intended to serve as a transparent proxy for the device functionality. In some cases, however, internal decisions are made to filter out certain data types. In the above case all the FW-supported streaming formats are listed on the right, while Librealsense's decides to pass through all but `UYVY`.

RealSenseCustomerSupport, 2019-06-28:
----------------------------------------------

@themikem Did you get chance to try as ev-mp suggested to collect realsense-viewer logs with DEBUG level? And did you get any issue for depth?

RealSenseCustomerSupport, 2019-07-02:
----------------------------------------------

@themikem Could you please update if you still need support on this? Thanks!

themikem, 2019-07-06:
@RealSenseCustomerSupport I noticed that new firmware has been posted since I tried this.  Let me update everything and see if I can still reproduce.

RealSenseCustomerSupport, 2019-07-11:
----------------------------------------------

@themikem Did you get it through with the latest firmware? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-15:
----------------------------------------------

@themikem How about your testing result with the latest firmware? Could you please update.Thanks!

RealSenseCustomerSupport, 2019-07-18:
----------------------------------------------

@themikem Any luck with the latest firmware? Thanks!

themikem, 2019-07-18:
@RealSenseCustomerSupport Yes!  Was able to get everything working.

However, I did have to manually re-point the git repo to a different tag in the kernel patch script to get the kernel modules to build on hwe.  I'm on 4.18, but the current head on the hwe branch is now pointing at 5.0, causing the patches to fail.

I'll file under a separate ticket.

themikem, 2019-07-18:
See https://github.com/IntelRealSense/librealsense/issues/4464 for what has delayed addressing this

RealSenseCustomerSupport, 2019-07-22:
----------------------------------------------

@themikem Thanks for the feedback! I saw Dorodnic is looking at issue#4464. Close this issue accordingly.

closed 2019-07-22 by RealSenseCustomerSupport