realsenseai/librealsense #4082: Remote repetitive structures and halos cause near-range noise

McJun1, 2019-05-28:
Required Info |  
-- | --
Camera Model | D435
Firmware Version | 05.11.01.100
Operating System & Version | Ubuntu
Platform | RK3399
SDK Version | 2.18.1
Language | c++
Segment | Robot

D435 is in high precision mode， the resolution is 848x480. We use it to avoid obstacles in the restaurant.But the near-range noise will make the robot stop!

The closer the red pixel is, the closer it is.
[image]
(The repetitive structures cause near-range noise)

[image]
(The halos cause near-range noise)

Please tell me how to filter out these noises, thank you!!!

AntoineWefit, 2019-05-28:
Could you specify the range of detection you are working with ?
I'm no expert whatsoever, but it might be easier to set a small viewing range (50 cm to 150 cm maybe ?) and to focus on improving this "narrowed" point cloud.
Setting the _**Depth Units**_, _**Depth Clamp Min & Max**_ and adjusting the _**Disparity Shift**_ under _Stereo Module/Advanced Controls/Depth Table_ will do that.
I'm not sure about what output you are expecting (completely filled mesh ? sharp objects ?) though...

With those cameras, there is always a risk of losing data depending on the environment. Things like office lights or reflective surfaces (like this big fridge in your picture) tends to mess a bit with the reconstruction. Also, as you stated, repetitive patterns are not helping !
-> Setting the exposure on auto could help, try adjusting the laser power too.

Anyway, once you've set your parameters, I would advise you to take a look at the post processing filters, those can really improve the overall quality of your point cloud.

Hope this will help !

McJun1, 2019-05-28:
> Could you specify the range of detection you are working with ?
> I'm no expert whatsoever, but it might be easier to set a small viewing range (50 cm to 150 cm maybe ?) and to focus on improving this [...]

0-3 meter I am working with

dorodnic, 2019-05-28:
One solution that tested well in the field is adding a physical filter in front of the camera, that would pass IR and block visible light, and increasing the laser power. 
At close range it would break repetitive patterns by introducing a little bit of texture from the projector, and at longer range it will filter the false alarms because less light from that area will make it into the camera. This has proven to work fairly well for customers operating indoor for collision avoidance use case. 
We also tried different software based solutions, but it's hard to filter the outliers without significantly reducing the total number of points.

McJun1, 2019-05-29:
> One solution that tested well in the field is adding a physical filter in front of the camera, that would pass IR and block visible light, and increasing the laser power.
> At close range it would [...]

Do you have an adapted physical filter recommendation?Or how can I find this physical filter?

McJun1, 2019-05-29:
I can't find some  adapted physical filter.Do you have an adapted physical filter recommendation?Or how can I find this physical filter?Thank you!!!

AntoineWefit, 2019-05-29:
Haven't personally tested it, but I think that you are looking for something like the followings : 
Either [a long-pass filter](https://www.edmundoptics.fr/p/1quot-diameter-optical-cast-plastic-ir-longpass-filter/5417/), which will let pass anything around 750 nm and beyond.
Or [a band-pass filter](https://www.edmundoptics.fr/p/850nm-cwl-10nm-fwhm-125mm-mounted-diameter/22452/), which will let pass a precise range of wavelengths.

In any case, you'll have to aim for a 850 nm wavelength (the IR emitter) for your choices.

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

Hi McJun1,
Do you find the filter? You may try filter Lee #87.

closed 2019-07-19 by RealSenseCustomerSupport