realsenseai/librealsense #4085: How to get D400 series camera status that is it ON or OFF or Disconnected

ambikesh13499, 2019-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-05-28:
@itachi134999 , can you elaborate what is the actual use-case?
My guess that you refer to the librealsense's `context` object, which provides (among others):
- Device factory
- System  Bookkeeper - handle OS-level device connect/disconnect event.

In order to be notified when device is attached/disconnect one needs to register a callback.
Getting the list of currently attached devices on demand is achieved with `auto dev_list = context.query_devices()`.

With regards to the camera's on/off status - I'm not sure whether these terms can be translated directly. Please review the SDK's core capabilities, especially with regards to multi-cam/multi-app setups that are listed [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model) .

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@ambikesh13499 With ev-mp's previous reply, did you get it through now? Looking forward to your update. Thanks!

itachi134999, 2019-06-25:
Yeah this issue is solved. Thanks

closed 2019-06-26 by RealSenseCustomerSupport