realsenseai/librealsense #4086: Firmware 5.11.4.0 issues with D435

SirDifferential, 2019-05-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.11.4.0 | 
| Operating System & Version |    Ubuntu 18.04.2 LTS | 
| Kernel Version (Linux Only)    |  4.15.0-37-generic, emutex Ubuntu 18 for Upboard | 
| Platform                                 | Upboard  |
| SDK Version                            |   Latest master  | 
| Language                            |  C  | 

### Issue Description

Using the latest development firmware `Signed_Image_UVC_5_11_4_0.bin`, sometimes RealSense D435 devices don't appear to work. The problem manifests as follows:

- lsusb lists the device as `8086:0ad4` when normally the device appears as `8086:0b07`
- The device cannot be accessed via the RealSense API
- kern.log contains numerous errors (see the attached kern.log.gz)

We set up a test platform of 6 Upboard SoCs with their own D435 sensors, all using the same cables. 2 out of 6 devices showed this problem. We eliminated cables and Upboards as a problem, rebooted the devices and isolated the issue to be the RealSenses themselves, then downgraded the firmware back to `Signed_Image_UVC_5_11_1_100.bin`, after which the RealSenses began to function normally. See kern2.log.gz for what the kernel prints with the same Upboard, cables and RealSense with the older firmware. There are still errors, but the device appears to work.

[kern.log.gz](https://github.com/IntelRealSense/librealsense/files/3227462/kern.log.gz)
[kern2.log.gz](https://github.com/IntelRealSense/librealsense/files/3227463/kern2.log.gz)

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@SirDifferential Could you please check if the latest firmware 5.11.6.250 have the issue or not? Thanks!
https://downloadmirror.intel.com/28870/eng/D400_Series_Production_FW_5_11_6_250.zip

SirDifferential, 2019-06-26:
The new firmware appears to work on the same 6 realsenses. Thank you for the continued updates.

closed 2019-06-26 by SirDifferential