realsenseai/librealsense #4089: Pyrealsense2 - Terminal Works, Pycharm Doesn't

kat-mcg-lin, 2019-05-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Firmware Version                   | ?| 
| Operating System & Version |   Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    | 4.9.14                                         | 
| Platform                                 | NVIDIA Jetson Nanno  |
| SDK Version                            |  2.0                          | 
| Language                            |  Python                       | 

I have been using pycharm on my jetson nano with small simple codes (I'm new to python and wanted to test out basic commands) and now that I want to start using the cameras I went to install pyrealsense2. I followed the instructions here: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python
but pycharm never recongizes pyrealsense2 when I try to import it

> Traceback (most recent call last):
>   File "/home/user/documents/projects_pycharm/GetPixel.py", line 3, in <module>
>     import pyrealsense2
> ModuleNotFoundError: No module named 'pyrealsense2'

However, when I use python3 command in my terminal I can import pyrealsense2 just fine- is there something I'm missing to make Pycharm work?

kat-mcg-lin, 2019-05-28:
#3838

exact issue as this person!

pedrombmachado, 2019-05-29:
Hi @kat-mcg-lin ,
  Did you add 
> export PYTHONPATH="$PYTHONPATH:/usr/local/lib"
to your `~/.bashrc` and 
```
$ nano ~/.bashrc
$ source ~/.bashrc
```

kat-mcg-lin, 2019-05-30:
Hi! 

I have run the export PYTHONPATH line before and even tried different locations, although I don't know/understand how to add it to the ~/.bashrc, is that a physical file I can open?

Also just tried running the other two lines you mentioned, the nano one returned the error:
`bash: nano: command not found`

the second line had no return but also no error

pedrombmachado, 2019-05-30:
Hi @kat-mcg-lin ,
  Apparently, you need to install nano (text editor) or use a different text editor (e.g. vim, kate, gedit, etc)
  I suggest you read about [bashrc](https://www.maketecheasier.com/what-is-bashrc/) before you edit it.

kat-mcg-lin, 2019-05-31:
Sounds good! 
Thanks for your help and the article! @pedrombmachado :)

closed 2019-05-31 by kat-mcg-lin