realsenseai/librealsense #4091: on Android USB 3.2, the example `capture` constantly report "bulk_transfer returned error, error: Out of memory"

GucciPrada, 2019-05-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.11.04 | 
| Operating System & Version |  MacOS 10.13.6+Android Studio 3.4.1 | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | Andriod 8.1(linux kernel 4.5) +USB 3.1 Gen2 c-to-c cable  |
| SDK Version                            |  2.22.0     | 
| Language                            |  java &c++ }                          | 
| Segment			|  Smartphone                         | 

### Issue Description
#### short version:
   I ran the wrappers/android/examples/capture app on a Android Phone(has a usb 3.1 compatible type-c female interface) with D415 plugged in. But there was only black screen on the UI, which means the text "Connect to a Realsense Camera" was gone but no video data rendered. And at the same time, I found in the logcat of Android studio there reported constantly a plenty of warning messages: 
```
2019-05-29 00:01:49.276 27981-28475/com.intel.realsense.capture W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
```
the full log was something as follow:
<details>
<summary>Click me to expand</summary>

```
2019-05-29 00:01:45.036 27981-28098/com.intel.realsense.capture D/librs DeviceWatcher: Device: /dev/bus/usb/003/002 added successfully
2019-05-29 00:01:49.237 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.237 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.239 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.239 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.241 27981-27981/com.intel.realsense.capture D/librs capture example: try start streaming
2019-05-29 00:01:49.241 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.241 27981-27981/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/003/002
    path- /dev/bus/usb/003/002
    susb specification- 320
2019-05-29 00:01:49.276 27981-28475/com.intel.realsense.capture W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
......  many many repeated warning lines as above .....

```
</details>

**My Question is: Has anybody ever seen this problem before? What should I do to fix this?**

#### Longer version:

##### My setup:
 
"Android Phone Huawei P20 + USB 3.1 Gen 2 C Male to C Male cable + Realsense D415".
I've smiply checked the cable, it is compliant with USB 3.1 Gen 2 10/20Gbps 5A 100W.
I've also checked the Huawei P20's type c interface by plugging it into my Macbook and checking the 'USB 3.0 Bus' section of System Report of the MacBook. It showed the Speed is up to 5Gbps of "Huawei P20+ the c-to-c cable" combination. I guess that means either Huawei P20 or my Macbook has a USB 3.1 Gen 1 or USB 3.0 interface.

##### What I did and what I saw:
I downloaded the librealsense-2.22.0.aar and librealsense-2.22.0.zip from dl.bintray.com.
After that, I import them into the android project of wrappers/android, making both examples/native-example and examples/capture Modules to depend on them.

Then I modified the examples/native-example's source code to enumerate the device, the sensors and the stream profiles of each sensor. And save the enumerated result to a log file: [here](usb3.2-P20-enumerate-log.txt).  In a word, I could tell that the highest profiles (1280x720xZ16x30Hz Depth stream and the 1920x1080xRGB8x30Hz Color stream) are available.

In the next step, I build and ran examples/capture on the Huawei P20. After granting the USB permission the capture app, the screen was black and the message **"27981-28475/com.intel.realsense.capture W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory" ** was flooding the Logcat.

matkatz, 2019-05-29:
Hi @GucciPrada 
That seems to be the same issue that was reported in #3612.
I'm still not able to reproduce it on my RK3399 setup and I plan to try some other devices.
Did you see this issue on earlier versions of librealsense?

GucciPrada, 2019-05-30:
Hi,@matkatz thanks for your reply. I just read that similar issue you mentioned.
I didn't try earlier version of librealsense or firmware. I'll try it out today. And I'm also keep finding other android devices that support usb3.1/3.2 to try. I'll post any news I get at here.

For additional information, 
1.I've tried some android phones that only support usb 2.1(e.g. Oppo Find X/Vivo Nex), the 'capture' example are working nicely, successfully and stably delivering RGBD stream of VGA(640x480)@30Hz. That's quite a nice work from you guys!
However I wanna a higher resolution RGBD stream(.i.e 720p or 1080p).
 Have you or other people successfully made the D415+android deliver 720P or 1080P RGBD stream @30fps? which android device do they use? and the cable?

matkatz, 2019-05-30:
I tried to reproduce this issue on Pixel 2 (Android 9), Samsung S8 (Android 9) and FireFly RK3399 (Android 8.1) with no success.
I will try to get other platforms next week and hopefully reproduce the issue.
As for the high resolution streaming, all of the platforms above were capable to stream 720p depth with 1080p color but with significant frame drops, I get about 20 FPS for the depth stream and 15 FPS for the color.

GucciPrada, 2019-05-31:
> I tried to reproduce this issue on Pixel 2 (Android 9), Samsung S8 (Android 9) and FireFly RK3399 (Android 8.1) with no success.
> I will try to get other platforms next week and hopefully reproduce [...]

Thank you! That is quite informative. What about the USB 3.0 cable or adapter you used? any advice or recommendation?

matkatz, 2019-06-02:
I'm not sure what type my cable is it but you can probably test your cable with [realsense-viewer] (https://github.com/IntelRealSense/librealsense/releases/download/v2.21.0/Intel.RealSense.Viewer.exe) over Windows/Linux

GucciPrada, 2019-06-03:
OK, Thank you! I tested my cable with realsense-viewer on Macbook. I'll test it on Windows later.

GucciPrada, 2019-06-04:
I tried with my new Android Phone, Honor V20 (another brand of Huawei), the device was enumerated as usb2.1 device but actually the Phone is declared to support USB 3.1 Gen1. And this time, the same 'out-of-memory' problem happened with firmware 5.11.4 & librealsense 2.22.0 or 2.21.0.  Here is the log while running firmware 5.11.4+librealsense 2.21.0:
<details>
<summary>Click me to expand</summary>

```
2019-06-04 14:14:57.981 28713-28713/com.intel.realsense.capture I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
2019-06-04 14:14:58.140 28713-28713/com.intel.realsense.capture I/librs UsbUtilities: grantUsbPermissions:
    device: UsbDevice[mName=/dev/bus/usb/002/002,mVendorId=32902,mProductId=2771,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Intel(R) RealSense(TM) Depth Camera 415 ,mProductName=Intel(R) RealSense(TM) Depth Camera 415 ,mVersion=50.114,mSerialNumber=null,mSubProductId =-1,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=55,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=135,mAttributes=3,mMaxPacketSize=64,mInterval=6]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Y,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=4,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2019-06-04 14:15:18.593 28713-28713/com.intel.realsense.capture W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@e6c6c4e
2019-06-04 14:15:18.621 28713-28713/com.intel.realsense.capture I/librs Enumerator: onReceive: USB_CONTROL_PERMISSION
2019-06-04 14:15:18.623 28713-28752/com.intel.realsense.capture I/librs MessagesHandler: handleMessage: realsense device attached
2019-06-04 14:15:18.623 28713-28752/com.intel.realsense.capture I/librs Enumerator: notifyOnAttach
2019-06-04 14:15:18.624 28713-28752/com.intel.realsense.capture I/usbhost: usb_device_new /dev/bus/usb/002/002 fd: 64
2019-06-04 14:15:18.624 28713-28752/com.intel.realsense.capture I/usbhost: usb_device_new read returned 1230 errno 19
2019-06-04 14:15:18.645 28713-28752/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.645 28713-28752/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.672 28713-28713/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.672 28713-28713/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.677 28713-28713/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.677 28713-28713/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- @
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:15:18.679 28713-28713/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
[... 74 lines cut ...]
    susb specification- 210
2019-06-04 14:00:37.874 27346-27346/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- P
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:00:37.875 27346-27346/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- P
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-04 14:00:37.875 27346-27346/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- P
    path- /dev/bus/usb/002/002
    susb specification- 210
```

</details>

dreifelix, 2019-06-06:
Encountered this issue as well, using D435 + Pixel3 and D435 + Samsung S7 (same results). Both devices are not rooted, and are detected as usb 210. Firmware 5.11.04, librealsense 2.22. Tried both `camera` and `capture` apps.

For `camera` app, it starts with black screen, then the Settings page is displayed after a few seconds. I tried selecting different settings, including the lowest frame rate and resolution and only one active stream, but I still get the Out of memory issue.  

Next, I tried building the AAR and created my own app. Same result for both devices. Let me know if you want me to try something else.  

See sample log below, if it helps:  

<details>  
<summary>capture app logs for D435 + S7</summary>

``` 06/06 21:53:29: Launching capture
$ adb shell am start -n "com.intel.realsense.capture/com.intel.realsense.capture.MainActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Client not ready yet..Waiting for process to come online
Connected to process 20619 on device samsung-sm_g930f-192.168.43.1:5555
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/zygote64: no shared libraies, dex_files: 1
I/InstantRun: starting instant run server: is main process
D/librs UsbUtilities: hasUsbPermission
D/librs UsbUtilities: grantUsbPermissions
I/librs UsbUtilities: grantUsbPermissions:
    device: UsbDevice[mName=/dev/bus/usb/001/002,mVendorId=32902,mProductId=2823,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Intel(R) RealSense(TM) Depth Camera 435 ,mProductName=Intel(R) RealSense(TM) Depth Camera 435 ,mVersion=2.16,mSerialNumber=null,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=55,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 435 with RGB Module Depth,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=135,mAttributes=3,mMaxPacketSize=64,mInterval=6]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 435 with RGB Module Depth,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 435  Y,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 435 with RGB Module RGB,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=4,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 435 with RGB Module RGB,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
D/librs: Librealsense VERSION: 2.22.0
D/OpenGLRenderer: HWUI GL Pipeline
D/ViewRootImpl@a2905e4[MainActivity]: setView = DecorView@59b2a4d[MainActivity] TM=true MM=false
D/ViewRootImpl@a2905e4[MainActivity]: dispatchAttachedToWindow
V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
D/ViewRootImpl@a2905e4[MainActivity]: Relayout returned: old=[0,0][0,0] new=[0,0][1080,1920] result=0x7 surface={valid=true 529795960832} changed=true
I/OpenGLRenderer: Initialized EGL, version 1.4
D/OpenGLRenderer: Swap behavior 2
D/libGLESv1: STS_GLApi : DTS, ODTC are not allowed for Package : com.intel.realsense.capture
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
D/OpenGLRenderer: eglCreateWindowSurface = 0x7b5a4f64a0
D/SurfaceView: BG show() Surface(name=Background for - SurfaceView - com.intel.realsense.capture/com.intel.realsense.capture.MainActivity@e82a902@0) com.intel.realsense.librealsense.GLRsSurfaceView{e82a902 V.E...... ......ID 0,0-1080,1920 #7f070037 app:id/glSurfaceView}
V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
D/SurfaceView: surfaceCreated 1 com.intel.realsense.librealsense.GLRsSurfaceView{e82a902 V.E...... ......ID 0,0-1080,1920 #7f070037 app:id/glSurfaceView}
D/SurfaceView: surfaceChanged (1080,1920) 1 com.intel.realsense.librealsense.GLRsSurfaceView{e82a902 V.E...... ......ID 0,0-1080,1920 #7f070037 app:id/glSurfaceView}
D/ViewRootImpl@a2905e4[MainActivity]: MSG_RESIZED_REPORT: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 63 - 0, 0) vi=Rect(0, 63 - 0, 0) or=1
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:2
D/ViewRootImpl@a2905e4[MainActivity]: Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x1 surface={valid=true 529795960832} changed=false
D/ViewRootImpl@a2905e4[MainActivity]: MSG_WINDOW_FOCUS_CHANGED 1
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@6acc213 nm : com.intel.realsense.capture ic=null
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
D/librs: Librealsense VERSION: 2.22.0
    Librealsense VERSION: 2.22.0
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@8629c50 nm : com.intel.realsense.capture ic=null
D/ViewRootImpl@a2905e4[MainActivity]: MSG_RESIZED: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 63 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
[... cut ...]
    time_diff_keeper::update_diff_time - unlock
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
W/librs: bulk_transfer returned error, endpoint: �, error: Out of memory
```
</details>

matkatz, 2019-06-06:
Thanks for the inputs!
I'm trying to put my finger on the changes that cause this issue.
I understand that it does not exists on 2.19.2 and it does on 2.21.0
What about 2.20.0 (uploaded to Bintray today)?

matkatz, 2019-06-07:
There is a change on USB transfer method on v2.21.0 that might cause this issue.
I created a branch that partly revert this change: [android_bulk_transfer](https://github.com/matkatz/librealsense/tree/android_bulk_transfer)
Its purpose is only to verify that this is indeed the cause so it might be very buggy.
You can also consume it by changing the bintray repo to:
 https://dl.bintray.com/intel-realsense/librealsense-dev
And specify version 2.23.0.372

dreifelix, 2019-06-07:
Happy to share that your branch works! I have tested it on S7 + D435, no errors in the log. Will test it on the Pixel3 on Monday when I have the device with me. It worked with all three streams (color, depth, infrared) turned on simultaneously at low resolution and 15 fps each.  

Thanks so much for the fix!

GucciPrada, 2019-06-08:
Thanks!  I'll give it a try and report the results at here.

GucciPrada, 2019-06-11:
I tried librealsense 2.20.0 + firmware 5.11.04 on Honor V20 android phone, the example `capture` can work normally now.
Here is the log:
<details>
<summary>Click me to expand</summary>

```
2019-06-11 11:05:24.597 31246-31246/com.intel.realsense.capture I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
2019-06-11 11:05:24.597 31246-31246/com.intel.realsense.capture I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
2019-06-11 11:05:24.625 31246-31246/com.intel.realsense.capture I/librs UsbUtilities: grantUsbPermissions:
    device: UsbDevice[mName=/dev/bus/usb/002/002,mVendorId=32902,mProductId=2771,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Intel(R) RealSense(TM) Depth Camera 415 ,mProductName=Intel(R) RealSense(TM) Depth Camera 415 ,mVersion=50.114,mSerialNumber=null,mSubProductId =-1,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=55,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=135,mAttributes=3,mMaxPacketSize=64,mInterval=6]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Y,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=4,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2019-06-11 11:05:24.781 31246-31276/com.intel.realsense.capture W/libEGL: EGLNativeWindowType 0x7cbe276010 disconnect failed
2019-06-11 11:05:24.795 31246-31271/com.intel.realsense.capture W/libEGL: EGLNativeWindowType 0x7cbfc3f010 disconnect failed
2019-06-11 11:05:24.804 31246-31246/com.intel.realsense.capture I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
2019-06-11 11:05:28.181 31246-31246/com.intel.realsense.capture W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@cc8dd49
2019-06-11 11:05:28.222 31246-31246/com.intel.realsense.capture I/librs Enumerator: onReceive: USB_CONTROL_PERMISSION
2019-06-11 11:05:28.224 31246-31273/com.intel.realsense.capture I/librs MessagesHandler: handleMessage: realsense device attached
2019-06-11 11:05:28.224 31246-31273/com.intel.realsense.capture I/librs Enumerator: notifyOnAttach
2019-06-11 11:05:28.225 31246-31273/com.intel.realsense.capture I/usbhost: usb_device_new /dev/bus/usb/002/002 fd: 75
2019-06-11 11:05:28.225 31246-31273/com.intel.realsense.capture I/usbhost: usb_device_new read returned 1230 errno 0
2019-06-11 11:05:28.245 31246-31273/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.245 31246-31273/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.245 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.245 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.246 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.246 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.247 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.247 31246-31246/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- K
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:05:28.314 31246-31356/com.intel.realsense.capture E/librs: Cannot queue request: Out of memory
2019-06-11 11:05:28.319 31246-31246/com.intel.realsense.capture I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
2019-06-11 11:05:31.978 31246-31265/com.intel.realsense.capture I/alsense.captur: Starting profile saver IsSaveProfileNow end.
```

</details>

GucciPrada, 2019-06-11:
And the example app `capture` built with librealsense-dev 2.23.0.372 can also work normally.
Here is the log:
<details>
<summary>Click me to expand</summary>

```
2019-06-11 11:20:43.757 1412-1412/com.intel.realsense.capture I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
2019-06-11 11:20:43.795 1412-1412/com.intel.realsense.capture I/librs UsbUtilities: grantUsbPermissions:
    device: UsbDevice[mName=/dev/bus/usb/002/002,mVendorId=32902,mProductId=2771,mClass=239,mSubclass=2,mProtocol=1,mManufacturerName=Intel(R) RealSense(TM) Depth Camera 415 ,mProductName=Intel(R) RealSense(TM) Depth Camera 415 ,mVersion=50.114,mSerialNumber=null,mSubProductId =-1,mConfigurations=[
    UsbConfiguration[mId=1,mName=null,mAttributes=192,mMaxPower=55,mInterfaces=[
    UsbInterface[mId=0,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=135,mAttributes=3,mMaxPacketSize=64,mInterval=6]]
    UsbInterface[mId=1,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Depth,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=130,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=2,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  Y,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=131,mAttributes=2,mMaxPacketSize=512,mInterval=0]]
    UsbInterface[mId=3,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=1,mProtocol=0,mEndpoints=[]
    UsbInterface[mId=4,mAlternateSetting=0,mName=Intel(R) RealSense(TM) Depth Camera 415  RGB,mClass=14,mSubclass=2,mProtocol=0,mEndpoints=[
    UsbEndpoint[mAddress=132,mAttributes=2,mMaxPacketSize=512,mInterval=0]]]]
2019-06-11 11:20:43.951 1412-1446/com.intel.realsense.capture W/libEGL: EGLNativeWindowType 0x7cac44f010 disconnect failed
2019-06-11 11:20:43.965 1412-1442/com.intel.realsense.capture W/libEGL: EGLNativeWindowType 0x7cac4e6010 disconnect failed
2019-06-11 11:20:43.976 1412-1412/com.intel.realsense.capture I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
2019-06-11 11:20:45.569 1412-1412/com.intel.realsense.capture W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@cc8dd49
2019-06-11 11:20:45.611 1412-1412/com.intel.realsense.capture I/librs Enumerator: onReceive: USB_CONTROL_PERMISSION
2019-06-11 11:20:45.613 1412-1444/com.intel.realsense.capture I/librs MessagesHandler: handleMessage: realsense device attached
2019-06-11 11:20:45.613 1412-1444/com.intel.realsense.capture I/librs Enumerator: notifyOnAttach
2019-06-11 11:20:45.614 1412-1444/com.intel.realsense.capture I/usbhost: usb_device_new /dev/bus/usb/002/002 fd: 77
2019-06-11 11:20:45.614 1412-1444/com.intel.realsense.capture I/usbhost: usb_device_new read returned 1230 errno 0
2019-06-11 11:20:45.615 1412-1444/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.615 1412-1444/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.615 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.615 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.616 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.616 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.616 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 0
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.616 1412-1412/com.intel.realsense.capture I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- ad3
    mi- 3
    unique_id- /dev/bus/usb/002/002
    path- /dev/bus/usb/002/002
    susb specification- 210
2019-06-11 11:20:45.672 1412-1515/com.intel.realsense.capture I/librs: USB pipe 130 reset successfully
2019-06-11 11:20:45.681 1412-1517/com.intel.realsense.capture I/librs: USB pipe 132 reset successfully
2019-06-11 11:20:49.215 1412-1428/com.intel.realsense.capture I/alsense.captur: Starting profile saver IsSaveProfileNow end.
```

</details>

GucciPrada, 2019-06-11:
Hi @matkatz ,
  Do you have any thoughts about the problem of the USB protocol version? That is, librealsense running on a USB 3.1 compliant android phone recognizes the D415 camera as a USB 2.1 device.

I've checked the cable and the phone using both a windows tool named 'ChipGenius' and the MacOS X System Report App on MacBook Pro. 

On windows, I plugged my Honor V20 android phone into the USB 3.2 type-c port of the PC laptop, using the same c-to-c cable that I used to connect the android phone with the D415 camera. And the ChipGenius give me a USB detection result of the Honor V20 phone as follow:
```
Device Description: USB Composite Device (HUAWEI PCT-AL10)
  Equipment Type: Supplier Customization

  Protocol version: USB 3.10
  Current speed: High Speed
  Power consumption: 504mA

  USB device ID: VID = 12D1 PID = 107E
Equipment serial number: 5EN0219223001846

Equipment supplier: HUAWEI
  Equipment Name: PCT-AL10
Equipment revision: 0299

  Host/Master Controller model: Unknown
```

As you can see, the phone + the cable is recognized as USB 3.1 device.

on MacBook Pro (Retina, 13-inch, Early 2015 version), I got almost the same result:
I think the `Speed:     Up to 5Gb/sec` implies that the phone is connected to USB 3.0 port of my MacBook.

[image]

So why it's `susb specification- 210` when the D415 is connected to the same phone using the same cable? Am I missing something?

matkatz, 2019-06-11:
Thanks for the feedback!

I'm testing the library on 4 different platforms:
- FireFly RK3328 Android 7
- FireFly RK3399 Android 8.1
- Samsung S8 Android 9
- Pixel 2 Android 9

On all of those systems the camera(D415 FW 5.11.4.0) is identified as USB 3.2.
It is possible that the camera will be detected as USB 2 in some cases, but unplug and plug back the camera should solve that.

I'm still not capable to reproduce the memory issue on those platforms, currently I'm working on both directions:
- Trying to reproduce and solve the memory issue (preferred).
- Revert the USB bulk transfer changes (this is causing other problems that I still need to solve)

GucciPrada, 2019-06-16:
I tried to build the example `capture` with librealsense-dev 2.23.0.372, and ran it on Huawei P20.
something similar to issue #4215 happened. See this screencast video: 

[image]

You could see that the frames were fitfully updated, the screen got stuck for a while and got updated later again, off and on repeatedly.
I could see from the logcat a warning and an error message alternately appeared:
```
W/librs: Timeout reached waiting for response
E/librs: Cancel request failed: Invalid argument
```
The warning log seems to be similar to issue # 3612.

But after switching librealsense to v2.20.0, and build the `capture` again, It worked correctly. It even could deliver 1280x720x30fps RGBD stream, since the D415 was enumerated as USB 3.2 device.

matkatz, 2019-06-16:
I was finally able to reproduce the issue and able to see where is it coming from.
From a reason that I’m not yet understand, the USB bulk transfer started to fail when the frame resolution size was larger than ~600KB.

To solve that I changed the read buffer size from default hard coded size to a value that is determined according to the current stream resolution.

Reducing the buffer size allows to configure and stream lower resolutions in case of “Out of memory” error.

I also tried to reassemble the frame from few smaller USB payloads but that didn’t work for me (I think it requires isochronous endpoint which is not supported by Android).

BTW, a reboot to the device that the issue was reproduced on (Pixel 2 - Android 9), eliminated the issue.

I will keep looking into this but first I want to apply the fix if it's mitigates the issue.

Once again I will ask you to check my fix before I push it to the development branch.

This is the branch: [android-read-length-fix](https://github.com/matkatz/librealsense/tree/android-read-length-fix)
And it is also available in the dev repo (https://dl.bintray.com/intel-realsense/librealsense-dev) on bintray, version: 2.23.0.13

Thanks in advance!

GucciPrada, 2019-06-17:
That's great! I'd like to try it out later in this day

dreifelix, 2019-06-17:
The `android-read-length-fix` branch fails to stream using the `camera` sample app, even at lowest resolution at 6 fps when tested on my S7 + D435.  

<details>
  <summary>Logs on first run, with D435 already attached to USB port</summary>  

```
$ adb shell am start -n "com.intel.realsense.camera/com.intel.realsense.camera.DetachedActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Client not ready yet..Waiting for process to come online
Connected to process 14436 on device samsung-sm_g930f-192.168.43.1:5555
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/zygote64: no shared libraies, dex_files: 1
I/InstantRun: starting instant run server: is main process
D/librs: Librealsense VERSION: 2.23.0
D/librs UsbUtilities: hasUsbPermission
I/librs MessagesHandler: handleMessage: realsense device attached
I/librs Enumerator: notifyOnAttach
D/librs DeviceWatcher: Adding device: /dev/bus/usb/001/002
D/librs: AddUsbDevice, previous device count: 0
    AddUsbDevice, adding device: /dev/bus/usb/001/002, descriptor: 75
    AddUsbDevice, create
D/librs: AddUsbDevice, notify
I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- b07
    mi- 0
    unique_id- /dev/bus/usb/001/002
    path- /dev/bus/usb/001/002
    susb specification- 210
    Found UVC Device vid: id- 
    vid- 8086
    pid- b07
    mi- 3
    unique_id- /dev/bus/usb/001/002
    path- /dev/bus/usb/001/002
    susb specification- 210
D/librs: AddUsbDevice, current device count: 1
I/librs: Found UVC Device vid: id- 
    vid- 8086
    pid- b07
    mi- 0
    unique_id- /dev/bus/usb/001/002
    path- /dev/bus/usb/001/002
    susb specification- 210
    Found UVC Device vid: id- 
    vid- 8086
    pid- b07
    mi- 3
    unique_id- /dev/bus/usb/001/002
    path- /dev/bus/usb/001/002
    susb specification- 210
D/librs: try_fetch_usb_device(...) failed.
D/librs: Creating UVC Device from path: /dev/bus/usb/001/002
D/librs: interrupt_request: 1, 3, 0, 7, 0, 0, 
E/librs: SetXU result:-1
D/librs: interrupt_request: 1, 3, 0, 7, 0, 0, 
D/OpenGLRenderer: HWUI GL Pipeline
D/ViewRootImpl@edad6cd[DetachedActivity]: setView = DecorView@2e1b82[DetachedActivity] TM=true MM=false
D/ViewRootImpl@edad6cd[DetachedActivity]: dispatchAttachedToWindow
D/librs: Creating UVC Device from path: /dev/bus/usb/001/002
V/Surface: sf_framedrop debug : 0x4f4c, game : false, logging : 0
D/ViewRootImpl@edad6cd[DetachedActivity]: Relayout returned: old=[0,0][0,0] new=[0,0][1080,1920] result=0x7 surface={valid=true 506472833024} changed=true
I/OpenGLRenderer: Initialized EGL, version 1.4
D/OpenGLRenderer: Swap behavior 2
D/libGLESv1: STS_GLApi : DTS, ODTC are not allowed for Package : com.intel.realsense.camera
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
D/OpenGLRenderer: eglCreateWindowSurface = 0x75e64b1a40
D/librs DeviceWatcher: Device: /dev/bus/usb/001/002 added successfully
D/ViewRootImpl@edad6cd[DetachedActivity]: MSG_RESIZED_REPORT: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 63 - 0, 0) vi=Rect(0, 63 - 0, 0) or=1
D/ViewRootImpl@edad6cd[DetachedActivity]: MSG_WINDOW_FOCUS_CHANGED 1
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@24b12c9 nm : com.intel.realsense.camera ic=null
[... 596 lines cut ...]
D/ViewRootImpl@c065583[SettingsActivity]: Relayout returned: old=[0,0][0,0] new=[0,0][1080,1920] result=0x7 surface={valid=true 505948639232} changed=true
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, egl_color_buffer_format *, EGLBoolean) returns 0x3000,  [1080x1920]-format:1
D/OpenGLRenderer: eglCreateWindowSurface = 0x75e64b19f0
D/AbsListView:  in onLayout changed 
D/AbsListView:  in onLayout changed 
I/zygote64: Do full code cache collection, code=504KB, data=342KB
I/zygote64: After code cache collection, code=503KB, data=303KB
D/AbsListView:  in onLayout changed 
D/ViewRootImpl@c065583[SettingsActivity]: MSG_RESIZED_REPORT: frame=Rect(0, 0 - 1080, 1920) ci=Rect(0, 63 - 0, 0) vi=Rect(0, 63 - 0, 0) or=1
D/ViewRootImpl@c065583[SettingsActivity]: MSG_WINDOW_FOCUS_CHANGED 1
V/InputMethodManager: Starting input: tba=android.view.inputmethod.EditorInfo@b2208f0 nm : com.intel.realsense.camera ic=null
I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
I/zygote64: Do partial code cache collection, code=504KB, data=310KB
    After code cache collection, code=504KB, data=310KB
    Increasing code cache capacity to 2MB
D/OpenGLRenderer: eglDestroySurface = 0x75ce20e630
D/ViewRootImpl@68badc3[DetachedActivity]: Relayout returned: old=[0,0][1080,1920] new=[0,0][1080,1920] result=0x5 surface={valid=false 0} changed=true
D/ViewRootImpl@68badc3[DetachedActivity]: dispatchDetachedFromWindow
D/InputEventReceiver: channel 'fa99c0a com.intel.realsense.camera/com.intel.realsense.camera.DetachedActivity (client)' ~ Disposing input event receiver.
    channel 'fa99c0a com.intel.realsense.camera/com.intel.realsense.camera.DetachedActivity (client)' ~NativeInputEventReceiver.
D/ViewRootImpl@a51e193[Toast]: dispatchDetachedFromWindow
D/InputEventReceiver: channel '58399ab Toast (client)' ~ Disposing input event receiver.
    channel '58399ab Toast (client)' ~NativeInputEventReceiver.
```
</details>

GucciPrada, 2019-06-17:
I got a similar result to the one from @dreifelix. I was running the example app `capture` on Huawei P20 + librealsense-dev v2.23.0.13.
Here is the code snippet to config and start the stream:
[image]

The critical error in logcat:
```
2019-06-17 23:22:47.432 26320-27063/com.intel.realsense.capture W/librs: bulk_transfer returned error, endpoint: 130, error: Out of memory, number: 12
2019-06-17 23:22:47.432 26320-27063/com.intel.realsense.capture E/librs: USB transfer failed, the max read buffer size is smaller than the current resolution requires, try to configure lower resolution
```

<details>
<summary>Click to expand the detailed log</summary>

```
2019-06-17 23:22:19.037 26320-26320/? I/zygote64: Late-enabling -Xcheck:jni
2019-06-17 23:22:19.052 26320-26320/? I/zygote64: Reinit property: dalvik.vm.checkjni= false
2019-06-17 23:22:19.068 26320-26320/? D/ActivityThread: Attach thread to application
2019-06-17 23:22:19.098 26320-26320/com.intel.realsense.capture D/HwFLClassLoader: get used feature list :/feature/used-list failed!
2019-06-17 23:22:19.098 26320-26320/com.intel.realsense.capture D/HwFLClassLoader: USE_FEATURE_LIST had not init! 
2019-06-17 23:22:19.111 26320-26320/com.intel.realsense.capture E/MemoryLeakMonitorManager: MemoryLeakMonitor.jar is not exist!
2019-06-17 23:22:19.112 26320-26320/com.intel.realsense.capture E/Minikin: Could not get cmap table size!
2019-06-17 23:22:19.124 26320-26320/com.intel.realsense.capture I/HwCust: Constructor found for class android.app.HwCustActivityImpl
2019-06-17 23:22:19.124 26320-26320/com.intel.realsense.capture D/HwCust: Create obj success use class android.app.HwCustActivityImpl
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64: Rejecting re-init on previously-failed class java.lang.Class<android.support.v4.view.ViewCompat$OnUnhandledKeyEventListenerWrapper>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/view/View$OnUnhandledKeyEventListener;
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.support.v4.view.ViewCompat.setOnApplyWindowInsetsListener(android.view.View, android.support.v4.view.OnApplyWindowInsetsListener) (ViewCompat.java:2203)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at android.view.ViewGroup android.support.v7.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:637)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.support.v7.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:518)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.support.v7.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:466)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.support.v7.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:140)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void com.intel.realsense.capture.MainActivity.onCreate(android.os.Bundle) (MainActivity.java:45)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.Activity.performCreate(android.os.Bundle, android.os.PersistableBundle) (Activity.java:7352)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:7343)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1219)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:3110)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:3268)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1888)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:109)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.os.Looper.loop() (Looper.java:166)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:7367)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2019-06-17 23:22:19.158 26320-26320/com.intel.realsense.capture I/zygote64:     at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:469)
[... cut ...]
2019-06-17 23:22:47.438 26320-27064/com.intel.realsense.capture I/librs: USB pipe 132 reset successfully
2019-06-17 23:22:48.443 26320-26320/com.intel.realsense.capture E/librs capture example: streaming, error: UNKNOWN_ERROR
2019-06-17 23:22:48.448 26320-26320/com.intel.realsense.capture D/ActivityThread: Remove activity client record, r= ActivityRecord{97431af token=android.os.BinderProxy@3c1d5b6 {com.intel.realsense.capture/com.intel.realsense.librealsense.DeviceWatcherActivity}} token= android.os.BinderProxy@3c1d5b6

```
</details>

matkatz, 2019-06-17:
Thanks again for your help!

@dreifelix, do you see the same behavior with the Pixel? did you try to reboot the phone?
Also, I see some strange issues with specific configurations (6 FPS) on my platforms when connecting via USB2.
If you get to test it again please try to config only depth 640x480x30 as first step.

I also want to share with you the reason for trying to solve this issue and not revert the backend changes.

This is the performance of v2.23.0 (when it does work (: )
[image]

While this is v2.20.0 results:
[image]

I'll keep digging...

dreifelix, 2019-06-18:
**Rebooting helped! :)**

In my S7, I am able to stream up to 60 fps when running 2 streams at a time, and up to 30 fps when running all 3 streams simultaneously, at the lowest resolution (424x240 for color, 480x270 for infrared and depth). At 640x480 resolution, only infrared works, up to 30 fps. It's probably the S7's available memory that's limiting the streams now - it only has 512MB.

I noticed the performance improvement as well. I wasn't able to reach the above frame rates in the `android_bulk_transfer` branch - the streams were freezing intermittently at 30 fps when running simultaneously.

As you noted, I am unable to run any of the streams at 6 fps at any resolution, but that's fine :) I will try it on the Pixel 3 when I get the chance - i'm just borrowing the device from another project team.

Looking forward to have your changes merged in the next release. Thanks so much for looking into this!

matkatz, 2019-06-18:
Thanks @dreifelix 
I still need to understand what is bringing the phone to this state.

@GucciPrada, I guess reboot didn't help with the P20?

GucciPrada, 2019-06-19:
Hi @matkatz I've never tried  to reboot the phone. I'll try it today.

GucciPrada, 2019-06-25:
Sorry for this late report!

I've tried the librealsense-dev 2.23.0.13, ran `capture` on Huawei P20 for 640x480@30fps RGBD. 
Before restarting the phone, I got the same result as before: the app cannot receive any frame correctly. 
<details>
<summary> log before Huawei P20 restarted</summary>

```
2019-06-25 23:11:37.497 2406-2406/? I/zygote64: Late-enabling -Xcheck:jni
2019-06-25 23:11:37.511 2406-2406/? I/zygote64: Reinit property: dalvik.vm.checkjni= false
2019-06-25 23:11:37.524 2406-2406/? D/ActivityThread: Attach thread to application
2019-06-25 23:11:37.550 2406-2406/? D/HwFLClassLoader: get used feature list :/feature/used-list failed!
2019-06-25 23:11:37.550 2406-2406/? D/HwFLClassLoader: USE_FEATURE_LIST had not init! 
2019-06-25 23:11:37.565 2406-2413/? I/zygote64: Debugger is no longer active
2019-06-25 23:11:37.568 2406-2406/? E/MemoryLeakMonitorManager: MemoryLeakMonitor.jar is not exist!
2019-06-25 23:11:37.569 2406-2406/? E/Minikin: Could not get cmap table size!
2019-06-25 23:11:37.579 2406-2406/? I/HwCust: Constructor found for class android.app.HwCustActivityImpl
2019-06-25 23:11:37.579 2406-2406/? D/HwCust: Create obj success use class android.app.HwCustActivityImpl
2019-06-25 23:11:37.611 2406-2406/? I/zygote64: Rejecting re-init on previously-failed class java.lang.Class<android.support.v4.view.ViewCompat$OnUnhandledKeyEventListenerWrapper>: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/view/View$OnUnhandledKeyEventListener;
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.support.v4.view.ViewCompat.setOnApplyWindowInsetsListener(android.view.View, android.support.v4.view.OnApplyWindowInsetsListener) (ViewCompat.java:2203)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at android.view.ViewGroup android.support.v7.app.AppCompatDelegateImpl.createSubDecor() (AppCompatDelegateImpl.java:637)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.support.v7.app.AppCompatDelegateImpl.ensureSubDecor() (AppCompatDelegateImpl.java:518)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.support.v7.app.AppCompatDelegateImpl.setContentView(int) (AppCompatDelegateImpl.java:466)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.support.v7.app.AppCompatActivity.setContentView(int) (AppCompatActivity.java:140)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void com.intel.realsense.capture.MainActivity.onCreate(android.os.Bundle) (MainActivity.java:45)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.Activity.performCreate(android.os.Bundle, android.os.PersistableBundle) (Activity.java:7352)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.Activity.performCreate(android.os.Bundle) (Activity.java:7343)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.Instrumentation.callActivityOnCreate(android.app.Activity, android.os.Bundle) (Instrumentation.java:1219)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at android.app.Activity android.app.ActivityThread.performLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent) (ActivityThread.java:3110)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.ActivityThread.handleLaunchActivity(android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:3268)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.ActivityThread.-wrap12(android.app.ActivityThread, android.app.ActivityThread$ActivityClientRecord, android.content.Intent, java.lang.String) (ActivityThread.java:-1)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.ActivityThread$H.handleMessage(android.os.Message) (ActivityThread.java:1888)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.os.Handler.dispatchMessage(android.os.Message) (Handler.java:109)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.os.Looper.loop() (Looper.java:166)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void android.app.ActivityThread.main(java.lang.String[]) (ActivityThread.java:7367)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at java.lang.Object java.lang.reflect.Method.invoke(java.lang.Object, java.lang.Object[]) (Method.java:-2)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run() (RuntimeInit.java:469)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at void com.android.internal.os.ZygoteInit.main(java.lang.String[]) (ZygoteInit.java:963)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64: Caused by: java.lang.ClassNotFoundException: Didn't find class "android.view.View$OnUnhandledKeyEventListener" on path: DexPathList[[zip file "/data/app/com.intel.realsense.capture-IVc-vmqH7F7WSw1gEP4S2w==/base.apk"],nativeLibraryDirectories=[/data/app/com.intel.realsense.capture-IVc-vmqH7F7WSw1gEP4S2w==/lib/arm64, /data/app/com.intel.realsense.capture-IVc-vmqH7F7WSw1gEP4S2w==/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64, /product/lib64]]
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:125)
2019-06-25 23:11:37.611 2406-2406/? I/zygote64:     at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:379)
[... cut ...]
   Then I tried to kill the app several times or tried to keep the D415 plugged into the phone, then reopen the `capture`  app, everything was ok again. That was wired! 

And one more trial,  after the phone was rebooted, I reinstalled the `capture` app through Android Studio. Then I got the same error result as the very beginning, i.e. both color and depth stream was stuck while the D415 plugged in.  So I tried to reboot the phone and reopen or kill the `capture` app with the D415 plugged in, sometimes the app could work correctly.

In a word, it's not stable and I didn't find a consistent way to make the app work correctly.

matkatz, 2019-06-30:
Thanks @GucciPrada.
I don't have a solution for the "Out of memory" state yet.
Will update once I'll have some news regarding this issue.

GucciPrada, 2019-07-01:
OK,  thanks!

matkatz, 2019-07-11:
@xtrawurst  let's continue the conversation on this thread.

I'm trying to revert the bulk transfer changes and use USB requests instead.
After removing some locks and extra copies I'm getting the same performance as I got with bulk transfer, hopefully without  the OOM error.
Since I'm not capable to reproduce that error on my devices I wanted to ask you to help me verify that I'm on the right direction.
This is the branch: [android-usb-request](https://github.com/matkatz/librealsense/tree/android-usb-request)
You can also consume it by changing the bintray repo to:
https://dl.bintray.com/intel-realsense/librealsense-dev
And specify version 2.24.0.67

xtrawurst, 2019-07-11:
@matkatz 
Great, I'll give it a try and let you know!

xtrawurst, 2019-07-11:
@matkatz 
Confirmed! I no longer see the issue on my Samsung Galaxy Tab S4, even when using Depth @ 1280x720 and RGB @ 1920x1080. Will test on other devices too.

Quick question: What's, roughly, the processing being done CPU-sided on each frame in libRS after the USB transmission completes? At full size capture the frame rate on my S4 is below 30 fps and I'm trying to figure out if it's
a) The USB transfer taking a long time,
b) CPU-sided processing in libRS or
c) CPU-sided processing on our end.

GucciPrada, 2019-07-13:
@matkatz 
Glad to hear this. I'll give it a try today.

@xtrawurst 
May I ask, what frame rate specifically you got at full-size capture(Depth @ 1280x720 and RGB @ 1920x1080)?

matkatz, 2019-07-15:
@xtrawurst 
Thanks! 
This is good news, except that now I get OOM error on high bandwidth configuration with devices that worked with no issues.
So I need to check that now... ):

As for processing, depends on the image format you request, a conversion may be done over the CPU.
For example requesting RGB8 requires more CPU than YUYV which is the native format and doesn't requires any processing except a single memcopy.

On the USB transactions I see strange behavior with the color stream on high resolutions (720p & 1080p).
In both resolutions the transaction time seems to be longer than the time measured on the depth stream although the resolution, bytes per pixel and frequency are identical for the 720p.
Currently with both 720p and 1080p color I get 15 FPS, while depth streams @ ~27-30 FPS

GucciPrada, 2019-07-15:
Hi @matkatz 
I got the same good news as @xtrawurst, that the example app `capture` running on Huawei P20 device can deliver 1920x1080 Color and 1280x720 Depth both at 15fps:
[image]

it can also deliver 640x360 Depth@23-26fps and 1280x720 Color @15fps.
The firmware version is still 5.11.04 on my D415 camera. I'll update it to the newest and test it again.

Yes, it's really weird that the color stream ran at a much lower frame rate than the depth stream with the same resolution and pixel size.

matkatz, 2019-07-15:
@GucciPrada 
Thanks for the update, I hope to push this change to the development branch soon, but first I need to understand why my RK3399 started to show OOM error.

ndiremdjian, 2019-07-23:
Hi Guys, 
Not sure if this issue is fully SW related, I was experiencing the same on Sony Xperia but I came to suspect my USB3 cable (I used so far a 2m from nedis). 
With this cable the capture app last about 27s before crashing with the bulk_transfer error above at 60fps VGA.
Lowering frame rate was just delaying the crash (30 fps about 50s etc...)

I then switched for another cable (1m from belkin) and I wasn't able to reproduce the error anymore.

Sounds a bit too magic and witchware but still, if it may help one of you...

Cheers

GucciPrada, 2019-07-24:
Hi @ndiremdjian 
Which version of librealsense and firmware did you use?
When built with librealsense v2.22.0 and firmware 5.11.4, the capture app, running at VGA@30fps,  didn't crash on my phone with a USB3.1 type-c to type-c cable.

And thanks for your tips, I will try some other cables.

ndiremdjian, 2019-07-24:
Hi 
I have the same versions as yours. My initial cable was a 2m 3.1 type c to type c 5Gbps.
My second cable was a Thunderbolt 3 USB-C 20Gbps with backward compatibility USB3.1 10Gbps.
With the later, no error is showing up even at 60fps VGA with Depth and both IR streams on.
Will try to add RGB as well.

Cheers

xtrawurst, 2019-08-10:
@matkatz Thanks for the information. Any updates on the Android-USB front?

matkatz, 2019-08-13:
I will try to merge #4428 to the development branch before our next release.

xtrawurst, 2019-08-13:
@matkatz Thanks

> On the USB transactions I see strange behavior with the color stream on high resolutions (720p & 1080p).
In both resolutions the transaction time seems to be longer than the time measured on the depth stream although the resolution, bytes per pixel and frequency are identical for the 720p.
Currently with both 720p and 1080p color I get 15 FPS, while depth streams @ ~27-30 FPS

Has this been resolved?

matkatz, 2019-08-13:
Not yet, currently both implementations shows this behavior.

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@GucciPrada @xtrawurst The latest SDK v2.30 has improved android stability according to our testing. Could you please try if you still get any issue? Thanks!

GucciPrada, 2019-11-07:
Yes, I will try it in the following days.
Thank you!

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@GucciPrada Thanks! Looking forward to your update.

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@GucciPrada @xtrawurst Any update with SDK v2.30? Thanks!

kafan1986, 2019-11-21:
@matkatz @RealSenseCustomerSupport I have received same OOM error on SDK 2.30.0 on android 9 (S922x SBC). My colour config is 1920x1080x15fps and depth is 1280x720x15. It works OK for few minutes and then crashes.

`
I/librs: endpoint 132 read buffer size: 4148224
I/librs: USB pipe 132 reset successfully
I/librs: usb_request_queue returned error, endpoint: 132 error: Out of memory, number: 12
E/librs: failed to submit UVC request, error: -11
`

GucciPrada, 2019-11-26:
Sorry for this late reply.
I ran the [RS Camera App v2.30.0.1174](https://play.google.com/store/apps/details?id=com.intel.realsense.camera&hl=en) on two Android Phones, Honor V20 and Huawei P30 Pro. 
The app could not open a single DEPTH stream(1280x720@15fps or 30fps) on those phones.
Specifically, when I tried changing to the 15fps configuration on the `Settings` page, the App showed me a toast message:"Failed to set streaming configuration". Here is the captured [log file](https://drive.google.com/file/d/1LAvHfX3zdXwUxrgafXXg4Hk8-61EqQwo/view?usp=sharing) when I changed the single DEPTH stream from 1280x720@6fps to 1280x720@15fps.
But it could open a single 1280x720@6fps DEPTH stream or RGBD stream(RGB8-1280x720@6fps and DEPTH-1280x720@6fps).
Here is the [log.txt](https://drive.google.com/file/d/1Qc6w6lS2m3oEh4bpL8Gfd4ZiznmZ5PK8/view?usp=sharing) while runing the 1280x720@6fps RGBD stream.

I guess the reason is: the connection type is considered as USB 2.1
I might need to try more Android Phones or USB cables.
[image]

kafan1986, 2019-12-02:
@matkatz @RealSenseCustomerSupport 
At 15 fps with RGB at 1280x720 and Depth at 1280x720, the camera will throw lots of error and would eventually would stop producing frames, after few minutes. The only way I could make it work stable for longer duration was to set FPS at 6, at 6 Hz no error is thrown for a long time and I could stream RGB at rather highest resolution of 1920x1080 and Depth at 1280x720 simultaneously without crashes for longer period. After around 3-4 hours, I saw the depth frame was getting updated but somehow the RGB stream was stuck on some previous frame. Stats displayed 0 fps for RGB stream. I was able to resume the camera at same 1920x1080 resolution, without any restart or reset by simply going to the settings activity and then going back to preview activity.

Also FYIP, my device is USB 3.0, but the test app displays it as 3.2, not sure if it affects stability in any way.

The error which was thrown after 3-4 hours was this:
```
2019-12-02 06:26:22.086 7396-7396/com.intel.realsense.camera E/librs api: hwmon command 0x10 failed. Error type:  (1).
2019-12-02 06:26:22.632 7396-7804/com.intel.realsense.camera E/librs: failed to submit UVC request, error: -11

```

I really need this to work at 15 Hz stable. Do confirm whether solving this issue is on your high on your priority or not. I need to make some business decision regarding my product design accordingly. 

@rafaelspring Did you ever share your hacked libusb code to Matkatz, which did not face same "out of memory" issue. I saw a previous thread regarding the same.

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@GucciPrada As your connection type is USB 2.1, only 6fps is supported for 1280x720. If you want higher fps such as 15fps or 30fps, you need switch to USB3 connection.

@kafan1986 What's your android device when you get the issue with latest v2.30? And could you please provice the full logcat log when the issue happened? Thanks!

kafan1986, 2019-12-02:
> @GucciPrada As your connection type is USB 2.1, only 6fps is supported for 1280x720. If you want higher fps such as 15fps or 30fps, you need switch to USB3 connection.
>
> @kafan1986 What's your [...]

My device is Odroid N2 running android 9 and I am using SDK v 2.30. I have pasted logs on separate issue thread https://github.com/IntelRealSense/librealsense/issues/5328

kafan1986, 2019-12-05:
@matkatz @RealSenseCustomerSupport @dorodnic @RealSense-Customer-Engineering 
I have tested and tried many things on android platform and none of them helped with the Out of memory error. I am using Odroid N2 SBC running android 9 through USB 3.0. At 15 or even 6 fps I get OOM error after few minutes and the device stalls. Depth set at 1280x720 and colour at 1920x1080.

I have written custom android code to recover the device using hardware_reset API provided by librealsense SDK but it keeps on trying to recover device and I have also tried it with and without Odroid's solution to reset the USB hub but without much success as even after reboot most times at least one the stream is stuck. The code for this class is as below.

[Streamer.java.txt](https://github.com/IntelRealSense/librealsense/files/3924799/Streamer.java.txt)

Following this https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485 , https://stackoverflow.com/questions/23005708/kernel-crashes-due-to-oom-error-usb-submit-urb , https://discuss.aerospike.com/t/how-to-tune-the-linux-kernel-for-memory-performance/4195 I increased usbfs_memory_mb value.
```
/sys/module/usbcore/parameters/usbfs_memory_mb 128        #Increased from default 16mb 

/proc/sys/vm/min_free_kbytes 40000                                        #Changed to 1% percent of my device 4GB RAM
```
Following https://www.bo-yang.net/2015/03/30/debug-kernel-space-memory-leak I looked for any memory leaks. The plots and logs are attached below.

[image]
[image]
[monitor_output_localhost.txt](https://github.com/IntelRealSense/librealsense/files/3924687/monitor_output_localhost.txt)

Part of dmesg from android during error is attached below.
[android_dmesg.txt](https://github.com/IntelRealSense/librealsense/files/3924690/android_dmesg.txt)

The logcat messages generated by librs is attached below.
[logcat.txt](https://github.com/IntelRealSense/librealsense/files/3924692/logcat.txt)

I have asked this forum multiple times but haven't heard anything promising. Given the issue was first reported almost 6-7 months back and has not been resolved yet, I hope someone soon takes a look at this issue and gets this resolved.

matkatz, 2019-12-08:
@alowenst01 please take a look.

kafan1986, 2019-12-13:
@alowenst01 @matkatz Was isochronous data transfer ever tried for Android? I found a repo where someone implemented the same for Android. https://github.com/Peter-St/Android-UVC-Camera

kafan1986, 2019-12-16:
**UPDATE**
I am running streaming depth and colour frame on android for last 36+ hours at 720p 15fps without any crash. I do get some messages like below every few minutes but it has yet to give any additional critical error like OOM.

`W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32`

Basically currently my code is just getting the depth and colour frame with just some post processing filter and nothing else. Top command shows the CPU usage to be around 40-45%. Earlier I was also drawing the eachb colour frame to UI and usually the app would crash in minutes.

It means although the data transfer is the culprit and the memory leak is not in the SDK itself but in the way how the RSUSB is implemented. When the CPU is not stressed then RSUSB can work fine but when it is stressed the USB request fail and then I believe the method of cancelling USB request, that should cancel the filled URB queues held with the linux kernel is not working properly releasing the queue and thus causes Out of memory error. 

As expected in such case hardware reset or even custom USB controller reset is not able to recover and only option is to reboot the entire system.

RealSenseCustomerSupport, 2020-02-01:
----------------------------------------------

@Kafan1986 Glad to see 720p 15fps work for you. We did see some stability issue with HD or FHD resolution due to Android platform limitation. So please use the working configuration as workaround at this point. Thanks!

kafan1986, 2020-02-01:
> @kafan1986 Glad to see 720p 15fps work for you. We did see some stability issue with HD or FHD resolution due to Android platform limitation. So please use the working configuration as workaround at this point. Thanks!

Actually I posted a little earlier. Although the camera was working but the colour frame was stuck, which I did not know at that point of time. Currently at 720p the device works from 4-7 hrs before it eventually faces the colour frame stuck or out of memory error. Happens in 1 - 1.5 hrs when set up at 1080p.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@kafan1986 So previously what you said "I am running streaming depth and colour frame on android for last 36+ hours at 720p 15fps without any crash" is not always what you get, right? What format did you set for color frame? Could you please try YUYV format to see if any improvement?

kafan1986, 2020-02-03:
> @kafan1986 So previously what you said "I am running streaming depth and colour frame on android for last 36+ hours at 720p 15fps without any crash" is not always what you get, right? What format did you set for color frame? Could you please try YUYV format to see if any improvement?

What I mean is that when I made that comment, my test setup was incorrect. I was working on the depth data only. When one streams color frame too then the system will crash quickly on Android. This either happens silently, i.e. the color stream gets stuck and then one needs to either check visually or check by comparing frame id with previous frame id to know this programmatically; otherwise the it throws out of memory error and it leads to frame timeout.

At 720p colour frame stream this eventually happens every 5-7 hours and at 1080p it happens within 1-2 hours. In all above cases, the only confirmed way to recover is to reboot the entire system. "Resetting" device through official API or even resetting USB controller on the system does not guarantee recovery.

rafaelspring, 2020-02-25:
@matkatz @RealSenseCustomerSupport @RealSense-Customer-Engineering @dorodnic 

Sorry for the late silence. I changed my GitHub name (used to be @xtrawurst).

We have tried the latest 2.32.1 librealsense and we still get frequent loss of connection on the Samsung Galaxy Tab S4 running Android 9.

Interestingly we sometimes get a more stable connection using the "RS Camera" Android app instead of our capturing app, even if we use the same configuration (resolutions and frame rate).

Would it be possible to open-source the RS Camera app code so the community can see how it performs configuration etc?

[9 more replies]

closed 2020-06-05 by Mad-Thanos