realsenseai/librealsense #4095: D415: depth data comparison between realsense viewer and python wrapper

redM0nk, 2019-05-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       | 
| Firmware Version                   | 05.11.04.00                                             | 
| Operating System & Version |  Win 10                                                   |   
| SDK Version                           |  pyrealsense2 (version- 2.22.0.864)        | 
| Language                               | python                                                    | 
| Segment			        |  other                                                      | 

### Issue Description
Getting different depth data between Intel realsense viewer and python script.

**Test Setup:** 
Milk can placed at _1.151m_ away from the camera. I'm measuring the distance at pixel _(342, 240)_

**Intel Realsense viewer:** 
[RGB frame](https://drive.google.com/open?id=1J8XuP7O_Q92TDT0_wrs3FPpvBg-Hr8wZ
)
**Depth frame**
[Depth frame (after post processing filters)](https://drive.google.com/open?id=1FHenTq7MLRfOHE57LtT1J41RENYS6bVU)

**Python script (provided below) output:**

> Enabling emitter
> Emitter Enabled? 1.0
> depth at (342, 240): 1282mm

**Python example code:**

```python

    pipeline = rs.pipeline()  # type: rs.pipeline
    config = rs.config()
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)
    profile = pipeline.start(config)  # type: rs.pipeline_profile
    device = profile.get_device()  # type: rs.device
    depth_sensor = device.first_depth_sensor()  # type: rs.depth_sensor

    if depth_sensor.supports(rs.option.emitter_enabled):
        print('Enabling emitter...')
        depth_sensor.set_option(rs.option.emitter_enabled, 1.0)

    print('-------------')
    is_emitted_enabled = depth_sensor.get_option(rs.option.emitter_enabled)
    print('Emitter Enabled? {}'.format(is_emitted_enabled))
    print('-------------')

    # Skip 5 first frames to give the Auto-Exposure time to adjust
    for index in range(5):
        pipeline.wait_for_frames()

    colorizer = rs.colorizer()
    spatial = rs.spatial_filter()
    spatial.set_option(rs.option.filter_magnitude, 5)
    spatial.set_option(rs.option.filter_smooth_alpha, 1)
    spatial.set_option(rs.option.filter_smooth_delta, 50)
    spatial.set_option(rs.option.holes_fill, 3)
    temporal = rs.temporal_filter()
    decimation = rs.decimation_filter()
    decimation.set_option(rs.option.filter_magnitude, 4)
    hole_filling = rs.hole_filling_filter()

    depth_to_disparity = rs.disparity_transform(False)
    disparity_to_depth = rs.disparity_transform(False)

    frames = []
    for index in range(10):
        frameset = pipeline.wait_for_frames()
        frames.append(frameset.get_depth_frame())
    # take the last frame and save it as a colored graph
    colorized_depth = np.asanyarray(colorizer.colorize(frames[-1]).get_data())
    plt.imshow(colorized_depth)
    plt.savefig('depth_frame_before_filters.png')
    frame = None
    for index in range(len(frames)):
        frame = frames[index]
        #frame = decimation.process(frame)
        #frame = depth_to_disparity.process(frame)
        frame = spatial.process(frame)
        frame = temporal.process(frame)
        #frame = disparity_to_depth.process(frame)
        frame = hole_filling.process(frame)

    colorized_depth = np.asanyarray(colorizer.colorize(frame).get_data())
    depth_arr_np = np.asanyarray(frame.get_data())
    print('depth at (342, 240): {}mm'.format(depth_arr_np[342,240]))
    plt.imshow(colorized_depth)
    plt.savefig('depth_frame_after_filters.png')
    pipeline.stop()
```

The Realsense viewer.exe is very close to the exact value (1.13-1.14m. actual value ~1.15m) whereas the python script is far off from the actual value (1.282m, actual value ~1.15m).
I'm not sure if I'm missing any parameters in the python script. The result stays same with/without the post-processing functions.

Any help is appreciated.

ev-mp, 2019-05-29:
@redM0nk hello,
IMO you should starts with checking the filters, as these obviously manipulate depth data.
And of those the hole_filling filter can introduce the most unexpected artifacts, so I'd start with isolating that first.

There is also a modification you made, so now all the filters are applied to depth rather than disparity data - which we do not recommend.
Is it possible that you switched the disparity transform off due to the issue in these lines? -
>    depth_to_disparity = rs.disparity_transform(False)
    disparity_to_depth = rs.disparity_transform(**~~False~~**True)

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@redM0nk Any update from your side with the suggestion of ev-mp? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@redM0nk Any other questions about this ticket?

RealSenseCustomerSupport, 2019-07-08:
----------------------------------------------

@redM0nk If no other questions about this ticket, close it accordingly. Thanks!

closed 2019-07-08 by RealSenseCustomerSupport