realsenseai/librealsense #4096: using kinfu

gowooyan, 2019-05-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<22>.<0> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I'm going to use the kinfu example
However, this error occurs
I followed the installation as it was, but it's like this
I don't know what the problem is
I need help. 

[image]

gowooyan, 2019-05-29:
also I am debuging this part 
[image]

point_queue and normals_queue's dim value is zero... 

I thought that this is causing the error.
[image]

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@gowooyan Sorry for late reply! Did you still get the issue now?  And is the camera working well with other samples?

RealSenseCustomerSupport, 2019-07-05:
----------------------------------------------

@gowooyan If no other questions, will close the ticket.

closed 2019-07-05 by RealSenseCustomerSupport

aalarcon96, 2019-09-11:
What was the solution to this problem? I am also getting this error.